espressif/esp-idf #601: undefined reference to `gpio_intr_handler_register'

ankitdaf, 2017-05-14:
Trying to call the method  gpio_intr_handler_register(gpio_intr_handler_fn_t fn, void *arg) throws an undefined reference error. 

This happens in spite of having included  "rom/gpio.h" in the source. Printing the verbose build command shows that there is a section "-T esp32.common.ld" which contains the location of the method, but apparently that is not enough. I suspect that "rom/gpio.h" is not in the include path, and adding it might cause this error to go away 

Can someone suggest a good way of fixing this ? I am using eclipse to build, and don't want to stick in a hardcoded build path

ankitdaf, 2017-05-14:
Similar with gpio_intr_pending

kewalmshah, 2017-05-18:
Hi @ankitdaf  interrupt functions in rom/gpio.h are generally not to be used in sdk(as these functions are in ROM and slow enough to execute can cause issues with FreeRTOS, you can check the description of each function).  These functions are generally used during bootloader. You can use driver/gpio.h for interrupt functions.

ankitdaf, 2017-05-18:
@kewalmshah 
Well, there are a couple of methods that I'd like to use which don't seem to be available in driver/gpio.h .
Specifically, I want to check : 

1. Check which GPIO pin has an interrupt pending with gpio_intr_pending
2. Call one handler to service all interrupts with gpio_intr_handler_register

Both of these were available in the ESP8266 non-os SDK, and I am porting to the 32. They are essential because if not available the alternative will be to setup interrupts for every gpio (about 10 of them) as well as individual handlers for each of them. 

Is there a better way ?

costaud, 2017-05-19:
I think gpio_isr_register can set the single handler for all gpio interrupts.

closed 2017-11-21 by FayeY