espressif/esp-idf #603: Self heat stops ble/wifi

ahmedjouirou, 2017-05-14:
Hi , i have a problem with my ESP32 (thing board from sparkfun) , i put it in a HCI continuos listening for BLE ADV packets , after 2 minutes or so it's still working ( printf and GPIO stuff) but BLE stops recieving packets . at first i was connecting also to wifi as station with BLE Wifi Coexists enabled in meuconfig , after ward i stopped the connecion to wifi and just debugged if the BLE will stops also , and in eather ways it STOPs listening. I putted the module in the FREEZER and every thing works absolutly fine for several hours . the module gets hot ( like 70 degrees or some like that) outside the freezer , and i think its drifting something inside the chip or even the crystal it self ... so please any work around to void poutting my module in the freezer while continiously listening to ble and preferrably always connectec to wifi also .
Thanks

edmund-huber, 2017-05-16:
Hi @ahmedjouirou , for what it's worth, the Sparkfun ESP32 Thing doesn't seem to have any heat spreading features. You might want to try the same experiment with a board based on the WROOM-32 module, such as the board that Adafruit is selling. If you're looking for a workaround this might be the most expedient route..

ahmedjouirou, 2017-05-17:
well i tried the ESP32-S module , same problem , it stops after 5 of 7 minutes from receiving ble

FHFS, 2017-05-26:
It might not be because of heat. #618 when I had CONFIG_SW_COEXIST_ENABLE on, Bluetooth would init but wouldn't receive advertisements. Try increasing your scan interval while keeping your window the same or lower.

TianHao-Espressif, 2017-06-08:
@ahmedjouirou , ESP32 can bear high temperature.  I'm agree @FHFS . Could you disable the CONFIG_SW_COEXIST_ENABLE, then report the result？Thanks.

FayeY, 2017-07-07:
Hi @ahmedjouirou, is there any update? Did the above suggestion help you resolve the issue?

closed 2017-07-13 by FayeY