espressif/esp-idf #605: 'mbedtls_pk_parse_public_key' always complaining about key

ThisNameIsNotAllowed, 2017-05-15:
Using `mbedtls_pk_parse_public_key` will always result in:  
`PK - The algorithm tag or value is invalid : ASN1 - ASN1 tag was of an unexpected value`  
```
  int error_code = 0;

//only for github (Key taken from "https://tls.mbed.org/kb/cryptography/asn1-key-structures-in-der-and-pem"
  unsigned char internal_key[] =  "-----BEGIN PUBLIC KEY-----\n"\
          "MIGfMA0GCSqGSIb3DQEBAQUAA4GNADCBiQKBgQDMYfnvWtC8Id5bPKae5yXSxQTt\n"\
          "+Zpul6AnnZWfI2TtIarvjHBFUtXRo96y7hoL4VWOPKGCsRqMFDkrbeUjRrx8iL91\n"\
          "4/srnyf6sh9c8Zk04xEOpK1ypvBz+Ks4uZObtjnnitf0NBGdjMKxveTq+VE7BWUI\n"\
          "yQjtQ8mbDOsiLLvh7wIDAQAB\n"\
          "-----END PUBLIC KEY-----\n";

  mbedtls_pk_context pk;
  mbedtls_pk_init(&pk);

  ESP_LOGI(MY_TAG, "READING PUBLIC KEY TO VARIABLE\n");
  //Reading rsa public key
  if( ( error_code =  mbedtls_pk_parse_public_key(&pk, &internal_key, sizeof(internal_key))) == 0){

    ESP_LOGI(MY_TAG, "READING SUCCESSFULL\n");
  } else {
    ESP_LOGE(MY_TAG, "ERROR READING PUBLIC KEY\n");
    char error_output[500];
    mbedtls_strerror(error_code, &error_output, sizeof(error_output));
    ESP_LOGE(MY_TAG, "%s\n", &error_output[0]);
  }
```

Spritetm, 2017-05-15:
That is because you're passing a pointer to the array containing the key text to mbedtls_pk_parse_public_key. Remove the & in front of internal_key in that call and it should work.

ThisNameIsNotAllowed, 2017-05-15:
Sry, this has been my fault while changing code to open this issue.  
Even with this line changed (also changed it in the initial comment), it's not working as expected.  
`if( ( error_code =  mbedtls_pk_parse_public_key(&pk, internal_key, sizeof(internal_key))) == 0){`  
still generates:  
`PK - The algorithm tag or value is invalid : ASN1 - ASN1 tag was of an unexpected value`

Spritetm, 2017-05-16:
Huh. This indeed seems to be ESP32-specific; I copy-pasted your code to run on my laptop and it works just fine.

ThisNameIsNotAllowed, 2017-05-17:
In case you pass the public key in "cer" format, there won't be any complains.
The "mbedtls" website tells you to use keys in "pem" format, but as shown above, these will cause errors.

projectgus, 2017-08-22:
Hi @ThisNameIsNotAllowed ,

Sorry for the very slow reply. I'm not able to reproduce using the code snippet you gave above, either on the V2.1 release or the current master branch (fc85cb683).

The full code I used is: https://gist.github.com/projectgus/bcf22209af2a6e5bc21da6b90214b887

My best guess is that you may have experienced a memory corruption bug, similar to #711, if you included `mbedtls/config.h` directly from your app's source. See the comments at the end of the linked issue for a discussion of the bug and a workaround. I can't tell for sure if this was the issue without seeing a full code sample.

I'm going to close this issue, but please reopen if you have steps (and full code) to reproduce on either the v2.1 release or the current master branch.

Angus

closed 2017-08-22 by projectgus