espressif/esp-idf #606: I2C module: problems with STOP operation after END

rojer, 2017-05-15:
My code performs a long read operation in several READ/END, READ/END, ..., READ/STOP operations.
It all works as expected, until the last batch, which ends in STOP, where I2C master behaves weirdly: it gets stuck, generating SCL continuously.
 
Here's a logic capture of the operation: 
[image]
The operation involves reading 34 bytes in 5 transactions:
1) Start, address 0x60, for reading, read 1 byte (which is the length of the packet): 
[image]
2) Read 10 bytes, continue (ends with an END opcode): 
[image]
3) Same as 2 (read 10, END): 
[image]
4) One more: 
[image]
5) And final transaction, which is going to be the last one, consists of the following op pipeline:
```
esp32_i2c_exec           00001003 2
esp32_i2c_exec           00001401 2
esp32_i2c_exec           00001800 3
```
which is READ 3 bytes with ack, then READ 1 with a NAK (as per the spec, final byte should be NAK'd by the master) and a STOP. However, it does not go as planned and instead i see this on the logic analyzer: 
[image]

This is bizarre. a START condition, followed by continuous SCL pulsing.
The execution of the commands continues ad infinitum, however, my waiting loop terminates after 10000 iterations and i see this:
```
esp32_i2c_exec         10000 loops, ok? 0, ints 0x00000242 -> 0x00000242, tx_fifo 0/-1 rx_fifo 0/4, status 0x33000010
esp32_i2c_exec           00001003 2
esp32_i2c_exec           00001401 2
esp32_i2c_exec           00001800 3
```
no commands completed, nothing received.

i found a workaround for this: do not end the sequence with STOP, finish with END, and then execute a STOP as a single-command pipeline. strangely, i also had to remove the "read with NAK" operation, because is also somehow confuses the FSM.
so, for final step i have tried these pipelines:
a) 1003, 1401, 1800 -> SCL stuck
b) 1003, 1401, 2000 -> SCL stuck
c) 1004, 1800 -> SCL stuck
d) 1004, 2000 -> ok, then in separate transaction 1800 -> ok
FWIW, this last option produces the expected result on the analyzer: 
[image] and 
[image]

rojer, 2017-05-15:
fwiw, i'm attaching full capture file which can be examined with the [SIGMA software](http://www.asix.net/dwnld_sigma-omega.htm)

[stop_bug.stf.zip](https://github.com/espressif/esp-idf/files/1001338/stop_bug.stf.zip)

rojer, 2017-05-15:
happens on write too

kewalmshah, 2017-05-17:
Hi @rojer Was just going through your issue! Can I have a look at your test code please ? It seems to have some changes in i2c.c driver file, is it ?

rojer, 2017-05-17:
our I2C driver is completely different and does not use IDF's i2c.c in any way. [here](https://github.com/cesanta/mongoose-os/blob/05b136d73abb58b461794380c2f31d1219ee01d9/fw/platforms/esp32/src/esp32_i2c_master.c) it is.

PerMalmberg, 2017-09-10:
+1 for this. I've seen the same thing symptom, though I haven't been able to determine why it happens.

stickbreaker, 2017-11-25:
@PerMalmberg I have rewritten the I2C driver for the arduino-esp32. [Here](https://github.com/stickbreaker/arduino-esp32) It solves this problem.

I believe you are running into a querk of the i2c Statemachine.  When you used the `END` command, you cannot reuse that position when you refill the queue.  It has to remain an `END` command.  Also, it is illegal to have a `START` precede a `END`. my I2C rewrite uses an ISR to service the StateMachine(SM), the fillCmdQueue() function enforces these rules.  

@rojer Your original problem can be explained if during your last fill of the Command[] queue, you reused the `END` entry with a `STOP`.  It is easy to duplicate.

Chuck.

rojer, 2017-11-25:
@stickbreaker thanks Chuck. i actually noticed updated language in the latest update to the TRM about clearing the end ints and also not reusing the position previosuly used by the end command. i made a note to revisit my driver, but haven't got around to it yet. it's good to get a confirmation that the issue can be solved, thanks!

stickbreaker, 2017-11-25:
What is TRM?  I am from the Arduino fork.

Chuck.

rojer, 2017-11-25:
[ESP32 techinical reference manual](http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf).
latest update was about I2C and mentions this quirk.

stickbreaker, 2017-11-25:
Thanks

Chuck.

rojer, 2017-11-25:
ok, i'm going to close this issue, since the cause and workaround is now known.

closed 2017-11-25 by rojer

stickbreaker, 2017-11-25:
Wow, they really butchered it.  I read a previous version, this version has changed the definitions for the interrupt registers, It looks like they are removing all references to FIFO.

The Fifo_conf register only describe NON fifo usage.
And the interrupt control registers int_* omit rx_fifo_Full,tx_fifo_empty,rx_fifo_ovf, slave_tran_comp

Are the deleting the Fifo from the hardware?

Chuck.