espressif/esp-idf #609: Hello world not compiling

harshcht, 2017-05-16:
I installed espressif idf on ubuntu everything went fine as per the instructions but when I try to run the example hello_world it gives me this error message:
missing destination file operand after '/examples/get-started/hello_world'
Try 'cp --help' for more information.

can anyone please help?

krzychb, 2017-05-16:
Hi @harshcht,

Do you see this error in step [Start Project](http://esp-idf.readthedocs.io/en/latest/get-started/index.html#start-a-project)?

If so, then maybe you are missing a `.` at the end in copy command:

```
cd ~/esp
cp -r $IDF_PATH/examples/get-started/hello_world .
```
This dot is the copy destination and means current directory, i.e. `~/esp`.

harshcht, 2017-05-18:
@krzychb thanks for the clarification.0 It compiles now.

closed 2017-05-18 by harshcht

sravanth005, 2019-03-20:
Hi all,
I am trying to work with ESP32-CAM (https://robotzero.one/esp32-camera-module/#comment-3073) i can able to take a still with the camera but face detection and face reorganization is not happening, Do any one know for what reason it is not happening.please help me out to solve this issue.

dhcps: send_offer>>udp_sendto result 0
I (63186) camera_httpd: JPG: 6335B 1573ms
I (66456) camera_httpd: vflip = 1
dhcps: send_offer>>udp_sendto result 0
I (72716) camera_httpd: JPG: 7267B 118ms
I (78976) camera_httpd: face_detect = 1
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (1752586) wifi: max connection, deauth!
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (1873156) wifi: max connection, deauth!
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (2301736) wifi: station: b4:ef:fa:79:e6:e2 leave, AID = 1, bss_flags is 134243
I (2301736) wifi: new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:6
I (2301736) camera wifi: station:b4:ef:fa:79:e6:e2leave, AID=1
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0

thanks in advance

projectgus, 2019-03-20:
@sravanth005 Is there a reason you commented on this existing issue?

You haven't specified, but if you're using the [esp-who](https://github.com/espressif/esp-who/) framework then please open an issue there if you think there's a bug. If you're using another hardware/software combination then please contact whoever produced that hardware/software.

projectgus, 2019-03-20:
@sravanth005 Also, I deleted the other comment that you copy-pasted to a different unrelated Github issue. Please do not spam github issues with comments that are unrelated to the issue. If you think there's a bug in ESP-IDF, open a new issue (however this doesn't look like a bug in ESP-IDF). If you have a general question, try the https://esp32.com/ forums.