espressif/esp-idf #610: Examples System/OTA - randomly fails

PhilColbert, 2017-05-16:
Followed the instructions, all compiles and runs fine.

Seems to randomly stop in the middle of writing the image.....

Attempt 1 
( esp running )
I (3774) ota: Have written image length 10621
I (3784) ota: Have written image length 11645
I (3804) ota: Have written image length 12669
I (3814) ota: Have written image length 13693
I (3814) ota: Have written image length 14413
I (11374) wifi: pm start, type:0
(esp hangs)

Reset and attempt 2
I (7514) ota: Have written image length 183509
I (7534) ota: Have written image length 184533
I (7564) ota: Have written image length 185557
I (7584) ota: Have written image length 186581
I (7604) ota: Have written image length 187605
I (7614) ota: Have written image length 188629
I (7614) ota: Have written image length 188761
I (11154) wifi: pm start, type:0

Reset and attempt 3 
I (3804) ota: Have written image length 33697
I (3804) ota: Have written image length 33981
I (3834) ota: Have written image length 35005
I (3834) ota: Have written image length 35441
I (11344) wifi: pm start, type:0

Reset and attempt 4

I (7514) ota: Have written image length 183509
I (7534) ota: Have written image length 184533
I (7564) ota: Have written image length 185557
I (7584) ota: Have written image length 186581
I (7604) ota: Have written image length 187605
I (7614) ota: Have written image length 188629
I (7614) ota: Have written image length 188761
I (11154) wifi: pm start, type:0

Any ideas?

PhilColbert, 2017-05-16:
Seems its perhaps a timings issue, if I put a vTaskDelay in the while(flag) loop then it has a much better chance of working, still only works now and then....

Or as soon as you put a lot of ESP_LOGI statements in the code, it works.....

projectgus, 2017-05-17:
Is there any possibility you could take a tcpdump from the server side, as the OTA download is happening?

projectgus, 2017-05-17:
> Is there any possibility you could take a tcpdump from the server side, as the OTA download is happening?

Never mind, I'm able to reproduce this here. Will let you know we find.

PhilColbert, 2017-05-17:
Sorry, I was asleep, in the UK here :)

Thanks for the help.

FHFS, 2017-05-17:
a vTaskDelay(20) fixed it for me, ESP_LOG did nothing for me

PhilColbert, 2017-05-17:
doesnt work for me... and it takes forever !

vTaskDelay(3) and some debug statements seems to be best, altho there will be a proper solution I am sure :)

ESP-EOS, 2017-05-20:
vTaskDelay(20) worked for me, too. Thanks for the advice.

BTW, I'm experiencing a similar issues with OTA on Arduino (some time works, but most of times there are errors during the sketch download). Has anybody the same problem?

skateone, 2017-05-25:
i also tried the vTaskDelay and gave me some degree of success but still failed on occasion. What worked for me was to reduce the BUFFER sizes from 1024 to 512... that gave me the best results.

rsnbusa, 2017-05-25:
For me it hangs (forever), always in the recv() call but never in the same place in the file, that is, sometimes it reads 20 lines sometime 100, but it NEVER ends and ALWAYS hangs on the recv().

EXACTLY the same application with the OLD library works perfectly EVERY TIME.

Tried the delay but to no avail. Also the buffer size 4096, 2048, 1024 and 512 they all hang in the new version and ALL work in the previous version.

If you compile with this version  the OTA example it works. My routines are a copy of this source code with minor mods.  Very disturbing.

PhilColbert, 2017-05-31:
Any news on this ? :)

projectgus, 2017-06-05:
Hi everyone,

Sorry for the slow update. This bug should be fixed in IDF master branch as of commit 365b66b9.

Angus

closed 2017-06-05 by projectgus

rsnbusa, 2017-06-05:
Thanks Ill give it a try.

rsnbusa, 2017-06-05:
Working as far as I can test it. Thanks

lixianyu, 2017-06-05:
3 Jun, I clone the master, this issue still have...

rsnbusa, 2017-06-05:
> 
> Closed #610 <https://github.com/espressif/esp-idf/issues/610>.
In my case its working, 700-800k about 10 times.

Regards.

projectgus, 2017-06-05:
> 3 Jun, I clone the master, this issue still have...

What is the output of the following two commands:
```
git describe
git submodule status
```

lixianyu, 2017-06-05:
@projectgus ,Hi,
Just now, I clone the master, it's work fine!^_^
Thank you.

PhilColbert, 2017-06-07:
Tested a few times, works great, thanks :)