espressif/esp-idf #614: OTA example hangs midway

eightdot, 2017-05-17:
The OTA example hangs midway recieving the image,

at that point it also stops responding to ping and arp, 

then (after some time)  looses wifi connection, and cannot reconnect

with [*] Run FreeRTOS only on first core  selected it works

hardware:
Ai-Thinker 	ESP-32S Breakout Board
also happens on custom board

build steps:
`mkdir ~/proj && cd ~/proj && mkdir esp32 && cd esp32`
`wget https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz`
`tar -xzf xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz`
`export PATH=$PATH:$HOME/proj/esp32/xtensa-esp32-elf/bin`
`git clone https://github.com/espressif/esp-idf `
`cd esp-idf/`
`git submodule update --init`
`export IDF_PATH=~/proj/esp32/esp-idf`
`cd .. && cp -r esp-idf/examples/system/ota ota && cd ota`
`make flash`
`stty -F /dev/ttyUSB0 115200 min 1 && cat /dev/ttyUSB0`

config changes:
> CONFIG_WIFI_SSID="tdwb"
> CONFIG_WIFI_PASSWORD="qwertyuiop"
> CONFIG_SERVER_IP="172.29.29.1"

git details
esp-idf: commit 1ddf40feb73386df26ac51bd80aac2210229b164 
Submodule path 'components/aws_iot/aws-iot-device-sdk-embedded-C': checked out '7132505b00d2dd57f48478e75efa636021919aae'
Submodule path 'components/bt/lib': checked out 'cbdf2b94b13a9e7df47c83a45fb82c71ba718f7c'
Submodule path 'components/coap/libcoap': checked out '6468887a12666f88b8704d797fc176cd4f40ee4c'
Submodule path 'components/esp32/lib': checked out '0c6d1e05a127dc1fef7c7b66d5ebf7c14111a77a'
Submodule path 'components/esptool_py/esptool': checked out '96698a3da9acc6e357741663830f97524b688ade'
Submodule path 'components/micro-ecc/micro-ecc': checked out '14222e062d77f45321676e813d9525f32a88e8fa'

eightdot, 2017-05-17:
log (hit reset after 1st try):

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4404
load:0x40078000,len:11092
load:0x40080000,len:252
entry 0x40080034
I (44) boot: ESP-IDF v2.0-rc1-813-g1ddf40fe 2nd stage bootloader
I (45) boot: compile time 10:19:05
I (78) boot: Enabling RNG early entropy source...
I (79) boot: SPI Speed      : 40MHz
I (79) boot: SPI Mode       : DIO
I (87) boot: SPI Flash Size : 4MB
I (100) boot: Partition Table:
I (111) boot: ## Label            Usage          Type ST Offset   Length
I (134) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (157) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (180) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (204) boot:  3 factory          factory app      00 00 00010000 00100000
I (227) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (250) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (273) boot: End of partition table
I (287) boot: Disabling RNG early entropy source...
I (304) boot: Loading app partition at offset 00010000
I (1004) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (1004) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x09d70 ( 40304) map
I (1021) boot: segment 2: paddr=0x00029d80 vaddr=0x3ffb0000 size=0x02af8 ( 11000) load
I (1052) boot: segment 3: paddr=0x0002c880 vaddr=0x40080000 size=0x00400 (  1024) load
I (1074) boot: segment 4: paddr=0x0002cc88 vaddr=0x40080400 size=0x13324 ( 78628) load
I (1138) boot: segment 5: paddr=0x0003ffb4 vaddr=0x400c0000 size=0x00000 (     0) load
I (1139) boot: segment 6: paddr=0x0003ffbc vaddr=0x00000000 size=0x0004c (    76) 
I (1159) boot: segment 7: paddr=0x00040010 vaddr=0x400d0018 size=0x47220 (291360) map
I (1185) cpu_start: Pro cpu up.
I (1196) cpu_start: Starting app cpu, entry point is 0x40080e60
I (0) cpu_start: App cpu up.
I (1229) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1252) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1272) heap_alloc_caps: At 3FFB8408 len 00027BF8 (158 KiB): DRAM
I (1293) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1315) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1336) heap_alloc_caps: At 40093724 len 0000C8DC (50 KiB): IRAM
I (1357) cpu_start: Pro cpu start user code
I (1415) cpu_start: Starting scheduler on PRO CPU.
I (202) cpu_start: Starting scheduler on APP CPU.
I (272) wifi: wifi firmware version: 6c86a1c
I (272) wifi: config NVS flash: enabled
I (272) wifi: config nano formating: disabled
I (272) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (282) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (312) wifi: Init dynamic tx buffer num: 32
I (312) wifi: Init dynamic rx buffer num: 32
I (312) wifi: wifi driver task: 3ffbce00, prio:23, stack:4096
I (322) wifi: Init static rx buffer num: 10
I (322) wifi: Init dynamic rx buffer num: 32
I (322) wifi: Init rx ampdu len mblock:7
I (332) wifi: Init lldesc rx ampdu entry mblock:4
I (332) wifi: wifi power manager task: 0x3ffc3904 prio: 21 stack: 2560
I (342) ota: Setting WiFi configuration SSID tdwb...
I (342) wifi: wifi timer task: 3ffc496c, prio:22, stack:3584
W (352) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (522) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 2
I (532) wifi: mode : sta (24:0a:c4:01:c1:44)
I (532) ota: Starting OTA example...
I (532) ota: Running partition type 0 subtype 0 (offset 0x00010000)
I (1252) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2232) wifi: state: init -> auth (b0)
[... 164 lines cut ...]
I (454702) wifi: pm stop, total sleep time: 0/448390486

I (454702) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (454702) ota: Connection closed, all packets received
I (454712) ota: Total Write binary data length : 37774
E (454762) esp_image: invalid segment length 0xffffffff
E (454762) ota: esp_ota_end failed!
E (454762) ota: Exiting task due to fatal error...
I (454792) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (454792) wifi: state: auth -> auth (b0)
I (454802) wifi: state: auth -> assoc (0)
I (454812) wifi: state: assoc -> run (10)
I (457922) wifi: state: run -> auth (3c0)
I (457922) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (458042) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (458042) wifi: state: auth -> auth (b0)
I (458042) wifi: state: auth -> assoc (0)
I (458052) wifi: state: assoc -> run (10)
I (461162) wifi: state: run -> auth (3c0)
I (461162) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (461282) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (461282) wifi: state: auth -> auth (b0)
I (461282) wifi: state: auth -> assoc (0)
I (461292) wifi: state: assoc -> run (10)
I (464402) wifi: state: run -> auth (3c0)

eightdot, 2017-05-17:
tcpdump (from the wandboard that run the AP and python -m SimpleHTTPServer 8070:

10:33:21.885309 IP 172.29.29.1 > 172.29.29.113: ICMP echo request, id 6070, seq 17, length 64
10:33:21.889900 IP 172.29.29.113 > 172.29.29.1: ICMP echo reply, id 6070, seq 17, length 64
10:33:22.887336 IP 172.29.29.1 > 172.29.29.113: ICMP echo request, id 6070, seq 18, length 64
10:33:22.894390 IP 172.29.29.113 > 172.29.29.1: ICMP echo reply, id 6070, seq 18, length 64
10:33:23.112294 IP 172.29.29.1.8070 > 172.29.29.113.48644: Flags [P.], seq 1:18, ack 59, win 29200, length 17
10:33:23.114004 IP 172.29.29.1.8070 > 172.29.29.113.48644: Flags [.], seq 18:1478, ack 59, win 29200, length 1460
10:33:23.114137 IP 172.29.29.1.8070 > 172.29.29.113.48644: Flags [.], seq 1478:2938, ack 59, win 29200, length 1460
10:33:23.114218 IP 172.29.29.1.8070 > 172.29.29.113.48644: Flags [P.], seq 2938:4398, ack 59, win 29200, length 1460
10:33:23.119414 IP 172.29.29.113.48644 > 172.29.29.1.8070: Flags [R.], seq 59, ack 18, win 5840, length 0
10:33:23.132596 IP 172.29.29.113.48644 > 172.29.29.1.8070: Flags [R.], seq 59, ack 1478, win 5840, length 0
10:33:23.133180 IP 172.29.29.113.48644 > 172.29.29.1.8070: Flags [R.], seq 59, ack 2938, win 5840, length 0
10:33:23.133802 IP 172.29.29.113.48644 > 172.29.29.1.8070: Flags [R.], seq 59, ack 4398, win 5840, length 0
10:33:23.137053 IP 172.29.29.1.8070 > 172.29.29.113.57619: Flags [P.], seq 1:18, ack 59, win 29200, length 17
10:33:23.138355 IP 172.29.29.1.8070 > 172.29.29.113.57619: Flags [.], seq 18:1478, ack 59, win 29200, length 1460
10:33:23.138493 IP 172.29.29.1.8070 > 172.29.29.113.57619: Flags [.], seq 1478:2938, ack 59, win 29200, length 1460
10:33:23.138649 IP 172.29.29.1.8070 > 172.29.29.113.57619: Flags [P.], seq 2938:4398, ack 59, win 29200, length 1460
10:33:23.144114 IP 172.29.29.113.57619 > 172.29.29.1.8070: Flags [R.], seq 59, ack 18, win 5840, length 0
10:33:23.147404 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [P.], seq 1:18, ack 59, win 29200, length 17
10:33:23.147766 IP 172.29.29.113.57619 > 172.29.29.1.8070: Flags [R.], seq 59, ack 1478, win 5840, length 0
10:33:23.148552 IP 172.29.29.113.57619 > 172.29.29.1.8070: Flags [R.], seq 59, ack 2938, win 5840, length 0
10:33:23.148895 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 18:1478, ack 59, win 29200, length 1460
10:33:23.148965 IP 172.29.29.113.57619 > 172.29.29.1.8070: Flags [R.], seq 59, ack 4398, win 5840, length 0
10:33:23.149084 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 1478:2938, ack 59, win 29200, length 1460
10:33:23.149183 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [P.], seq 2938:4398, ack 59, win 29200, length 1460
10:33:23.156966 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 1478, win 4363, length 0
10:33:23.168657 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 1478, win 5840, length 0
10:33:23.168785 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 4398:5858, ack 59, win 29200, length 1460
10:33:23.168894 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 5858:7318, ack 59, win 29200, length 1460
10:33:23.296610 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 4398, win 2920, length 0
10:33:23.388561 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 4398, win 4968, length 0
10:33:23.388676 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 7318:8778, ack 59, win 29200, length 1460
10:33:23.423120 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 7318, win 2048, length 0
10:33:23.425134 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 7318, win 3944, length 0
10:33:23.425218 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 8778:10238, ack 59, win 29200, length 1460
10:33:23.482535 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 7318, win 5840, length 0
10:33:23.482646 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [P.], seq 10238:11698, ack 59, win 29200, length 1460
10:33:23.482752 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 11698:13158, ack 59, win 29200, length 1460
10:33:23.544528 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 8778, win 5840, length 0
10:33:23.544668 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 13158:14618, ack 59, win 29200, length 1460
10:33:23.586494 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 10238, win 5840, length 0
10:33:23.586620 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [P.], seq 14618:16078, ack 59, win 29200, length 1460
10:33:23.645385 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 13158, win 2920, length 0
10:33:23.691880 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 13158, win 4380, length 0
10:33:23.691992 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 16078:17538, ack 59, win 29200, length 1460
10:33:23.733235 IP 172.29.29.113.13257 > 172.29.29.1.8070: Flags [.], ack 16078, win 1460, length 0
10:33:23.889023 IP 172.29.29.1 > 172.29.29.113: ICMP echo request, id 6070, seq 19, length 64
10:33:24.210729 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 16078:17538, ack 59, win 29200, length 1460
10:33:24.890889 IP 172.29.29.1 > 172.29.29.113: ICMP echo request, id 6070, seq 20, length 64
10:33:25.170741 IP 172.29.29.1.8070 > 172.29.29.113.13257: Flags [.], seq 16078:17538, ack 59, win 29200, length 1460
10:33:25.890975 IP 172.29.29.1 > 172.29.29.113: ICMP echo request, id 6070, seq 21, length 64
[... cut ...]
        0x0030:  0000 0000 0000 0000 0000 0004 7464 7762  ............tdwb
        0x0040:  0108 8b96 8284 0c18 3060 3014 0100 000f  ........0.0.....
        0x0050:  ac04 0100 000f ac04 0100 000f ac02 0000  ................
        0x0060:  3204 6c12 2448 2d1a 6e11 1eff 0000 0000  2.l.$H-.n.......
        0x0070:  0000 0000 0000 0001 0000 0000 0000 0000  ................
        0x0080:  0000 0000                                ....
11:01:28.189917 EAPOL key (3) v2, len 95
11:01:28.290820 EAPOL key (3) v2, len 95

negativekelvin, 2017-05-17:
https://github.com/espressif/esp-idf/issues/610

projectgus, 2017-05-18:
Thanks for reporting this @eightdot . As @negativekelvin mentions, I think this is a duplicate of #610 - which I'm looking into right now. I'm going to close this issue, but please follow the link to #610 and click the "Subscribe" button under "Notifications" on the right, in order to get updates on that issue.

closed 2017-05-18 by projectgus

eightdot, 2017-05-18:
ok