espressif/esp-idf #615: mbstowcs crashing

johsely1234, 2017-05-17:
Simple example, crashes

```
#include <iostream>
#include <clocale>
#include <cstdlib>

extern "C" void app_main(void)
{

    std::setlocale(LC_ALL, "en_US.utf8");
    std::wcout.imbue(std::locale("en_US.utf8"));
    const char* mbstr = u8"z\u00df\u6c34\U0001f34c"; // or u8"zß水🍌"
                        // or "\x7a\xc3\x9f\xe6\xb0\xb4\xf0\x9f\x8d\x8c";
    wchar_t wstr[5];
    std::mbstowcs(wstr, mbstr, 5);
    std::wcout << "wide string: " << wstr << '\n';
}
```

Anyone tried this?
wcstombs(...) works for me

igrr, 2017-05-17:
newlib supports only one locale — "C" locale. That doesn't explain crashing though. Do you have a stack trace which is normally decoded by the `make monitor` tool?

johsely1234, 2017-05-17:
```
abort() was called at PC 0x400e3d0f on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40009203:0x3ffb4bc0 0x00000000:0x3ffb4be0 0x400f26e5:0x3ffb4c00 0x400dd4dd:0x3ffb4c20 0x400dd9d0:0x3ffb4c70 0x400dc797:0x3ffb4cd0 0x400d1159:0x3ffb4d10

Entering gdb stub now.
$T1f#eb

```

I cant get any info in the gdb stub:

```
(gdb) target remote /dev/ttyUSB0
Remote debugging using /dev/ttyUSB0
0x40009203 in ?? ()
(gdb) bt
#0  0x40009203 in ?? ()
(gdb) 
```

But if I uncomment the locale stuff (first 2 lines) it works nicely, so I can work with this.

igrr, 2017-05-22:
Indeed, looking into libstdc++ source we find that for 'generic' locale option only "C" locale is supported. Support for other locales is delegated to the libc library. Attempting to create locale objects for unsupported locales causes an exception to be thrown. Because we build with `-fno-exceptions`, this results in an `abort()` being called.

Newlib library provided with the ESP-IDF is compiled without `--enable-newlib-mb`, disabling locales other than "C". It is possible to replace newlib provided with the ESP-IDF with a custom build, however we don't support this for now. We will consider adding an option to build newlib from source with custom configuration options in the next ESP-IDF version.

closed 2017-05-22 by igrr