espressif/esp-idf #616: SPI slave crash

gctronic, 2017-05-17:
Hi,
I was playing with SPI slave, I modified the example given in the esp-idf framework and attached and ESP32 module to another board (STM discovery) that is acting as SPI master.
The communication worked with commit 1a19747678129494245ceebc3b31788705a5db3d (27 April 2017), but after updating the esp-idf repo, the ESP32 keep crashing as soon as the SPI communication starts.
Trying come back to commit 1a19747678129494245ceebc3b31788705a5db3d and the communication works again.
Can you confirm my finding?

negativekelvin, 2017-05-17:
You will get more help if you post crash log/backtrace and relevant source code.

gctronic, 2017-05-18:
I narrowed the cause of the problem to commit f7e88565208451a7c3415be3a2b94d32377441f6.
As a workaround I setup FreeRTOS to run on a single core.

Spritetm, 2017-05-19:
As negativekelvin stated, can you perhaps give a crash dump or code causing it or something? Now all we know is that that commit made your code puke. I can go look for something that breaks and fix it now, but I have no idea if it's the same issue you're having; for all I know, it may be something in your code that is wrong but happened to work with the previous esp-idf.

Spritetm, 2017-05-22:
We've fixed something that could have caused your crash, (specifically in [this](https://github.com/espressif/esp-idf/commit/a38cecefa74afa7e619db6ccf1d5fa1ceb86d417) commit. Could you maybe pull the latest master and see if that solves the bug you're experiencing?

FayeY, 2017-07-07:
Hi @gctronic , any update? Is this problem solved now?

closed 2017-07-13 by FayeY