espressif/esp-idf #618: Memory leak. WiFi + BT fails after 1.5h of operating.

mkonnov, 2017-05-17:
Worked for ~1.5h (104min).
BT is scanning for nearby beacons and publishes a data related to each detected device into MQTT topic.

esp-idf #https://github.com/espressif/esp-idf/commit/1ddf40feb73386df26ac51bd80aac2210229b164 is used.
[wifi_fail_log.txt](https://github.com/espressif/esp-idf/files/1008671/wifi_fail_log.txt)

The following params used:

BLE gattc:
```
static esp_ble_scan_params_t ble_scan_params = {
        .scan_type                        = BLE_SCAN_TYPE_ACTIVE,
        .own_addr_type            = BLE_ADDR_TYPE_RPA_PUBLIC,
        .scan_filter_policy      = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_interval            = **220**,
        .scan_window                    = **25**
 };
```

From sdkconfig:
```
BTC_TASK_STACK_SIZE = 4096
SW_COEXIST_ENABLE = y
```

Scan duration is 60sec.

Tried to play with interval/window values. Seems to be a little helpful, e.g. after changing it from interval/window = 220/30 it increases the operating time before fail on ~15min.
But this values is inapplicable in our environment as we need longer window and shorter interval.

I've inserted the logging of free heap on every scan completion and it looks OK before wifi fail.
After such, it leaks without a hesitation.

Please pay an attention on my comments into log.

```
/*
 * ...
 *
 * A lot of similar logs reporting ~18000 bytes of free heap every minute when scan completes
 *
 * ...
 */

I (4701142) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (4701142) gattc: free heap: 18000
E (4761152) BT: btc_search_callback  BLE observe complete. Num Resp 130

I (4761152) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (4761152) gattc: free heap: 18000
E (4821162) BT: btc_search_callback  BLE observe complete. Num Resp 116

I (4821162) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (4821162) gattc: free heap: 17680
E (4881172) BT: btc_search_callback  BLE observe complete. Num Resp 78

I (4881172) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (4881172) gattc: free heap: 18000

/*
 * And here it starts the memory leak
 */

I (4930742) wifi: bcn_timout,ap_probe_send_start
I (4933252) wifi: ap_probe_send over, resett wifi status to disassoc
I (4933252) wifi: state: run -> init (1)
I (4933252) wifi: pm stop, total sleep time: 0/626181837

I (4933252) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
E (4934012) aws_iot: failed! mbedtls_ssl_write returned -0x4e

/*
 * An error of IoT_Error_t (NETWORK_ATTEMPTING_RECONNECT)
 */

E (4934202) mqtt_task: rc = 4
E (4941182) BT: btc_search_callback  BLE observe complete. Num Resp 253

I (4941182) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (4941182) gattc: free heap: 59384
E (4941792) aws_iot: failed! mbedtls_net_connect returned -0x52
E (4941792) mqtt_task: attempting reconnect failed, rc = 4

/*
 * After wifi fail, this fail occurs
 */

E (4949192) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
E (5001192) BT: btc_search_callback  BLE observe complete. Num Resp 253

I (5001192) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (5001192) gattc: free heap: 62564
E (5009172) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
E (5061202) BT: btc_search_callback  BLE observe complete. Num Resp 253

I (5061202) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (5061202) gattc: free heap: 61044
E (5069162) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
E (5121202) BT: btc_search_callback  BLE observe complete. Num Resp 253

I (5121202) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (5121202) gattc: free heap: 58760
E (5181202) BT: btc_search_callback  BLE observe complete. Num Resp 253

I (5181202) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (5181202) gattc: free heap: 55944
E (5241202) BT: btc_search_callback  BLE observe complete. Num Resp 253

/*
 *...
 * Look into attached file
 * ...
 */

I (6201202) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (6201202) gattc: free heap: 4496
E (6261202) BT: btc_search_callback  BLE observe complete. Num Resp 253

I (6261202) gap_cb: ESP_GAP_SEARCH_INQ_CMPL_EVT
I (6261202) gattc: free heap: 1476
E (6261202) BT:  bta_dm_ble_observe start observe failed. status=0x3

E (6261212) gattc: Scan start failed
W (7229192) wifi: alloc eb len=76 type=2 fail

W (7229192) wifi: m f probe req l=0

W (8853012) wifi: alloc eb len=76 type=2 fail

W (8853012) wifi: m f probe req l=0

/*
 * nothing more
 */
```

murraysst, 2017-05-26:
interesting - I too am trying to implement something like this, and find the BLE gap observe just eventually stops seeing anything.

FHFS, 2017-05-26:
Try turning CONFIG_SW_COEXIST_ENABLE off, and increase the scan interval in esp_ble_scan_params_t from the 0x50 from the examples to 0x100
`	esp_ble_scan_params_t _params =
	{
		.scan_type = BLE_SCAN_TYPE_ACTIVE,
		.own_addr_type = BLE_ADDR_TYPE_PUBLIC,
		.scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
		.scan_interval = 0x100,
		.scan_window = 0x30
	};`

mkonnov, 2017-05-29:
@FHFS Thanks for reply.
Turning CONFIG_SW_COEXIST_ENABLE hasn't helped, the picture is the same.
I think that problem is that I'm publishing MQTT messages too frequently.
I'll report here if I'll succeed.

FayeY, 2017-06-09:
Hi @mkonnov , how about this issue now? Is there any update?

mkonnov, 2017-06-09:
Hi @FayeY . 
Sadly not. I have bypassed this problem just by picking esp_restart() in bta_dm_ble_observe if status == BTM_NO_RESOURCES.

I will return to this issue in a nearest days so once I will fix it, I will share it here.
BTW, maybe it's not a memory leak at all, but it behaves as a leak: the issue makes itself felt after ~1h of operating.

TianHao-Espressif, 2017-06-12:
@mkonnov , could help make sure that is MQTT cause memory leak or Bluetooth or Wifi something cause memory leak? We are doing testing of excluding MQTT.

TianHao-Espressif, 2017-06-12:
@mkonnov , I try BLE scan and wifi connect AP without SW_COEXIST_ENABLE and MQTT, I cannot reproduce your problem. So could you try a test without MQTT, then we can make sure which module cause the memory leak.

murraysst, 2017-06-12:
@TianHao-Espressif could you post or pm the test code you are using? I am also trying to diagnose similar issues where either BLE scan stops seeing things and/or wifi disconnects and won't reconnect after an hour or so, but I am using mongoose-os (and their cloud build system) which complicates things. I would like a simple way to start with something that actually works (without mongoose, mqtt etc) and add back in things until it breaks.

TianHao-Espressif, 2017-06-13:
@murraysst. I try BLE scan and wifi connect AP without SW_COEXIST_ENABLE and MQTT. It have no memory leak , then I put AP into sheildbox, then ESP32 station disconnect and try to reconnect, there's no memory leak, too. The test code is in attachment.
[Archive.zip](https://github.com/espressif/esp-idf/files/1070969/Archive.zip)

mkonnov, 2017-06-22:
Hi guys ! My apologies for long silence.

After I have found that BTM_NO_RESOURCES can indicate the lack of free heap to allocate and at the same time I have estimated the decreasing of free heap, therefore I concluded that it was a memory leak without a hesitation.

Now after a few updates of esp-idf, I'm not estimating such free heap decrease anymore, but my latest investigations brought me to my mistake.
I'm making BLE advertising and scan consequently in one task and I have found that
`esp_ble_gap_start_advertising` together with `esp_ble_gap_start_scanning` 
is not starting advertising/scanning immediately due to callback-style API. We need to wait for event to be sure that scan/adv has started. Or not started.

This way, my error has been caused by the lack of synchronization between this two routines.
So I have used `xEventGroupSetBits` for appropriate events in `esp_gap_cb` callback and awaiting by  `xEventGroupWaitBits` this events in my app.
@TianHao-Espressif, @murraysst Thank you guys for your help.

I have tried the codes @TianHao-Espressif has shared and yes - there's no memory leak.

FayeY, 2017-06-23:
Is this problem solved?

mkonnov, 2017-06-25:
@FayeY The problem is not solved, but things starts to be clear.
I can say exactly: **it's not a memory leak**.
The problem has been reproduced by turning off WiFi connection (on WiFi router).
After WiFi has tried to reconnect, it tooks too much memory it has been available and this way it tooks all the memory available. What leads to inability to scan and do whatever what uses memory allocation.

This way, I have walked through possible places that takes too much heap and noticed that my MQTT task takes 16kb for stack (btw, it has bee token from example) and recuded it to 10kb. And it works fine. Also, I have put a suspension of BLE task when MQTT task notices the reconnection attempt and resume it only after network is reconnected. All errors is from IoT_Error_t.

But for now I see the following error in a loop:
`aws_iot: failed! mbedtls_net_connect returned -0x52`

2 modules is running simultaneously in the same environment and this error occured only on one of them.
So do anybody have some suggestions ?

mkonnov, 2017-06-26:
Also I often see the current fail:
`ASSERT_ERR(0), in ld_fm.c at line 185Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
`
Do smbd know what it related to and where to look for solution ?
I know, it's a wdt timeout, I know how to increase its period, but I need to know the reason of that fail.
I have made a coredump and it says nothing bad.

TianHao-Espressif, 2017-06-27:
@mkonnov , please check something:
1. do you use classic bt?
2. do you use dual core?
please update esp-idf to the newest.

mkonnov, 2017-06-27:
@TianHao-Espressif  thanks for your reply.
1. I use BT dual mode
2. As far as I know, this is controlled from checkbox "Run FreeRTOS only on first core". If so, then yes, I'm running dual core.

esp-idf is the newest.

TianHao-Espressif, 2017-06-27:
@mkonnov ，I'm very interesting at "often see the current fail".  So could you provide me your tests code?

mkonnov, 2017-06-27:
@TianHao-Espressif please test it with with a huge amount of BT advertisers, as more as you can.
[ble_mqtt_test.zip](https://github.com/espressif/esp-idf/files/1105895/ble_mqtt_test.zip)

Also hardcode your AWS certificates and WiFi credentials in aws_iot_mqtt_task.c. Sorry for crappy hardcodes, I tried to make it ASAP for you.
Thanks a lot.

mkonnov, 2017-06-27:
@TianHao-Espressif 
I have an updates. I have run this code with just a few advertisers and it fails.
You may be interested in the following crash report.
But I'm very interested in a reason of such fail, so please give me a direction to look for.
Pasting the crash report of the code I have attached above. Code hasn't been changed instead of my AWS key and WiFi credentials.

```
I (18472) gattc: REG_EVT                                                       
I (18472) gattc_cb: EVT 0, gattc if 4                                          
I (18472) gattc: REG_EVT                                                       
I (18482) BT: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT                          
ASSERT_ERR(bt_intcntl_finetgtintmsk_getf() == 1), in ea.c at line 497Guru Medit
ation Error of type LoadProhibited occurred on core  1. Exception was unhandled
.                                                                              
Register dump:                                                                 
PC      : 0x40086121  PS      : 0x00060033  A0      : 0x400879df  A1      : 0x3ffd6790                                                                        
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc4f38  A5      : 0x3ffc4f38                                                                        
A6      : 0x00000001  A7      : 0x00060b20  A8      : 0x00000017  A9      : 0x00000002                                                                        
A10     : 0x3ffc1398  A11     : 0x00000000  A12     : 0x3ffc510c  A13     : 0x3ffe31e8                                                                        
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c                                                                        
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000                                                                        
                                                                               
Backtrace: 0x40086121:0x3ffd6790 0x400879dc:0x3ffd67b0 0x40087996:0x3ffd691c   
                                                                               
Guru Meditation Error o

```
Despite I have shared my source code, here's a link to actual elf for which this crash has take place:
https://drive.google.com/open?id=0B5Q2K4xNERwCZkUtY2N1Ung4a2M

mkonnov, 2017-06-28:
@TianHao-Espressif Hi there !
Have you tried the code I sent ? Is the problem I described reproducible ?
Thanks.

TianHao-Espressif, 2017-06-29:
@mkonnov , I tried your code. But there's some logs

> .[0;32mI (10502) gattc: REG_EVT.[0m
> .[0;32mI (10512) BT: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT.[0m
> I (12132) wifi: pm start, type:0
> 
> .[0;31mE (12602) mqtt_task: rc = -13.[0m
> .[0;31mE (12602) mqtt_task: An error occurred in the main loop (rc = -13), rebooting....[0m
> I (12602) wifi: state: run -> init (0)
> I (12602) wifi: pm stop, total sleep time: 0/473683

mkonnov, 2017-06-29:
@TianHao-Espressif  Sorry, I assumed you have some.
Can I send you the certificate set via the email to you will test it out ?

TianHao-Espressif, 2017-06-29:
@mkonnov , sure. Email is in my profile. You can click my head photo to see it.

closed 2017-10-10 by FayeY