espressif/esp-idf #619: BLE sniffer example

ahmedjouirou, 2017-05-18:
Im trying to make a simple ble sniffer (continuous scan no specific duration), i tried both, using HCI command and  GAP interface , both worked for a 5 to 6 minutes .. then BLE stopped receiving (but cpu is running correctly) . I m in a very crowded BLE enivrement (around 600 ble packets/second from multiple devices and i need to listen to them all at this rate and all time) , is there any suggestions ...
for HCI direct commands ( i scan at 0x50/0x30  window/interval ) i couldnt debug further the esp_vhci_host_register_callback(&vhci_host_cb); (the callback called when i got a BLE message) because its part of the closed binary blob of BLE .

murraysst, 2017-05-26:
ok, I'm finding the same sort of thing - I'm running back to back gap observe operations, but eventually it stops seeing BLE devices.

Yulong-espressif, 2017-06-11:
@ahmedjouirou We have fixed this bug in the new version of SDK, you can get it from the github. Thanks. Any question, please let me know.

murraysst, 2017-06-12:
Hi @ahmedjouirou can you please post or pm your code to test this bug fix against latest sdk? thanks

Yulong-espressif, 2017-09-27:
git commit id of my SDK is a3a4a20518db956108c94ad6cbfbc6079313d88c, I have try it with used the gatt_client demo to test the scanning issue, but it works find in my side, so can you provide me more information about it? Thanks.

Yulong-espressif, 2017-09-27:
@murraysst

closed 2017-10-10 by FayeY