espressif/esp-idf #621: Error in make command.

harshcht, 2017-05-18:
I followed the steps as given. I tried to compile the hello_world example . I went till  "make menuconfig" command.  But when I exit, It gives the following warning:

/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
MENUCONFIG.

Can anyone help me find what I did wrong?

krzychb, 2017-05-18:
Hi again :)

The warning indicates an issue with toolchain installation.

1. Does the issue concern Linux (ubuntu) as in #609?

2. Have you installed the toolchain according to  [Standard Setup of Toolchain](http://esp-idf.readthedocs.io/en/latest/get-started/index.html#standard-setup-of-toolchain) or [Customized Setup of Toolchain](http://esp-idf.readthedocs.io/en/latest/get-started/index.html#customized-setup-of-toolchain)?

3. If you installed the toolchain and the issue persists, then please enter in terminal and send result of:
```
$ printenc PATH
```

harshcht, 2017-05-19:
I installed using the standard setup of tool chain : 

[http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html#standard-setup-of-toolchain-for-linux](url)

the printenv PATH commands gives the following result:
/home/pankaj/bin:/home/pankaj/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

However the variable updated in the process was IDF_PATH
the command  "printenv IDF_PATH" return:
/home/pankaj/esp/esp-idf

krzychb, 2017-05-19:
You need to update `PATH` as well. 

From what I see above it does not contain the path to the toolchain.

Please check once more the configuration step under [Toolchain Setup](http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html#toolchain-setup)  that starts here:

[image]

Here is how my `PATH` looks like:

```
$ printenv PATH
/home/krzysztof/bin:/home/krzysztof/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/krzysztof/esp/xtensa-esp32-elf/bin
```
Comparing to your `PATH` it has an extra `:/home/krzysztof/esp/xtensa-esp32-elf/bin` added at the end.

In your case, if you install in standard `~/esp` location, it would be `:/home/pankaj/esp/xtensa-esp32-elf/bin`

harshcht, 2017-05-19:
@krzychb Thanks again for the help :) . It compiles and uploads now.

closed 2017-05-22 by igrr

ahmedawais, 2018-06-03:
It's not working for me please help 
Eclipse ide  
Ubuntu 
Esp32 
running flash on terminal 

awais@ubuntu:~/Downloads/espressifmaster-all/espressif-all/espressif-master/esp32/app/ESP32_ble_UART$ printenv IDF_PATH
/home/awais/esp32/esp-idf
awais@ubuntu:~/Downloads/espressifmaster-all/espressif-all/espressif-master/esp32/app/ESP32_ble_UART$ export PATH="$PATH:$HOME/esp/xtensa-esp32-elf/bin"
awais@ubuntu:~/Downloads/espressifmaster-all/espressif-all/espressif-master/esp32/app/ESP32_ble_UART$ make flash/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Flashing binaries to serial port ttyS0 (app at offset 0x10000)...
esptool.py v2.1
Traceback (most recent call last):
  File "/home/awais/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 2524, in <module>
    _main()
  File "/home/awais/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 2517, in _main
    main()
  File "/home/awais/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 2246, in main
    esp = chip_class(args.port, initial_baud)
  File "/home/awais/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 177, in __init__
    self._port = serial.serial_for_url(port)
  File "/usr/lib/python2.7/dist-packages/serial/__init__.py", line 85, in serial_for_url
    instance.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 294, in open
    raise SerialException(msg.errno, "could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port ttyS0: [Errno 2] No such file or directory: 'ttyS0'
/home/awais/esp32/esp-idf/components/esptool_py/Makefile.projbuild:54: recipe for target 'flash' failed
make: *** [flash] Error 1
awais@ubuntu:~/Downloads/espressifmaster-all/espressif-all/espressif-master/esp32/app/ESP32_ble_UART$ printenv IDF_PATH/home/awais/esp32/esp-idf
awais@ubuntu:~/Downloads/espressifmaster-all/espressif-all/espressif-master/esp32/app/ESP32_ble_UART$ printenv PATH

Please help

leandrosilva2151197, 2018-06-23:
This is closed topic but... I had same problem and you can try use this commands in case you using root user or maybe because $HOME is not set as it supose too: 
root@projeto:cd ~
root@projeto:pwd
and it should point to the main path where "esp" folder is to esp/esp-idf/... and esp/xtensa-esp32-elf/... is. If not... use nano or any other text editor /root/.bashrc or /home/<user>/.bashrc and make sure this vars:

export $HOME=/root
export $PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
export $ESP_IDF=~/esp/esp-idf

.bashrc should exist already you only have to append those comands in the end. Still not working?
export $HOME=/home/<user>
<user> is your username, you have to fix $HOME to point to the main path of /esp/esp-idf/... and  /esp/xtensa-esp32-elf/... 

if root:
your "printenv PATH":
/usr/local/sbin:/usr/local/bin:/usr/bin/... neet to have this ... :**/root/esp/xtensa-esp32-elf/bin**
your "printenv IDF_PATH":
/root/esp/esp-idf

if regular user:
our "printenv PATH":
/usr/local/sbin:/usr/local/bin:/usr/bin/... neet to have this ... :**/home/<user>/esp/xtensa-esp32-elf/bin**
your "printenv IDF_PATH":
/home/<user>/esp/esp-idf

<user> as your username.

Good luck.

sravanth005, 2019-03-05:
i try to run ESP32 Cam i came up till menu configuration but i could not flash the unit i fall up with error like:

admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
make: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
CC build/bootloader/bootloader_support/src/bootloader_flash.o
make[2]: xtensa-esp32-elf-gcc: Command not found
/home/admin-unizen/esp32/esp-idf/make/component_wrapper.mk:289: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 127
/home/admin-unizen/esp32/esp-idf/make/project.mk:551: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/admin-unizen/esp32/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/admin-unizen/esp-who/examples/single_chip/camera_web_server/build/bootloader/bootloader.bin' failed
make: *** [/home/admin-unizen/esp-who/examples/single_chip/camera_web_server/build/bootloader/bootloader.bin] Error 2

so please help me out how to solve this error.

sravanth005, 2019-03-07:
The following Python requirements are not satisfied:
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/Desktop/iotbits_esp32/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/Desktop/iotbits_esp32/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
make: *** [check_python_dependencies] Error 1

how to clear this error can any one suggest me the commands for Ubuntu 16.04.
i tried a lot cant make it out so please someone send the commands to correct that error.

krzychb, 2019-03-07:
@sravanth005 how `The following Python requirements are not satisfied:` is related to this issue report?

sravanth005, 2019-03-20:
Hi all,
I am trying to work with ESP32-CAM (https://robotzero.one/esp32-camera-module/#comment-3073) i can able to take a still with the camera but face detection and face reorganization is not happening, Do any one know for what reason it is not happening.please help me out to solve this issue.

dhcps: send_offer>>udp_sendto result 0
I (63186) camera_httpd: JPG: 6335B 1573ms
I (66456) camera_httpd: vflip = 1
dhcps: send_offer>>udp_sendto result 0
I (72716) camera_httpd: JPG: 7267B 118ms
I (78976) camera_httpd: face_detect = 1
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (1752586) wifi: max connection, deauth!
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (1873156) wifi: max connection, deauth!
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (2301736) wifi: station: b4:ef:fa:79:e6:e2 leave, AID = 1, bss_flags is 134243
I (2301736) wifi: new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:6
I (2301736) camera wifi: station:b4:ef:fa:79:e6:e2leave, AID=1
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0

thanks in advance

sravanth005, 2019-03-20:
hai @krzychb 
Even if i install cryptography 2.7.1 i fall up with same error only.

oliverjc, 2019-12-22:
I'm using Eclipse IDE and had to edit my path variable for my project.
In a terminal window type `echo $PATH`.
Copy that output exactly into the value for the PATH variable in Project Properties --> C/C++ Build --> Environment

ahrenswett, 2020-09-03:
having this issue but slightly different. specifically this part looks curious as my bin file does not have sh directory but does contain the  xtensa-esp32-elf-gcc:
it like its looking in the wrong place. thanks in advance!

`/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found`

```
nina-fw[master !?]$ printenv IDF_PATH
/mnt/c/Users/ahren/esp/esp-idf/
nina-fw[master !?]$ printenv PATH
$/mnt/c/Users/ahren/esp/xtensa-esp32-elf/bin/:$/mnt/c/Users/ahren/esp/xtensa-esp32-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/c/Program Files/AdoptOpenJDK/jdk-11.0.4.11-hotspot/bin:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files/Microsoft SQL Server/130/Tools/Binn/:/mnt/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn/:/mnt/c/Program Files/Git/cmd:/mnt/c/Gradle/gradle-5.6.2/bin:/mnt/c/Program Files/Calibre2/:/mnt/c/Program Files (x86)/IDM Computer Solutions/UltraEdit:/mnt/c/Users/ahren/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/ahren/AppData/Local/Programs/Microsoft VS Code/bin:/mnt/c/Users/ahren/AppData/Local/GitHubDesktop/bin:/mnt/c/Users/ahren/AppData/Local/Microsoft/WindowsApps:/snap/bin
nina-fw[master !?]$ make
make: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Toolchain path:
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-esp32-elf-gcc: Command not found
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/bootloader_support/src/bootloader_flash.o
make[2]: xtensa-esp32-elf-gcc: Command not found
/mnt/c/Users/ahren/esp/esp-idf/make/component_wrapper.mk:289: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 127
/mnt/c/Users/ahren/esp/esp-idf/make/project.mk:552: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/mnt/c/Users/ahren/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/mnt/c/Users/ahren/nina-fw/build/bootloader/bootloader.bin' failed
make: *** [/mnt/c/Users/ahren/nina-fw/build/bootloader/bootloader.bin] Error 2
```

siddyboy11, 2021-02-21:
@ahrenswett  I have been facing same issue
Can u help me resolve