espressif/esp-idf #623: API_vhci_host_register_callback problem

ahmedjouirou, 2017-05-18:
I got the BLE running for scanning ADV packets , after 5 minutes or so  it stops recieving (im using HCI interface) is there any way to get the sources of "API_vhci_host_register_callback" because it seem that i cannot dig more in my debugging due to closed binary blobs . thanks

mywang-espressif, 2017-06-07:
@ahmedjouirou , thanks for your information, but I am sorry that the sources for vhci are not open, and we only provide the library file. I will try to reproduce the issue myself first.

mywang-espressif, 2017-06-08:
@ahmedjouirou , I made up a simple example to run BLE scan, but did not find it stops receiving the LE advertising report event for more than half an hour or so when I ended the test.
Can you give me some details about your test scenario, e.g. which HCI commands are used,  and the parameters of HCI_LE_Set_Scan_Parameters and HCI_LE_Set_Scan_Enable?

FayeY, 2017-06-19:
Hi @ahmedjouirou , is there any update?

closed 2017-07-05 by FayeY