espressif/esp-idf #624: example aws_iot - subscribe_publish crash if using  xTaskCreatePinnedToCore

WilliamFrasson, 2017-05-18:
using the AWS_IoT example "subscribe_publish"
for creating  the aws_iot_task task, it is used xTaskCreatePinnedToCore

using my DevKitC  it crash when application call the xTaskCreatePinnedToCore
while using 
 xTaskCreate(&aws_iot_task, "aws_iot_task", stack_size, NULL, 5, xHandle);

the fw works.

why xTaskCreatePinnedToCore create this problem?

FayeY, 2017-11-21:
Since this is a very old issue, I'm going to close it. Please feel free to reopen it, if your problem is still unsolved on the latest IDF.

ustccw, 2017-11-22:
@WilliamFrasson 
plz try the latest code, we have used the `xTaskCreatePinnedToCore` at the beginning of the `app_main()`

but you need to replace the total `app_main()` with the following code:
```
void app_main()
{
    // Initialize NVS.
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // app partition table has a smaller NVS partition size than the non-OTA
        // partition table. This size mismatch may cause NVS initialization to fail.
        // If this happens, we erase NVS partition and initialize NVS again.
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    initialise_wifi();
#ifdef CONFIG_MBEDTLS_DEBUG
    const size_t stack_size = 36*1024;
#else
    const size_t stack_size = 36*1024;
#endif
    xTaskCreatePinnedToCore(&aws_iot_task, "aws_iot_task", stack_size, NULL, 5, NULL, 1);
}
```

commit 4b3997cf (2017-11-23): aws_iot: fix aws_iot examples without NVS initialization

WiFi used to initialise nvs automatically, but it doesn't now.
see more info: https://github.com/espressif/esp-idf/issues/624

Closes #624
closed 2017-11-22 by FayeY