espressif/esp-idf #626: Feature request: simple BT to UART bridge

xarrange, 2017-05-20:
Hi,
I have some experience with ESP8266 and HC-05. And I was kind of hoping to meet here any analog to this functionality. I have no idea what is HCI, Gatt, BT Classic, Bluedroid, etc.. which are used in examples.
What will be really helpful (I guess for lots of ESP32 users) is to have some "getting started" example which will contain something like Arduino's software serial :
`_if (uart0.available()) {
    uart1.write(uart0.read());
  }

  if (uart1.available()) {
    uart0.write(uart1.read());
  }_`
where uart0 is connected to PC through FTDI(usb to uart) and uart1 is connected to PC (or phone) via BT.

igrr, 2017-05-20:
There is no support for SPP at the moment (which is what HC-05 uses), so this may take some time to implement. Suggest reading a bit about GATT in the meantime...

xarrange, 2017-05-20:
@igrr, thanks for clear answer. Does anybody have planse to implement it?
Sure I'll get falimiar with GATT. just wanted to start with something simple

igrr, 2017-05-20:
We do plan to support SPP in ESP-IDF, but it's currently low priority. So maybe a couple of releases down the road...

PhilColbert, 2017-05-21:
Need this for my current project aswell, sooner please ? ;)

Yulong-espressif, 2017-06-03:
@PhilColbert We will release used uart command to control BLE in the next version(2.1) of SDK, thanks.

closed 2017-10-23 by FayeY

jesuisdiana, 2018-05-09:
I have a question. Using GATT let me connect an hc05 bluetooth module with BLE like in esp32?