espressif/esp-idf #629: Feature Request: BT PAN Over Ethernet

GeorgeIoak, 2017-05-20:
This might not be the most common usage but I was hoping if I posted the request I could see if anyone had worked on it. What this would do is allow a device (say a tablet) to connect over Bluetooth to the ESP32 and then send packets over Bluetooth to the internet through a wired ethernet connection.
This is useful for small data and lowest power since Bluetooth communication would be lower power than WiFi

negativekelvin, 2017-05-21:
Might be a start: https://github.com/kuc-arc-f?tab=repositories

GeorgeIoak, 2017-05-21:
It definitely looks like it's along the lines of what I'm after. I'll brush up on my Japanese and see what he's got done.

benjaminaigner, 2017-05-22:
Hi,

I'm currently working with 6lowpan over BLE, so I'm already sending MQTT
messages via Bluetooth to a host (no Ethernet by now, but this should be
no problem).

Are you interested?

On 20/05/17 21:27, George Ioakimedes wrote:
> This might not be the most common usage but I was hoping if I posted the request I could see if anyone had worked on it. What this would do is allow a device (say a tablet) to connect over Bluetooth [...]

GeorgeIoak, 2017-05-22:
Yes, I'm very interested in getting this working. Should we start a repo or how would you want to work together?

mringwal, 2017-05-22:
Another option you might try is to run [BTstack](http://btstack.org) instead of the Bluedroid stack on the ESP32 and then port the [example/panu_demo.c](https://github.com/bluekitchen/btstack/blob/master/example/panu_demo.c) to the IP stack of ESP-IDF. This would use Bluetooth Classic - more speed, but also requires more power than 6LowPAN over BLE.

GeorgeIoak, 2017-05-22:
@mringwal Thanks that looks interesting but I didn't see the ESP32 as one of the supported processors on the BTstack site.

benjaminaigner, 2017-05-22:
Hi,
sry for this late reply.
I will upload it to my account ASAP (comments, ...).
My implementation uses BTStack (thx @mringwal for that!), it depends on you if you want to spend the effort on implementing/using Bluedroid

I will post the link when I'm finished!
I'm looking forward for feedback, improvements and wishes ^^

GeorgeIoak, 2017-05-22:
@benjaminaigner Thanks! I look forward to checking it out. I've got a ENC28J60 breakout board and I just ordered up some more ESP32 modules to mess with.

mringwal, 2017-05-23:
@GeorgeIoak it's not on the main README.md yet, but check out ports/esp32.
I have been waiting for a bug in the ESP32 low level Bluetooth implementation to get fix and still got crashes I'd like to analyse and post issues about, before I add it to the front page. It's working well for @benjaminaigner it seems.

benjaminaigner, 2017-05-23:
@mringwal: I have no problems, even with an "old" version of BTStack!
I sent more than 5000 MQTT-SN packets (the MQTT-Spy stops counting after 5000 :-)), no problems. 

I've posted the network interface for RFC7668 at savannah (lwIP git), you need to put this driver in components/lwip/netif:

http://savannah.nongnu.org/bugs/?51094

I will post the implementation with BTStack via the university's GIT server soon!

benjaminaigner, 2017-05-23:
Hi again, you can clone my current implementation here:

https://es.technikum-wien.at/aignerb/esp32-rfc7668-public

Best wishes

BTW: Issue tracking is disabled, so either we discuss problems here or via another channel

GeorgeIoak, 2017-05-23:
I need to setup a new machine with the ESP-IDF so I'll work on that today and clone your repo.

GeorgeIoak, 2017-05-24:
I got my Windows machine setup finally and I cloned your repo into my Github account. I tried to compile the project but it looks like your repo is missing btstack_event.h:

benjaminaigner, 2017-05-24:
I'm not sure if I'm allowed to include BTStack within my repo. So I cloned the official BTStack repo from bluekitchen and put that to a folder in my projects directory. Please take care of modifying the components.mk file in my repo to reflect the **relative** path to the BTStack root.

mringwal, 2017-05-25:
@benjaminaigner hi. just had a first look. It's surprisingly little code, congrats! I'll send more comments by mail. anyway. I don't see "rfc7668.c and rfc7668.h/rfc7668_opts.h" in the repo. Where to get these, or are they just missing?

GeorgeIoak, 2017-05-25:
Is it perhaps forked from here, [here](http://savannah.nongnu.org/bugs/?51094)

benjaminaigner, 2017-05-26:
@GeorgeIoak  other way, I posted it there :-)

You can use these files from the link @GeorgeIoak  (and I) posted. 

I'm currently cleaning up the code, to have it in the master branch of lwIP.

mringwal, 2017-05-26:
@benjaminaigner ouch.. right! You could add the savannah link to the gitlab readme (and update it once there's progress.

benjaminaigner, 2017-05-29:
Hi again, I've updated my repo with the example code. The current lwIP files are now included (as they are currently provided to lwip as patch).

So:

- Clone my repository ( https://es.technikum-wien.at/aignerb/esp32-rfc7668-public/ )
- Clone BTStack ( https://github.com/bluekitchen/btstack )
- Follow the instructions in my README

Greetings

YelongHuang, 2017-06-01:
@GeorgeIoak I think you can try to transplant the contiki(a stack of 6lowpan, the github link is https://github.com/contiki-os/contiki) to the IDF, mabey it will helpful for you. Thanks.

closed 2017-07-05 by FayeY

nottledim, 2017-12-24:
I'm a bit late to the party but I'm trying to build the Benjamin Aigner test/demo https://es.technikum-wien.at/aignerb/esp32-rfc7668-public

And I get this compile error: No rule to make target '/home/beni/sync/Projects/FH/esp/esp-idf/components/esp32/./brownout.c', needed by 'brownout.o

I can't find where that path originates in the repo.  Can somebody help!

benjaminaigner, 2017-12-28:
@nottledim 
I'm currently not working on this proof of concept, simply because I'm waiting for BLE Mesh.
But this build error is mostly related to a absolute path I wrote somewhere (this is my personal dir tree).
Please do a "make clean" and "make" again & if it does not work, please post the resulting log here.
BTW: I cannot guarantee this is working with the latest esp-idf!

nottledim, 2017-12-28:
Thanks for responding.  I've given up with bluetooth for now. The learning curve was too steep to justify and btstack doesn't support PAN on esp yet.

EvanKrall, 2019-06-13:
I took a crack at getting PAN on ESP32 via btstack working:

https://github.com/EvanKrall/panu_demo_esp32

The core of it is https://github.com/EvanKrall/panu_demo_esp32/blob/efbc0a2b1787ffdc9d9f07b95d7c03d3a411f31e/main/btstack_network_lwip.c, which implements the same functions as e.g. https://github.com/bluekitchen/btstack/blob/789c3c6f9c87f37dfab2df5d9ed2c74951d065fd/platform/posix/btstack_network_posix.c -- those defined in https://github.com/bluekitchen/btstack/blob/789c3c6f9c87f37dfab2df5d9ed2c74951d065fd/src/btstack_network.h

I'm not aiming to forward to Ethernet like you are, just to use PAN like softAP.

This repo also sets up a dhcp server and the example http server.

It compiles, runs without crashing, and I can pair to it with iOS and macOS, but I am unable to access the web server or get assigned an address by DHCP. I'm sure I'm just missing something silly.

mringwal, 2019-06-13:
Hi Evan! Thanks for sharing. I've also spend some time on providing an HTTP server over BTstack's PAN using lwIP for all all platforms. I guess there could have been some synergies.

Anyway, I've started with plain BTstack and added lwIP in the NO_SYS = 1 bare metal version (to have something that works anywhere). It also adds a DHCP server and runs on desktop system (mac/linux/win) and some NOOS platforms.

After pondering for a while how we could support multiple PAN demos, but at the same time trying to hide the fact that we're using lwIP, I gave up, and pulled lwIP into the examples.
However, the BNEP packet handling is always the same for a given TCP/IP stack, so I've created a bnep_lwip.c adapter that is used instead of the btstack_network.h interface. The PAN HTTP + DHCP demo is here: https://github.com/bluekitchen/btstack/blob/develop/example/pan_lwip_http_server.c

Now, back to ESP32. I've also tried to integrate it with esp-idf, but it's not working yet - there seems to be bug that causes a crash on the first HTTP request. In fact, I wanted to look at it later today and try to find the bug. If you beat me to it, please let me know, it's on lwip-esp32 branch with the bnep_lwip.c adapter in platform/lwip.c

You can check it out

mringwal, 2019-06-13:
Found it! Well, it was a misconfiguration of the ESP32 port that was only relevant to PAN/BNEP.
Please try pan_lwip_http_server from current develop. Works for me! :)

RandyLoeb, 2019-08-13:
It seems like the discussion here is providing internet through the esp32 via bluetooth, I'm trying to pursue the reverse direction: esp32 uses bluetooth to get internet via some other device (phone, laptop) that bridges (I think this sometimes called tethering over bt). But I think there's common interest in getting a tcp/ip stack on top of bt? Curious if someone knows of sample code for my use case? Thanks

mringwal, 2019-08-14:
@RandyLoeb For Bluetooth Tethering, the UI to pick a smartphone can be a bit tricky. On the other hand, the smartphone could connect to the ESP32 which then could remember the address of the smartphone. 

That aside, the panu_demo in BTstack shows how to create an outgoing connection or might already do what you're looking for.

nicolasgarnet, 2020-12-08:
@RandyLoeb Have you found any code sample that provides internet access to ESP32 through phone via bluetooth tethering ? I'm also looking to do something like this.

SalimTerryLi, 2021-08-16:
I have made BT tethering works on ESP32 now, by adding TAP interface in esp_netif and manually wiring input and output to and from bnep. I'm going to try to make it available in IDF. At the current point DHCP client and ARP works fine when I connect esp32 to my Linux Mint20.2 and Android10(lineageos) with bluetooth.

FYI @nicolasgarnet @RandyLoeb