espressif/esp-idf #632: components starting with src/ don't work

MrBuddyCasino, 2017-05-21:
I added BearSSL as a git submodule, and the folders look like this:

/<my_project>/components/bear_ssl/src/codec
/<my_project>/components/bear_ssl/src/ec
etc.

This leads to errors: `fatal error: opening dependency file src/codec/ccopy.d: No such file or directory`

Making a symlink called bearsrc to src and using it in my COMPONENT_SRCDIRS path fixes the problem.

projectgus, 2017-05-22:
I wasn't able to reproduce this with a simple example, however I've seen errors like this sometimes when doing a partial build, if the structure of the source tree has changed from the last time "make" ran.

If you make "make clean" and then "make", does the 'src' directory work then?

If the answer is "no", could you please post the full component.mk and directory structure so we can try to reproduce?

igrr, 2017-05-22:
Could you try modifying the order of source directories so that src is in front of its inner directories? I.e. `COMPONENT_SRCDIRS = src/codec src/ec src`. I seem to remember something similar in our LwIP component.

projectgus, 2017-05-22:
I'm able to reproduce the bug Ivan mentions when using make 3.81 (this is the default make version on OS X, but most other systems have make 4.x). Looking into a fix now.

projectgus, 2017-05-22:
We're testing a fix for this internally now, but in the mean time reordering the COMPONENT_SRC_DIRS as suggested should work around the error.

MrBuddyCasino, 2017-05-22:
Indeed it does - thx!

FayeY, 2017-07-07:
Is this problem solved now? Can we close this issue?

igrr, 2017-07-23:
Fixed in 99771a2.

closed 2017-07-23 by igrr