espressif/esp-idf #633: When is the ESP-MESH feature available?

binsys, 2017-05-22:
ESP8266 has this feature。
I think ESP32 should not remove this feature.
Here is offical overview:
http://www.espressif.com/en/products/software/esp-mesh/overview

FayeY, 2017-05-26:
We are working on ESP32 mesh now. It will be added in the future release. 
Thanks for your interest in ESP32!

binsys, 2017-05-26:
Thank you for your reply.

closed 2017-05-26 by binsys

lixianyu, 2017-06-24:
Any news on this ? :)

Karusa, 2017-07-01:
How will this up updated on already sold chips?

Eric24, 2017-07-24:
I'd be interested to know if, when released for the ESP32, ESP-MESH will be compatible with the "official" [BLE mesh standard](https://www.bluetooth.com/what-is-bluetooth-technology/how-it-works/le-mesh).

binsys, 2017-07-25:
@Eric24  ESP-MESH looks based on WIFI.

jo-nas, 2017-08-04:
Is there any news on this topic?

micheljung, 2017-08-08:
For bluetooth mesh, see #819

micheljung, 2017-08-08:
@FayeY is there an issue that tracks the status of ESP-MESH? I'm not sure what "the future release" exactly means :)

No-0n3, 2018-01-09:
Any news of when it is going to be implemented?

FayeY, 2018-02-27:
Hi, thanks for your interest in the ESP32 and mesh, the ESP-MESH will be released with the IDF V3.1. Thanks.

yorickr, 2018-03-05:
When will V3.1 be released?