espressif/esp-idf #634: unhandleed error when ringbuffer is full

mybays, 2017-05-22:
example for :
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt_nec_tx_rx

log as follow:
Rb size 1000 free 47 rptr 16 freeptr 16 wptr 968
E (495212) rmt: RMT RX BUFFER FULL
Rb size 1000 free 47 rptr 16 freeptr 16 wptr 968
E (495242) rmt: RMT RX BUFFER FULL
Rb size 1000 free 47 rptr 16 freeptr 16 wptr 968
E (496072) rmt: RMT RX BUFFER FULL
Rb size 1000 free 47 rptr 16 freeptr 16 wptr 968
E (496482) rmt: RMT RX BUFFER FULL
Rb size 1000 free 31 rptr 16 freeptr 16 wptr 984
E (496862) rmt: RMT RX BUFFER FULL
Rb size 1000 free 15 rptr 16 freeptr 16 wptr 0
E (496912) rmt: RMT RX BUFFER FULL

Spritetm, 2017-05-23:
How specifically did you get this error? What did you expect that would happen?

mybays, 2017-05-23:
I use an IR remote controller send random ir signal,this error occur very soon.

chevdor, 2017-07-10:
I am running into the same issue with the following code, inspired by the NEC sample:
```
static void pwm_read_task()
{
    int channel = 0;
    pwm_rx_init();
    RingbufHandle_t rb = NULL;
    // Get RMT RX ringbuffer
    rmt_get_ringbuf_handler(channel, &rb);
    rmt_rx_start(channel, 1);
    while(rb) {
        size_t rx_size = 0;
        // Try to receive data from ringbuffer.
        // RMT driver will push all the data it receives to its ringbuffer.
        // We just need to parse the value and return the spaces of ringbuffer.
        rmt_item32_t* item = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, 1000);
        if(item) {
            int duration;
            int res = pwm_parse_items(item, &duration);
            if (res) ESP_LOGI(NEC_TAG, "%d µs", duration);
            //after parsing the data, return spaces to ringbuffer.
            vRingbufferReturnItem(rb, (void*)item);
        } else {
            break;
        }
    }
    vTaskDelete(NULL);
}
```

As long as I apply a PWM signal on my input, all works fine and it can run for a while.
The issue occurs as soon as I stop applying a PWM signal and reapply it after ~9 - 10s.

negativekelvin, 2017-07-11:
@chevdor well if your ringbuffer is empty and you break out of your while loop then nothing will read items from the ringbuffer.

chevdor, 2017-07-11:
@negativekelvin Put it like this, it totally makes sense 😄 
That was indeed the issue on my end. Thanks for your comment! 
Issue is solved AFAIK.

Spritetm, 2017-07-11:
Good to hear :)
@mybays : do you still have this issue? If so, can you post the code that triggers it? Or is it the default unmodified example?

FayeY, 2017-07-24:
Hi @mybays , any update about this issue?

closed 2017-08-07 by FayeY

raziAkbar, 2018-08-07:
@chevdor can u guide me with your IR remote?

KIMSUNIK, 2019-12-31:
You need to change like below.
this code does not stop into the while() loop.

static void pwm_read_task()
{
    int channel = 0;
    pwm_rx_init();
    RingbufHandle_t rb = NULL;
    // Get RMT RX ringbuffer
    rmt_get_ringbuf_handler(channel, &rb);
    rmt_rx_start(channel, 1);
    while(rb) {
        size_t rx_size = 0;
        // Try to receive data from ringbuffer.
        // RMT driver will push all the data it receives to its ringbuffer.
        // We just need to parse the value and return the spaces of ringbuffer.
        rmt_item32_t* item = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, 1000);
        if(item) {
            int duration;
            int res = pwm_parse_items(item, &duration);
            if (res) ESP_LOGI(NEC_TAG, "%d µs", duration);
            //after parsing the data, return spaces to ringbuffer.
            vRingbufferReturnItem(rb, (void*)item);
        } 
    }
    vTaskDelete(NULL);
}