espressif/esp-idf #635: Use PCNT with GPIO36 and GPIO39

SirRibeiro, 2017-05-22:
Hi,
I want to use the pulse counter module (pntc) example to make 8 counters on esp32.
The GPIOS are
GPIO 36
GPIO 39
GPIO 34
GPIO 35
GPIO 32
GPIO 33
GPIO 25
GPIO 26

Separately i can use all except for GPIO 36 and GPIO 39? Is there a special reason for that or i'am just doing something wrong?

Is it possible to work with all at the same time?
Can anyone give me a hint how to do that?

Thanks!

Spritetm, 2017-05-23:
GPIO36 and 39 are indeed 'special' because they're input-only; every GPIO in the range 34-39 is. See [the pin list](http://www.espressif.com/sites/default/files/documentation/esp32_chip_pin_list_en.pdf) for more information.

negativekelvin, 2017-05-23:
@Spritetm true, but pcnt is an input based module.  however they are also special because they are used for the differential LNA and connected to capacitors.

Spritetm, 2017-05-23:
Ah, derp, I thought I read RMT instead :X That's what you get when you answer Github questions before drinking coffee. Not sure what's wrong with those GPIOs then, they should just work; maybe it's a bug in the GPIO software logic or something.

negativekelvin, 2017-05-23:
Well might need rtc_gpio_deinit (but i believe 32, 33 do too) or some other amp config so the caps don't filter out the pulses?

negativekelvin, 2017-05-23:
Pcnt example works for me on 36 and 39

SirRibeiro, 2017-05-23:
@negativekelvin can you please show me your code to see what am I doing wrong? 
And can you use it all together? 
Thanks for the responses guys!

SirRibeiro, 2017-05-23:
@negativekelvin and @Spritetm I don't Know why but it is working well now, must be the lack of Coffee from Last night as well. Thank you guys and sorry for the trouble 
I Just have to find out how to make it work all together.

FayeY, 2017-05-31:
Hi @SirRibeiro , is this problem solved? Can we close this issue?

SirRibeiro, 2017-06-03:
Hello @FayeY of course! Sorry for the delay in the response.

closed 2017-06-04 by FayeY