espressif/esp-idf #636: Cant find Example Configuration in make menuconfig

llewellynnu, 2017-05-23:
As mentioned in the Readme.md to configure the aws endpoint 

"Run make menuconfig, navigate to "Example Configuration" and change "AWS IoT Certificate Source" to "Load from SD card"."

Cant find the Example configuration menu .

igrr, 2017-05-23:
Which example does this happen with, `subscribe_publish` or `thing_shadow`?

Could you check if there is a Kconfig.projbuild file in the main/ directory of the example project?

llewellynnu, 2017-05-23:
Hi 
This is with a custom project created by me . 
I see that Kconfig.projbuild for my project and the subscribe_publish example is different .

How do i add those fields to the file ?

igrr, 2017-05-23:
If you are developing based on a different project and you want to merge Kconfig files, read about [Kconfig format](https://www.kernel.org/doc/Documentation/kbuild/kconfig-language.txt) first. Kconfig files are plain text, with blocks of text describing each option. So in general, merging two Kconfig files should be straightforward: just copy the blocks describing the options you need from the example into your file.

Then take note however that to make the added options useful, you need to have the same/similar code in your application:

https://github.com/espressif/esp-idf/blob/02fdf8271de3a6db2ab9d4d6f023c160c84ebdbe/examples/protocols/aws_iot/subscribe_publish/main/component.mk#L5-L22

https://github.com/espressif/esp-idf/blob/02fdf8271de3a6db2ab9d4d6f023c160c84ebdbe/examples/protocols/aws_iot/subscribe_publish/main/subscribe_publish_sample.c#L79-L96

https://github.com/espressif/esp-idf/blob/02fdf8271de3a6db2ab9d4d6f023c160c84ebdbe/examples/protocols/aws_iot/subscribe_publish/main/subscribe_publish_sample.c#L177-L208

llewellynnu, 2017-05-23:
Yes i have done that . But where is the code to embed the certificates in the binary ?
and do i have to mantain the variable names for embeded certificates?  
"extern const uint8_t aws_root_ca_pem_start[] asm("_binary_aws_root_ca_pem_start");"

igrr, 2017-05-23:
That is in the first of the three links at the end of my previous comment (component.mk):

https://github.com/espressif/esp-idf/blob/02fdf8271de3a6db2ab9d4d6f023c160c84ebdbe/examples/protocols/aws_iot/subscribe_publish/main/component.mk#L5-L22

`extern const uint8_t aws_root_ca_pem_start[] asm("_binary_aws_root_ca_pem_start");`

The first part (variable name used in C) can be changed as you wish. The second part (symbol name) is derived from the file name. If the file name is `foo-1.bar`, the corresponding variable name will become `_binary_foo_1_bar_start`. I can't find a documentation reference for this now, but this is how `objcopy` works.

llewellynnu, 2017-05-23:
Thank you for your support .

closed 2017-05-31 by FayeY

shirish47, 2017-12-14:
hi I think I have similar issue

every thing compiles fine but I think not able to link.
```
AR libmain.a
LD gatt_client_demo.elf
/Users/shirishbytelens/esp/bytelensproject_AWSIoT/build/main/libmain.a(gattc_demo.o):(.literal.aws_iot_task+0x18): undefined reference to `_binary_aws_root_ca_pem_start'
/Users/shirishbytelens/esp/bytelensproject_AWSIoT/build/main/libmain.a(gattc_demo.o):(.literal.aws_iot_task+0x1c): undefined reference to `_binary_certificate_pem_crt_start'
/Users/shirishbytelens/esp/bytelensproject_AWSIoT/build/main/libmain.a(gattc_demo.o):(.literal.aws_iot_task+0x20): undefined reference to `_binary_private_pem_key_start'
collect2: error: ld returned 1 exit status
make: *** [/Users/shirishbytelens/esp/bytelensproject_AWSIoT/build/gatt_client_demo.elf] Error 1
```
my folider does not have kconfig.projbuild

I have copied sub_pub examples component.mk as mine was empty. so how do I go with this ?

shirish47, 2017-12-14:
ok I resolved this by simply copying kconfig.projvbuild from subscribe_publish example.