espressif/esp-idf #637: AWS-IOT advanced configuration (change AWS_IOT_MQTT_TX_BUF_LEN)

Vlad-Iliescu, 2017-05-23:
Recently working on an aws-iot client I had the need to increase the ` AWS_IOT_MQTT_TX_BUF_LEN`. The only way I found to increase this was by editing the file [aws_iot_config.h](https://github.com/espressif/esp-idf/blob/master/components/aws_iot/include/aws_iot_config.h). However I find this not being the best way since on the next esp-idf update this will get overwritten. 

Is there another safe way to increase this without having to edit the component itself.

mkonnov, 2017-05-23:
I have changed it the same way. It's not overwriting. Refer to git basics.

"aws-iot-device-sdk-embedded-C" is a separate submodule. It means that bringing this setting into menuconfig (even when it seems more reasonable than direct file changing) will require change this file by esp-idf devs. Which means it must be forked or copied into a project (as FreeRTOS, for example).

Smeedy, 2017-05-23:
Also ran into this one. So you are saying that the only way right now is to change it in the submodule? Indeed, this does work, but it is a bit of a crude solution. Why doesn't an #undef and again #define should cope with this? Couldn't make this happen and indeed had to revert to changing the submodule itself.

mkonnov, 2017-05-23:
I'm not saying about the only right way.
And I forgot about #undef so he solution you proposed is better.

Smeedy, 2017-05-23:
I was not making myself clear. I /tried/ the #undef/#def, but this fails and I'm stuck with the original values. So I also reverted to altering the config.h.

mkonnov, 2017-05-23:
Get you.
I just noticed that I was wrong regarding the submodule.
aws_iot_config.h is not the part of "aws-iot-device-sdk-embedded-C" submodule, so it can be changed to be configured from themenuconfig.

knap1930, 2017-05-24:
even if it were in the aws-iot-device-sdk-embedded-C submodule, it has already been cloned...

https://github.com/espressif/aws-iot-device-sdk-embedded-C/

mkonnov, 2017-05-24:
This is just some repo that's have nothing common with 'esp-idf'.
I could copy it in my profile and say the same.

knap1930, 2017-05-24:
Not true, take a look at https://github.com/espressif/esp-idf/tree/master/components/aws_iot

Notice the submodule is pointing at https://github.com/espressif/aws-iot-device-sdk-embedded-C/tree/7132505b00d2dd57f48478e75efa636021919aae

espressif has cloned the AWS repo, made some [changes](https://github.com/espressif/aws-iot-device-sdk-embedded-C/compare/4e0d995c09cd2a98803012460f827cbd2e9b9453...7132505b00d2dd57f48478e75efa636021919aae), and pointed the submodule to their clone.

mkonnov, 2017-05-24:
I didn't notice that.
Thanks for explanation.

closed 2017-08-08 by igrr