espressif/esp-idf #638: PCNT with PCNT_UNIT_5

SirRibeiro, 2017-05-23:
Hello,
using the pcnt example when i change the PCNT_TEST_UNIT from PCNT_UNIT_0 to PCNT_UNIT_5 it doesn't work anymore, the same happens with the  PCNT_UNIT_6 and PCNT_UNIT_7, the other ones from 0 to 4 work well.

Is there a special configuration for this pcnt units?

Thanks

SirRibeiro, 2017-05-23:
Perfect!

By reading the [http://www.espressif.com/sites/default/files/documentation/esp32_chip_pin_list_en.pdf](url) 
I thought that could be the problem. 

I was just not sure what to change to solve it, thanks for the hint.

So I change the code in the pcnt.c from this:
` int input_sig_index = (channel == 0 ? PCNT_SIG_CH0_IN0_IDX + 4 * unit : PCNT_SIG_CH1_IN0_IDX + 4 * unit);
int ctrl_sig_index = (channel == 0 ? PCNT_CTRL_CH0_IN0_IDX + 4 * unit : PCNT_CTRL_CH1_IN0_IDX + 4 * unit);`

To this:

`int input_sig_index = 0;
    int ctrl_sig_index = 0;
    if(unit<5)
    {
	 input_sig_index = (channel == 0 ? PCNT_SIG_CH0_IN0_IDX + 4 * unit : PCNT_SIG_CH1_IN0_IDX + 4 * unit);
	 ctrl_sig_index = (channel == 0 ? PCNT_CTRL_CH0_IN0_IDX + 4 * unit : PCNT_CTRL_CH1_IN0_IDX + 4 * unit);
    }
    else 
    {
	input_sig_index = (channel == 0 ? PCNT_SIG_CH0_IN0_IDX + 4  * unit + 12 : PCNT_SIG_CH1_IN0_IDX + 4 * unit + 12);
ctrl_sig_index = (channel == 0 ? PCNT_CTRL_CH0_IN0_IDX + 4 * unit +12 : PCNT_CTRL_CH1_IN0_IDX + 4 * unit + 12);
    }`

and the problem is solved! 
Thanks once again @negativekelvin

SirRibeiro, 2017-05-23:
By the way,
the pcnt_counter_clear doesn't work, but if you use the code suggested by @SpenZerX on the forum it works fine.
`esp_err_t pcnt_counter_clear(pcnt_unit_t pcnt_unit)
{
   PCNT_CHECK(pcnt_unit < PCNT_UNIT_MAX, PCNT_UNIT_ERR_STR, ESP_ERR_INVALID_ARG);
   PCNT_ENTER_CRITICAL(&pcnt_spinlock);
   PCNT.ctrl.val |= (BIT(PCNT_PLUS_CNT_RST_U0_S + (pcnt_unit * 2)));
   PCNT.ctrl.val &= (~(BIT(PCNT_PLUS_CNT_RST_U0_S + (pcnt_unit * 2))));
   PCNT_EXIT_CRITICAL(&pcnt_spinlock);
   return ESP_OK;
}`

igrr, 2017-06-22:
Fixed in https://github.com/espressif/esp-idf/commit/769944b067b2b4be4c45cd10fa68981fe411ec2e.

closed 2017-06-22 by igrr