espressif/esp-idf #642: Build system does not support two components with the same name in different directories

Oromis, 2017-05-24:
It's not possible to compile a project with two identically-named components in different component directories. Example:

The project Makefile defines `COMPONENT_DIRS` as `$(CURDIR)/components $(CURDIR)/other_components $(IDF_PATH)/components`. This structure is useful because I want to build different versions of my application - one with the code from `other_components` and one without them. 

Both `$(CURDIR)/components` and `$(CURDIR)/other_components` contain a directory called `os`, with each of them containing a valid component (an empty `component.mk` and some `.cpp` files). 

When invoking make, only the code in `$(CURDIR)/components/os` is being compiled and linked, the other component is just being ignored. 

It would be nice if we could name our components any way we want without paying attention to name clashes. Additionally, the current behavior allows conflicts between user-defined components and system components. For example, if I were to place a component named `driver` in my app's `components` directory, it would be ignored completely because `$(IDF_PATH)/components` also contains a component named `driver`.

Spritetm, 2017-05-24:
This is very much on purpose. It allows you to e.g. copy a component from the esp-idf directory to your own project and modify it there, without having to upset esp-idf system-wide. Suggest you don't choose your components in such a way their names clash.

igrr, 2017-05-24:
See #457.

Oromis, 2017-05-24:
Okay, in this case this it would be helpful to explicitly state this fact in the build system docs. This has cost me several hours of chasing strange errors. 

But thank you for your quick response, your help is much appreciated!

Spritetm, 2017-05-25:
Fair enough, will give the documentation a readthrough to see if it's mentioned there.

igrr, 2018-02-12:
Documentation has been updated: http://esp-idf.readthedocs.io/en/latest/api-guides/build-system.html#multiple-components-with-the-same-name.

closed 2018-02-12 by igrr