espressif/esp-idf #643: Bluedroid GATT Server keeps old characteristics UUID

figueiredoAF, 2017-05-24:
I'm configuring the UUID characteristic on a service I created on my GATT server, and after the bluedroid event ESP_GATTS_ADD_CHAR_EVT I can see the characteristic was created with the right UUID.

 However, when I download a firmware in ESP32 and run the application, my GATT client (mobile app) sees the characteristic with an old UUID that I used early.

Does the bluetooth (bluedroid) stack have any memory that keeps the old settings even after downloading a new firmware?

I'm creating a GATT server attribute table (esp_gatts_attr_db_t) and using the bluedroid functions esp_ble_gatts_create_attr_tab and esp_ble_gatts_start_service to create and start my services.

collin80, 2017-05-24:
This is a known issue with the client side. The client will cache things
and then use the old data upon subsequent connections instead of requerying
the server side. Yes, this is horribly annoying. Look up the proper
procedure for clearing the cache on your mobile OS.

heyinling, 2017-05-25:
You may try to change service dynamically during connection (delete and re-create service). ESP32 GATTS will send service change event to client. client will refresh its gatt db (maybe need mobile app to redo service discovery). This works for ESP32 gatt client so it should work for Android.

ESP32 gatt client provides API `esp_ble_gattc_cache_refresh` which flush DB cache (latest master). You may call this API before service discovery after connected.

figueiredoAF, 2017-05-25:
Hello,

The problem really was the client cache. I have used an android API to refresh the cache before the service discovery.

Thank you very much !!!

closed 2017-05-25 by figueiredoAF