espressif/esp-idf #644: bt_controller_task crashes receiving LE ACL with Host Controller to Host Flow Control

mringwal, 2017-05-25:
I'm testing receiving LE ACL packets sent at full speed but slowing down reception with BTstack. 

Setup: LE Streamer example runs on Mac with USB Bluetooth dongle, LE Streamer Client runs on ESP32.

After start ESP32 looks for peer device and connects, looks for characteristics and enables notifications.

After receiving a few of them, it crashes in bt_controller_task. The num packets the host can accept was set to 10 in this test. Interestingly, if I reduce this, it crashes earlier. E.g. with num acl packets set to 1, I don't get a single ACL packet (it crashes before that).

The packet callback just stores the packets in a large ring buffer after checking that there's enough space: https://github.com/bluekitchen/btstack/blob/develop/port/esp32/template/components/btstack/main.c#L113

When I disable the Host to Host Controller Flow Control, I'll eventually get the message that there's no space in the ringbuffer and loose packets, but it runs stable otherwise.

Feel free to ask for more details/logs, or help to reproduce it on your site.
I'll repeat the same with Classic data next.

Log incl. crash: 
[le-streamer-client-crash-max-10-acl.txt](https://github.com/espressif/esp-idf/files/1029157/le-streamer-client-crash-max-10-acl.txt)

mringwal, 2017-05-25:
I've prepared a test to stream SPP data and get more or less the same behaviour. A  ACL packets during RFCOMM setup are acknowledged by BTstack, and then it crashes.

Log incl. crash:
[spp-streamer-client-crash-max-10-acl.txt](https://github.com/espressif/esp-idf/files/1029748/spp-streamer-client-crash-max-10-acl.txt)

TianHao-Espressif, 2017-06-01:
@mringwal , could you provide the git commit ID? Thanks.

mringwal, 2017-06-01:
@TianHao-Espressif I currently have 55fb4c54c8c00040439825ea9ef8132d1fa5a261 for esp-idf and 8f6c2c5 for components/bt/lib

It should be straight forward to run the test on your side, following https://github.com/bluekitchen/btstack/tree/master/port/esp32 

Host Controller to Host is disabled so far, but you can enable it by uncommenting it in btstack/port/esp32/template/component/btstack/include/btstack_config.h 
(#define ENABLE_HCI_CONTROLLER_TO_HOST_FLOW_CONTROL)

You could try spp_streamer[_client] or le_streamer[_client]

mywang-espressif, 2017-06-08:
@mringwal thank you very much for your information and example, and I could build and run the test code successfully. I have uncommented the line "#define ENABLE_HCI_CONTROLLER_TO_HOST_FLOW_CONTROL" in btstack_config.h as you mentioned, and found that the program may crash in either of following conditions:
1. ESP-IDF submodule "components/bt/lib" is earlier than(not include) commit ID 8f6c2c5
2. not disable the hci_dump or other major loggings, e.g. the "printf_hexdump" in le_streamer_client.c

For scenario 1, We fixed one bug in the specified commit(8f6c2c5), but there was an accident in the merge of code for ESP-IDF, the submodule components/bt/lib was mistakenly rolled back for several days and now the issue is fixed. So could you please check it again or do another test using the latest ESP-IDF and its submodules?

For scenario 2, the program is indeed easy to crash when too many loggings are enabled. After I have disabled these loggings, your example can run well. Therefore I guess it may not be the problem of HCI flow control logic.

To conclude, can you test again with the latest ESP-IDF with the major loggings disabled? Or can you tell me whether there is anything else I must do to carry on the test code on my side? For now I only uncomment the "#define ENABLE_HCI_CONTROLLER_TO_HOST_FLOW_CONTROL" in btstack_config.h, for the le_streamer[_client] and spp_streamer[_client] example.

mringwal, 2017-06-09:
@mywang-espressif Hi Karl! Thanks for looking into this - and even playing with BTstack. I'll pull and report back over the weekend.

mringwal, 2017-06-09:
@mywang-espressif hi again. I've updated to the current master revision and the first problem seems to be fixed now. Excellent!

You suggest to disable logging mostly. Without logging, it doesn't crash -that's correct, too. 

However, with logging enabled it still crashes. The logging happens on the BTstack thread, so the main effect of logging is to slow down reception. Instead of logging, I just tried to sleep using `vTaskDelay(100 / portTICK_PERIOD_MS);` -- same crash happens.

I assume that it's ok for BTstack to slow down reception - after all, that's the point of the Host Controller to Host Flow Control: with logging or delays, BTstack acknowledges received ACL packets slower than it gets them from the bt_controller_task. When this happens, I would expect the HCI implementation to signal to the peer device that it needs to pause sending, as there are no additional buffers on our side.

Please let me know if I misunderstood anything with the Host Controller to Host Flow Control and let me know why it crashes when logging or slowing down reception / and what to do. Slow reception is a likely scenario when using the ESP32 e.g. as a virtual serial port and the physical device raises it's RTS line, so we cannot send data to it.

When BTstack is slowed down, the 'host num completed'

mywang-espressif, 2017-06-12:
@mringwal, Hi Matthias, I agree with you on the flow control mechanism, when Host slows down in processing the received packets, Controller should inform the peer device with FLOW flag set to STOP, indicating that the reception buffer is full. I will study the data path in the Controller, and give my feedback to you once I have any progress.

mringwal, 2017-06-12:
@mywang-espressif Hi Karl. I'm glad you agree. I did not look how it's implemented in the Controller, but it should be possible. Good luck. Feel free to discuss Bluetooth details (although I need to read up as well).

mywang-espressif, 2017-06-22:
@mringwal Hi Matthias, I have fixed the bug that results crash in Controller, and initial result shows that the flow control mechanism works. I will update the library in a couple of days when more tests are done.

mringwal, 2017-06-22:
@mywang-espressif Awesome. Is this in the latest commit (0f68ece) to the Bluetooth Lib?

mywang-espressif, 2017-06-23:
@mringwal Yes, Commit(0f68ece) fixes the bug that causes the controller to crash. As far as I know, For classic Bluetooth, flow control mechanism works in the way that when RX buffer is full, the hardware sets the ARQN bit in packet header to be NAK and makes the peer device to retransmit the packet. I have tested the spp_streamer[_client] example and add delay in the host and reduce the host reception buffer size, the speed can slow down for now.
For BLE I think there is similar Flow Control mechanism, when I test the LE_streamer[_client], the controller will not crash but the connection is disconnected from the loggings of BTStack. I wonder where the problem lies, Host or Controller.
Please try with the update, and I am looking forward to your feedback.

mringwal, 2017-06-23:
@mywang-espressif Hi! Thanks for fixing the flow control for Classic. I just run spp_streamer[_client] with packet log enabled, which slows it down from 190 kB/s to 3.7 kB/s. This clearly shows that the flow control works.

For LE: do you have the "Bluetoooth Low Energy, The Developer's Handbook" by Robin Haydon? Chapter 7.8.6 mentions flow control: "... it is also possible to enable flow control by using the next expected sequence number. If a device does not have enough buffer space to process a message at a given time, it is not able to update the next expected sequence number. This forces the other device to retransmit this message, effectively pushing the buffering requirements onto the sending device and away from the receiving device."

Well, that's not particularly elegant, it clearly is sufficient to avoid packet loss and implement Host Controller to Host Flow Control.

Could you check if it's possible to implement this in the LE part? (not update expected sequence number if there are no free buffers to store the next packet). Thanks!

mywang-espressif, 2017-06-26:
@mringwal Hi Matthias, glad to hear that your Classic BT test passed. I do have the LE book you mentioned in hand and I will also read the specifications of the LE controller. I do find that SESN flag in the packet is used for flow control. Before that, I think I can do some packtcapture work and check where the problem lies.

mringwal, 2017-07-13:
@mywang-espressif Did you find a way to add flow control to LE? Anything to test for me?

mywang-espressif, 2017-07-14:
@mringwal I was engaged in another development task recently and sorry for keep you waiting. I think I can turn to this issue in one week or two and there will be further update then.

mringwal, 2017-07-14:
Thanks for the update. I'm glad it's still on your "radar"!

mringwal, 2017-08-06:
Any news about LE Flow Control?  Thanks!

mywang-espressif, 2017-08-07:
@mringwal Hi Matthias, I am sorry that there was a delay for this issue. I think I can begin to invesitigate in this case later this week, if I cannot I will try to find another teammate to carry on.

mringwal, 2017-08-08:
Thanks for your continuous support.

Yesterday, I've learned (by trying) that the ESP32 support BLE Date Length Extension - that's great.
And today, I tried to see if it can handle multiple BLE Peripheral Roles - yes, it can. That's awesome. (No idea if Bluedroid supports it, but your Controller implements it).

I've added the ESP32 to the Bluetooth Chipset list in BTstack (on develop branch only at the moment): https://github.com/bluekitchen/btstack/tree/develop/chipset

DVSUHZ-g, 2017-08-08:
> develop/chipset

So my cell phone could be (over BT) a peripheral to the ESP32 DevBorad?
Wondering about using phone to send it IMU data that then goes server-side?

mywang-espressif, 2017-08-17:
Hi @mringwal, I am studying the LE hci controller to host data flow control these days. Using your example le_streamer(_client). I added the occasional delay in the le_streamer_client(master) side which is receiving data. When the host task is delaying, the controller stops transmitting packet to host and soon the RX buffer of the controller gets exhausted. I can see the master device sends empty packet with NESN unchanged to make the peer device resending the packet. After some connection events of retransmission, the controller of master reports the disconnection complete event with reason "Connection Timeout".
You can see that the flow control mechanism works that the controller uses NESN bit for retransmission. The only issue now is that the Connection Timeout is not, maybe, an appropriate behaviour in this case.
I will update my progress to you later.

mringwal, 2017-08-17:
That sounds like a good start / not far from the goal :)

As long as the master receives a packet from the slave (the retransmitted packet that it didn't acknowledge yet), it should not disconnect. Looks like some 'connection alive/heartbeat' flag isn't updated if the received packet cannot be queued up, right? 

Thanks for debugging!

mywang-espressif, 2017-08-18:
Hi @mringwal , what you've said is almost the fact. The supervision timer is reset each time a valid packet is received, according to the core specification, which is also implemented in our controller. But in this issue, when controller RX buffer is full, no packet received will be reported since there is no resource for that, and even there is no interrupt for this any more, thus the supervision timer cannot be reset, leading to connection timeout. 
This issue is somehow complicated to find a proper solution, and let me think more about it.

mringwal, 2017-08-25:
From far away..., it looks like it should be possible to extract the code to reset the supervision timer. Then, the call to the reset-supervision-timer could be moved to an earlier point, just after the packet validation. :)

mywang-espressif, 2017-08-28:
Hi @mringwal, there is an update commit (4b54278b) along with submodule "components/bt/lib" commit(caec3249). Please use the latest ESP-IDF with this included. With this patch, LE HCI controller to host data flow control is possible, but the feature is limited by supervision timeout. This patch fixes the issue that when controller RX buffer is full caused by host flow control, controller will stop to send packets to host any more, even when the HCI host_number_of_cmpleted_packets command is sent by host. I think the patch is sufficient for most cases, When I turn on the loggings in the example LE_streamer, the throughput just gets lower.

mringwal, 2017-08-29:
Hi @mywang-espressif thanks for the update. Good point about the current version being good enough to use flow control now. I guess the 'disconnect by supervision timeout because of flow control' should be a new issue. I've updated BTstack to use flow control by default and also close this issue! Thanks & see you at other issues!

closed 2017-08-29 by mringwal