espressif/esp-idf #645: VSCode Debugging via MI causes internal GDB error

gojimmypi, 2017-05-25:
Much of this issue is documented here: [https://github.com/Microsoft/vscode-cpptools/issues/763](https://github.com/Microsoft/vscode-cpptools/issues/763) but in short, I am trying to use `xtensa-esp32-elf-gdb.exe` to remotely debug ESP32 connected to Olimex ARM-USB-OCD-H JTAG debugger. 

I'm using the `xtensa-esp32-elf-gdb.exe` from [https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz) : `GNU gdb (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 7.10`

I can successfully do this in both DOS (Windows 10) as well as Linux, but it appears that when called from the MI in VSCode, something odd happens:

```
C next: {"threadId":-2}
1: (812785) <-1022-exec-next
1: (812793) ->1022^running
1: (812793) ->*running,thread-id="all"
1: (812793) ->(gdb)
1: (812793) 1022: elapsed time 8
1: (812794) ->(gdb)
 R: {"success":true,"message":null,"request_seq":11,"command":"next","body":null,"running":false,"refs":null,"seq":0,"type":"response"}
1: (812903) ->@"esp32.cpu0: Target halted, pc=0x40080E77\n"
E output: {"category":"stdout","output":"@\"esp32.cpu0: Target halted, pc=0x40080E77\\n\"\n","data":null,"type":"output"}
@"esp32.cpu0: Target halted, pc=0x40080E77\n"
1: (812932) ->~"/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.\nA problem internal to GDB has been detected,\nfurther debugging may prove unreliable.\nQuit this debugging session? "
1: (812932) ->~"(y or n) [answered Y; input not from terminal]\n"
E output: {"category":"stdout","output":"/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.\nA problem internal to GDB has been detected,\nfurther debugging may prove unreliable.\nQuit this debugging session? ","data":null,"type":"output"}
1: (812932) ->&"\nThis is a bug, please report it."
/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? 

```
once this error occurs, the only way I have found to resolve is to power cycle the Olimex. It appears from issue #303 that others had VSCode debugging working back in January. Wondering what the current status is? Thanks.

FHFS, 2017-05-26:
what if you try a mode other than MI?

gojimmypi, 2017-05-29:
I'm interested in how that works... when I replace `"type": "cppdbg"` with `"type": "gdb"` VSCode gives me an error: `Configured debug type 'gdb' is not supported.`  Perhaps I need to have some other extension installed?

FHFS, 2017-05-30:
Ooh, wasn't aware of that. I have Native Debug Extension installed and the C/C++ extension

igrr, 2017-06-05:
@gojimmypi Could you please add `set remotelogfile gdb_log.txt` command to your `gdbinit` sequence, or as a command line argument (`-ex "set remotelogfile gdb_log.txt"`) and share the log with this issue happening?

gojimmypi, 2017-06-05:
@igrr (TL;DR jump to end to see most interesting log) 

I first tried the command-line argument. I assume you were referring to openocd; it did not accept that as a parameter:

```
sudo openocd -f interface/ftdi/olimex-arm-usb-ocd-h-1MHz.cfg -c "transport select jtag"  -f target/ESP32-RTOS-none.cfg  -ex "set remotelogfile gdb_log.txt"
Open On-Chip Debugger 0.10.0-dev-g372bb59 (2017-05-19-10:35)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
openocd: invalid option -- 'e'
openocd: invalid option -- 'x'
adapter speed: 1000 kHz
jtag
force hard breakpoints
Info : clock speed 1000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32.cpu0: Target halted, pc=0x40000400
esp32.cpu0: target state: halted
```
So I tried adding as (last) text parameter in VSCode launch.json file:
```{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "OpenOCD Debug",
            "type": "cppdbg",
            "request": "launch",
            "miDebuggerPath": "C:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb.exe",
            "program": "Z:/home/gojimmypi/esp/hello_world/build/hello-world.elf",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                },
                {
                    "text": "file 'Z:/home/gojimmypi/esp/hello_world/build/hello-world.elf'"
                },
                {
                    "text": "target remote 192.168.174.129:3333"
                },
                {
                    "text": "monitor reset init"
                },
                {
                    "text":  "set remotelogfile gdb_log.txt"
                }
            ],
            "externalConsole": false,
            "cwd": "Z:/home/gojimmypi/esp/hello_world/build/",
            "logging": {
                "trace": true,
                "traceResponse": true,
                "engineLogging": true
            }
        }
    ]
}
```

This is what openocd displayed when I pressed F5 in VSCode to debug:

```
0 $ sudo openocd -f interface/ftdi/olimex-arm-usb-ocd-h-1MHz.cfg -c "transport select jtag"  -f target/ESP32-RTOS-none.cfg
[... 253 lines cut ...]
r $00#60
c 1017-stack-list-frames 0 1000
c 1018-stack-list-arguments 2 0 1
c 1019-stack-list-variables 0
c 1020-var-create - * "arg"
c 1021-var-create - * "cpuid"
c 1022-var-create - * "__func__"
```
and VSCode looks like this:
[image]
but when I press a button to step into debugging.... it fails, leaving the ESP32 in the "needs to be reset" state, and just a few lines added to your log file:
```c 1023-exec-step
c step &
w $Z0,400f4ab0,2#35
r $OK#9a
w $s#73
r $O65737033322e637075303a205461726765742068616c7465642c2070633d307834303038304531330a#fd$T05#b9
w $g#67
r $130e0840001000000000000000000000000000000000000000ebfa3f000000000000000000000000e0eafa3f000000000000000000000000000000000000000000000000000000000000000000000000000000009999190000000000010000000000000014000000a5000000a50000000000000097460880603bfe3f7414fb3f020000000000000000000000000000008000000000000000403bfe3f0000000000000000f0000000506bfb3f000000000f000000ae3b0880203bfe3f6c14fb3f00000000000000ffffffff80000000000300000000000000003bfe3f000000009c9402ff0000007fffffff8002000000203cfe3f9a0c0880103cfe3fd843fb3fdcfcfa3f000000000000000000000000000000000200000006000000feffbcc296fec51c20000500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000050140e010000000018150e0100000000c0340e0100000000008810010000000001000000000000009a25347e0e7f00002000000030000000a09acc260000000000000000#98
w $m40080e13,1#8f
r $30#63
w $m40080e10,3#8e
r $364100#2e
```
does this help? Thanks for taking a look.

gojimmypi, 2017-06-05:
@FHFS is this the Native Debug Extension you are referring to?

[image]
[https://marketplace.visualstudio.com/items?itemName=webfreak.debug](https://marketplace.visualstudio.com/items?itemName=webfreak.debugurl)

@igrr I only currently have these extensions currently installed in VSCode; please advise if adding the Native Debug Extension would help:
[image]
thanks

FHFS, 2017-06-05:
@gojimmypi Yup, this is my launch json.
```
{
    "version": "0.1.0",
    "configurations": [
        {
            "name": "OpenOCD Debug",
            "type": "gdb",
            "request": "attach",
            "printCalls": true,
            "executable": "${workspaceRoot}/build/app.elf",
            "target": ":3333",
            "remote": true,
            "cwd": "${workspaceRoot}",
            "autorun": [
                "break app_main"
            ],
            "gdbpath": "/home/fhfs/ESP/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb"
        }
    ]
}
```

gojimmypi, 2017-06-05:
@FHFS thanks for the sample `launch.json` - I installed the webfreak Native Debug and adapted the sample config to my settings. With this, I'm not able to debug at all. When I press F5 the status bar at the bottom of the screen flashes orange for a moment and then nothing. No error, no debugging, no message. The "logging" option appears to not be supported.

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "OpenOCD Debug",
            "type": "gdb",
            "request": "attach",
            "printCalls": true,
            "executable": "Z:/home/gojimmypi/esp/hello_world/build/hello-world.elf",
            "target": "192.168.174.129:3333",
            "remote": true,
            "cwd": "Z:/home/gojimmypi/esp/hello_world/build/",
            "autorun": [
                "break app_main"
            ],
            "gdbpath": "C:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb.exe"
        }
    ]
}
```

the debug console shows a similar "internal GDB" error:
```
1-gdb-set target-async on
2-environment-directory "Z:/home/gojimmypi/esp/hello_world/build/"
3-target-select remote 192.168.174.129:3333
Reading symbols from Z:/home/gojimmypi/esp/hello_world/build/hello-world.elf...
done.
/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? 
(y or n) [answered Y; input not from terminal]
undefined
This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
This is a bug, please report it.
  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.
/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/inline-frame.c:171: internal-error: inline_frame_this_id: Assertion `!frame_id_eq (*this_id, outer_frame_id)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? 
(y or n) [answered Y; input not from terminal]
-gdb-exit

```

After resetting the ESP32 to start fresh. the debug console initially shows this:
```
1-gdb-set target-async on
2-environment-directory "Z:/home/gojimmypi/esp/hello_world/build/"
3-target-select remote 192.168.174.129:3333
Reading symbols from Z:/home/gojimmypi/esp/hello_world/build/hello-world.elf...
done.
ipc_task (arg=0x0) at /home/gojimmypi/esp/esp-idf/components/esp32/./ipc.c:45
45	{
Not implemented stop reason (assuming exception): undefined
4-break-delete
break app_main
5-break-insert -f "z:\\home\\gojimmypi\\esp\\hello_world\\main\\hello_world_main.c:19"
6-stack-list-frames 0 20
break app_main
Breakpoint 1 at 0x400f4b38: file /home/gojimmypi/esp/hello_world/main/./hello_world_main.c, line 46.
7-exec-continue
8-stack-list-variables --thread 1 --frame 0 --simple-values
esp32.cpu0: Target halted, pc=0x400F4B38
Breakpoint 
1, app_main () at /home/gojimmypi/esp/hello_world/main/./hello_world_main.c:46
46	{
9-stack-list-frames 0 20
10-data-evaluate-expression ipc_task
```
Curiously I am able to step into `main()` - however when attempting to step over the `printf`, it fails and is unable to recover with this error in VSCode:

`Unable to open 'puts.c': File not found (\Users\ivan\e\newlib_xtensa-2.2.0-bin\newlib_xtensa-2.2.0\newlib\libc\stdio\puts.c).
`
and this message in the debug console
```
11-exec-continue
esp32.cpu0: Target halted, pc=0x400F4B3B
esp32.cpu0: Target halted, pc=0x400F4AB3
Breakpoint 
2, main () at /home/gojimmypi/esp/hello_world/main/./hello_world_main.c:19
19	        printf("Hello world!\n");
12-stack-list-frames 0 20
13-stack-list-variables --thread 1 --frame 0 --simple-values
14-exec-next
esp32.cpu0: Target halted, pc=0x400F4AB6
esp32.cpu0: Target halted, pc=0x400E3C1C
Warning:
Cannot insert breakpoint 0.
Cannot access memory at address 0x400f4ab9
puts (s=0xa5 "") at ../../../.././newlib/libc/stdio/puts.c:137
137	../../../.././newlib/libc/stdio/puts.c: No such file or directory.
Not implemented stop reason (assuming exception): undefined
15-stack-list-frames 0 20
16-stack-list-variables --thread 1 --frame 0 --simple-values

```
additionally, unlike the other method of debugging - I am unable to place breakpoints with webfreak Native Debug; the button is gray, "unverified breakpoint" and does not get triggered.

gojimmypi, 2017-06-05:
I should also point out that I am using a JTAG esp32.cfg file in  `~/workspace/openocd-esp32/tcl/target` that looks like this:

```
#With no variables set, openocd will configure JTAG for the two cores of the ESP32 and
#will not automatic RTOS detection. This can be be adjusted:

#set ESP32_ONLYCPU 1        # Only configure the PRO CPU
#set ESP32_ONLYCPU 2        # Only configure the APP CPU

# Disable RTOS support
set ESP32_RTOS none        

#set ESP32_RTOS freertos    # Force RTOS to be FreeRTOS

if { [info exists CHIPNAME] } {
	set _CHIPNAME $CHIPNAME
} else {
	set _CHIPNAME esp32
}

if { [info exists CPUTAPID] } {
	set _CPUTAPID $CPUTAPID
} else {
	set _CPUTAPID 0x120034e5
}

if { [info exists ESP32_RTOS] } {
	set _RTOS "$ESP32_RTOS"
} else {
	set _RTOS "auto"
}

if { [info exists ESP32_ONLYCPU] } {
	set _ONLYCPU $ESP32_ONLYCPU
} else {
	set _ONLYCPU 3
}

proc esp_core_halt { tgt } {
	#We need to disable the watchdogs here.
	#TIMG1 WDT
	$tgt mww 0x3FF5F064 0x50D83AA1
	$tgt mww 0x3FF5F048 0x0
	#TIMG2 WDT
	$tgt mww 0x3FF60064 0x50D83AA1
	$tgt mww 0x3FF60048 0x0
	#RTC WDT
	#ToDo: Figure out how to kill the RTC WDT
}

proc configure_esp32_core { TGT } {
	$TGT configure -event halted [list esp_core_halt $TGT]
}

#Change the name of the CPU taps depending on if it's enabled or not. This way, the user
#gets immediate feedback in the openocd logs.
if { $_ONLYCPU == "1" } {
	set _CPU0NAME cpu0
	set _CPU1NAME ignored
} elseif { $_ONLYCPU == "2" } {
	set _CPU0NAME ignored
	set _CPU1NAME cpu1
} else {
	set _CPU0NAME cpu0
	set _CPU1NAME cpu1
}

#Do add both taps, even if one of the CPUs is disabled.
jtag newtap $_CHIPNAME $_CPU0NAME -irlen 5 -expected-id $_CPUTAPID
jtag newtap $_CHIPNAME $_CPU1NAME -irlen 5 -expected-id $_CPUTAPID
set _TARGETNAME1 $_CHIPNAME.cpu1
set _TARGETNAME2 $_CHIPNAME.cpu0

if { $_ONLYCPU != 2 } {
	if { $_RTOS == "none" } {
		target create $_TARGETNAME2 esp108 -endian little -chain-position $_TARGETNAME2
	} else {
		target create $_TARGETNAME2 esp108 -endian little -chain-position $_TARGETNAME2 -rtos $_RTOS
	}
	configure_esp32_core $_TARGETNAME2
}
if { $_ONLYCPU != 1 } {
	if { $_RTOS == "none" } {
	target create $_TARGETNAME1 esp108 -endian little -chain-position $_TARGETNAME1
	} else {
	target create $_TARGETNAME1 esp108 -endian little -chain-position $_TARGETNAME1 -rtos $_RTOS
	}
	if { $_ONLYCPU != 3 } {
		configure_esp32_core $_TARGETNAME1
	}
}

#Force hw breakpoints. Once we have a memory map, we can also allow software bps.
gdb_breakpoint_override hard
```
...which is significantly different that the one mentioned [here](http://esp-idf.readthedocs.io/en/latest/api-guides/openocd.html); I found another esp32.cfg file in my  `/./home/gojimmypi/esp/esp-idf/docs/api-guides/esp32.cfg` (which I assume is the one referred to in those docs) that looks like this:
```
#
# Example configuration file to hook up an ESP32 module or board to a JTAG
# adapter. Please modify this file to your local setup.
#
#

# Include the configuration for the JTAG adapter. We use the Tian TUMPA here.
# If you have a different interface, please edit this to include the
# configuration file of yours.
source [find interface/ftdi/tumpa.cfg]

# The ESP32 only supports JTAG.
transport select jtag

# The speed of the JTAG interface, in KHz. If you get DSR/DIR errors (and they
# do not relate to OpenOCD trying to read from a memory range without physical
# memory being present there), you can try lowering this.
adapter_khz 200

# With no variables set, openocd will configure JTAG for the two cores of the ESP32 and
# will do automatic RTOS detection. This can be be adjusted by uncommenting any of the
# following lines:

# Only configure the PRO CPU
#set ESP32_ONLYCPU 1
# Only configure the APP CPU
#set ESP32_ONLYCPU 2
# Disable RTOS support
#set ESP32_RTOS none
# Force RTOS to be FreeRTOS
#set ESP32_RTOS FreeRTOS

#Source the ESP32 configuration file
source [find target/esp32.cfg]

# The TDI pin of ESP32 is also a bootstrap pin that selects the voltage the SPI flash
# chip runs at. When a hard reset happens (e.g. because someone switches the board off
# and on) the ESP32 will use the current TDI value as the bootstrap value because the
# JTAG adapter overrides the pull-up or pull-down resistor that is supposed to do the
# bootstrapping. These lines basically set the idle value of the TDO line to a
# specified value, therefore reducing the chance of a bad bootup due to a bad flash
# voltage greatly.

# Enable this for 1.8V SPI flash
esp108 flashbootstrap 1.8
# Enable this for 3.3V SPI flash
#esp108 flashbootstrap 3.3
```
which has the target and interface included in one config file - however what is missing from my file are those last few lines:
```
# Enable this for 1.8V SPI flash
esp108 flashbootstrap 1.8
# Enable this for 3.3V SPI flash
#esp108 flashbootstrap 3.3
```
should I be using one of those voltage settings?

and for completeness, here's my adapter config:
```
#
# Olimex ARM-USB-OCD-H
#
# http://www.olimex.com/dev/arm-usb-ocd-h.html
#

interface ftdi
ftdi_device_desc "Olimex OpenOCD JTAG ARM-USB-OCD-H"
ftdi_vid_pid 0x15ba 0x002b

ftdi_layout_init 0x0908 0x0b1b
ftdi_layout_signal nSRST -oe 0x0200
ftdi_layout_signal nTRST -data 0x0100
ftdi_layout_signal LED -data 0x0800

adapter_khz 1000

```

FHFS, 2017-06-06:
When I have some time I will test my JTAG setup again, and post my logs. I have the same debugger too =) but since it was soo much trouble I just used printf.

gojimmypi, 2017-06-09:
@FHFS note there's a WIP OpenOCD here: [https://github.com/espressif/openocd-esp32/wiki/Work-in-progress-ESP32-dual-core-target](https://github.com/espressif/openocd-esp32/wiki/Work-in-progress-ESP32-dual-core-target).  I posted initial progress [on my blog here](https://gojimmypi.blogspot.com/2017/03/jtag-debugging-for-esp32.html). (not using the Native Debugger, rather the cppdbg)

tom-borcin, 2021-03-25:
Hello @gojimmypi, we are closing this issue now. It was resolved by commit: https://github.com/espressif/esp-idf/commit/3ba63a520c91b961a2632ad7385c8bac9039e197

If you think this issue should remain open, please let me know. Thank you!

closed 2021-03-25 by tom-borcin