espressif/esp-idf #647: SPI Without CS

hopkinskong, 2017-05-28:
Hi,

I want to use the SPI bus as master to drive a shift register (which does NOT require CS pin).

In spi_master.c, there is a check at `spi_bus_add_device`:
`SPI_CHECK(dev_config->spics_io_num < 0 || GPIO_IS_VALID_OUTPUT_GPIO(dev_config->spics_io_num), "spics pin invalid", ESP_ERR_INVALID_ARG);`

Any ways that I can bypass it to disable the use of CS pin on the SPI bus?

Thanks.

closed 2017-05-28 by hopkinskong

Spritetm, 2017-05-28:
Yes. As it says in the documentation, passing -1 will disable CS. Note that the check in SPI_CHECK has to be TRUE, otherwise the check fails, so a cs value of -1 will pass the check.

hopkinskong, 2017-05-28:
@Spritetm Yes, silly me, I just discovered it...lol

Spritetm, 2017-05-28:
Can you next time not delete/mangle the issue, though? If you just close it, people who run into a similar issue can find your question using a search engine. If you kill the issue like you did here, you're making that impossible. Better to just answer what you found and close it yourself.

hopkinskong, 2017-05-28:
@Spritetm  Sure, I reverted the question.

Flotschipopotschi, 2022-05-10:
> @Spritetm Yes, silly me, I just discovered it...lol
Seems like you are not the only silly person out there... 

> Can you next time not delete/mangle the issue, though? If you just close it, people who run into a similar issue can find your question using a search engine. If you kill the issue like you did here, you're making that impossible. Better to just answer what you found and close it yourself.
Thank you for the comment, this saved me!

lanistor, 2024-06-29:
@Spritetm  Well, Why `SPI-without-CS` not supported on `release v4.4` anymore?

> Yes. As it says in the documentation, passing -1 will disable CS. Note that the check in SPI_CHECK has to be TRUE, otherwise the check fails, so a cs value of -1 will pass the check.