espressif/esp-idf #651: WiFi in AP mode crash on device join

MicroJoe, 2017-05-30:
I am using a Sparkfun ESP32 Thing in AP mode to do connectivity tests.

The board will crash and not reboot upon device connection to the AP. No stack trace is provided during the crash so this is quite hard to debug for me right now. I am using the SDK on `master` branch, but same behavior happens on `release/v2.1` branch.

Testing code:

```c
void app_main(void)
{
  /* Initialize NVS flash memory access */
  ESP_ERROR_CHECK(nvs_flash_init());

  tcpip_adapter_init();

  ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );

  /* Initialize Wi-Fi */
  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
  ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );

  ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_AP) );

  wifi_config_t ap_config = {
    .ap = {
      .ssid = "ESP32-Test",
      .password = "1234567890",
      .ssid_len = strlen("ESP32-Test"),
      .authmode = WIFI_AUTH_WPA2_PSK,
      .max_connection = 1
    }
  };

  ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_AP, &ap_config) );
  ESP_ERROR_CHECK( esp_wifi_start() );
  //ESP_ERROR_CHECK( esp_wifi_connect() );

  /* Init GPIOs */
  gpio_set_direction(LED_BUILTIN, GPIO_MODE_OUTPUT);

  for(;;) {
    gpio_set_level(LED_BUILTIN, 1);
    DELAY(50);
    gpio_set_level(LED_BUILTIN, 0);
    DELAY(50);
    gpio_set_level(LED_BUILTIN, 1);
    DELAY(50);
    gpio_set_level(LED_BUILTIN, 0);
    DELAY(50);
    DELAY(500);
  }

  vTaskDelete(NULL);
}
```

Monitor output:

```
MONITOR
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbserial-DN02MPIP instead...
--- idf_monitor on /dev/cu.usbserial-DN02MPIP 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3424
load:0x40078000,len:10388
ho 0 tail 12 room 4
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

D (814) nvs: nvs_flash_init_custom start=9 count=6
D (834) nvs: nvs_open misc 1
D (834) nvs: nvs_get_str_or_blob log
I (835) wifi: wifi firmware version: 8a1ad86
I (835) wifi: config NVS flash: enabled
I (837) wifi: config nano formating: disabled
D (841) nvs: nvs_open nvs.net80211 1
D (845) nvs: nvs_get opmode 1
D (848) nvs: nvs_get country 1
D (851) nvs: nvs_get_str_or_blob sta.ssid
D (856) nvs: nvs_get_str_or_blob sta.mac
D (860) nvs: nvs_get sta.authmode 1
D (863) nvs: nvs_get_str_or_blob sta.pswd
D (867) nvs: nvs_get_str_or_blob sta.pmk
D (871) nvs: nvs_get sta.chan 1
D (874) nvs: nvs_get auto.conn 1
D (877) nvs: nvs_get bssid.set 1
D (881) nvs: nvs_get_str_or_blob sta.bssid
D (885) nvs: nvs_get sta.phym 1
D (888) nvs: nvs_get sta.phybw 1
D (891) nvs: nvs_get_str_or_blob sta.apsw
D (896) nvs: nvs_get_str_or_blob sta.apinfo
D (902) nvs: nvs_get_str_or_blob ap.ssid
D (904) nvs: nvs_get_str_or_blob ap.mac
D (908) nvs: nvs_get_str_or_blob ap.passwd
D (912) nvs: nvs_get_str_or_blob ap.pmk
D (916) nvs: nvs_get ap.chan 1
D (919) nvs: nvs_get ap.authmode 1
D (922) nvs: nvs_get ap.hidden 1
D (926) nvs: nvs_get ap.max.conn 1
D (929) nvs: nvs_get bcn.interval 2
D (933) nvs: nvs_get ap.phym 1
D (936) nvs: nvs_get ap.phybw 1
D (939) nvs: nvs_get ap.sndchan 1
D (943) nvs: nvs_set_blob sta.mac 6
D (958) nvs: nvs_set_blob ap.mac 6
I (959) wifi: Init dynamic tx buffer num: 32
I (959) wifi: Init dynamic rx buffer num: 64
I (960) wifi: wifi driver task: 3ffcf0fc, prio:23, stack:4096
I (965) wifi: Init static rx buffer num: 10
I (969) wifi: Init dynamic rx buffer num: 0
I (973) wifi: Init rx ampdu len mblock:7
I (976) wifi: Init lldesc rx ampdu entry mblock:4
I (981) wifi: wifi power manager task: 0x3ffd44d4 prio: 21 stack: 2560
I (1645) wifi: wifi timer task: 3ffd5564, prio:22, stack:3584
D (1646) nvs: nvs_open phy 0
D (1646) nvs: nvs_get cal_version 4
D (1647) nvs: nvs_get_str_or_blob cal_mac
D (1651) nvs: nvs_get_str_or_blob cal_data
D (1660) nvs: nvs_close 3
I (1675) wifi: mode : softAP (24:0a:c4:05:b0:85)
D (1678) tcpip_adapter: check: local, if=1 fn=0x4010c648
0x4010c648: tcpip_adapter_start_api at /Users/*****/esp/esp-idf/components/tcpip_adapter/./tcpip_adapter_lwip.c:801

D (1680) tcpip_adapter: dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
D (1683) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (1688) tcpip_adapter: check: remote, if=1 fn=0x4010c648
0x4010c648: tcpip_adapter_start_api at /Users/*****/esp/esp-idf/components/tcpip_adapter/./tcpip_adapter_lwip.c:801

I (6130) wifi: n:1 0, o:1 0, ap:1 1, sta:2????$?M??@?ɽ??j??
	?S?&?JHW-V-'H.]X?Z?K???҂?????Ҫ2?R????
?E??b:???j??
```

The ????? garbage is not helping me to find why the crash occurs. The board will not reboot upon crash. Led stops blinking, so I think FreeRTOS totally crashes. The interruption watchdog is enabled but not triggered.

liuzfesp, 2017-05-31:
HI @MicroJoe could you provide the complete test code please? When I run your example, everything looks good and no crash happens, and sta can connected to the softap correctly. Since I don't know the implementation of your event_handler, I set it to NULL: ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );  it will be better if you can show me the completed code list...

MicroJoe, 2017-05-31:
Here is the full code:
```c
/*
 * Copyright 2016 Espressif Systems
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

 #include "freertos/FreeRTOS.h"

 /* ESP32 includes */
 #include "esp_wifi.h"
 #include "esp_system.h"
 #include "esp_event.h"
 #include "esp_event_loop.h"
 #include "nvs_flash.h"
 #include "driver/gpio.h"

 /* lwIP includes */
 #include "tcpip_adapter.h"
 #include "lwip/sys.h"
 #include "lwip/netdb.h"
 #include "lwip/api.h"
 #include "lwip/err.h"

 /* LibC includes */
 #include "string.h"

/* Built-in led on ESP32 Thing board */
#define LED_BUILTIN GPIO_NUM_5

#define DELAY(n) vTaskDelay(n / portTICK_PERIOD_MS)

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    return ESP_OK;
}

void app_main(void)
{
  /* Initialize NVS flash memory access */
  ESP_ERROR_CHECK(nvs_flash_init());

  tcpip_adapter_init();

  ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );

  /* Initialize Wi-Fi */
  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
  ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
  ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_AP) );

  wifi_config_t ap_config = {
    .ap = {
      .ssid = "ESP32-Test",
      .password = "1234567890",
      .ssid_len = strlen("ESP32-Test"),
      .authmode = WIFI_AUTH_WPA2_PSK,
      .max_connection = 4
    }
  };

  ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_AP, &ap_config) );
  ESP_ERROR_CHECK( esp_wifi_start() );

  /* Init GPIOs */
  gpio_set_direction(LED_BUILTIN, GPIO_MODE_OUTPUT);

  for(;;) {
    gpio_set_level(LED_BUILTIN, 1);
    DELAY(50);
    gpio_set_level(LED_BUILTIN, 0);
    DELAY(50);
    gpio_set_level(LED_BUILTIN, 1);
    DELAY(50);
    gpio_set_level(LED_BUILTIN, 0);
    DELAY(50);
    DELAY(500);
  }

  vTaskDelete(NULL);
}
```

Here are the messages:
```
st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3424
load:0x40078000,len:10388
ho 0 tail 12 room 4
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

I (848) wifi: wifi firmware version: 8a1ad86
I (848) wifi: config NVS flash: enabled
I (848) wifi: config nano formating: disabled
I (864) wifi: Init dynamic tx buffer num: 32
I (864) wifi: Init dynamic rx buffer num: 64
I (864) wifi: wifi driver task: 3ffcda1c, prio:23, stack:4096
I (866) wifi: Init static rx buffer num: 10
I (870) wifi: Init dynamic rx buffer num: 0
I (874) wifi: Init rx ampdu len mblock:7
I (878) wifi: Init lldesc rx ampdu entry mblock:4
I (882) wifi: wifi power manager task: 0x3ffd2df4 prio: 21 stack: 2560
I (1546) wifi: wifi timer task: 3ffd3e84, prio:22, stack:3584
I (1563) wifi: mode : softAP (24:0a:c4:05:b0:85)
I (4684) wifi: n:1 0, o:1 0, ap:1 1, sta:25eD?MM6#\?˖?j??
	???S%?W-V-'H.]X?Z?K??2??
????*麊R????
?E??b:???j??
```

The strange characters at the end drive me crazy, no stacktrace. Maybe faulty hardware? I remember it was working fine for some time but now it always crash on device connection when AP WiFi is used.

liuzfesp, 2017-05-31:
HI @MicroJoe still can't reproduce this issue, try to flush the NVS flash with  "python /home/zfliu/work/code/esp32/official/new_idf/components/esptool_py/esptool/esptool.py erase_flash" and then download the image. If still not OK, send me your sdkconfig and I need to check whether your menuconfig is same as mine or not, thanks!

MicroJoe, 2017-06-02:
The esptool do not let the user modify the serial port from `/dev/ttyUSB0` to something else:
```
usage: esptool erase_flash [-h] [--spi-connection SPI_CONNECTION]
```

So I had to modify the serial port in the Python script to set it to `/dev/cu.usbserial-DN02MPIP` in my case. Once the flash is successfully erased I reflash the AP program above and get this first monitor output:

```
I (903) wifi: wifi firmware version: 8a1ad86
I (904) wifi: config NVS flash: enabled
I (904) wifi: config nano formating: disabled
I (919) wifi: Init dynamic tx buffer num: 32
I (919) wifi: Init dynamic rx buffer num: 64
I (920) wifi: wifi driver task: 3ffcd994, prio:23, stack:4096
I (922) wifi: Init static rx buffer num: 10
I (925) wifi: Init dynamic rx buffer num: 0
I (929) wifi: Init rx ampdu len mblock:7
I (933) wifi: Init lldesc rx ampdu entry mblock:4
I (938) wifi: wifi power manager task: 0x3ffd2d6c prio: 21 stack: 2560
I (1601) wifi: wifi timer task: 3ffd3dfc, prio:22, stack:3584
I (1619) wifi: mode : softAP (24:0a:c4:05:b0:85)
Gr eiainErr oe0pnce Itrutwttmoto P0)
x<`?҂????0 A    x348 A    x3 ?  x0000 001 
?????
%?????
5???j??????
U?ق???%		????*5j	????j?eEJ??????bU		т?b?		т?b??
	?????????jj$5????????????"??????jj$%?+떩?ets Jun  8 2016 00:22:
```

Then the WDT reset the board. And there is another crash after that:

```
st:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3424
load:0x40078000,len:10388
ho 0 tail 12 room 4
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

I (903) wifi: wifi firmware version: 8a1ad86
I (904) wifi: config NVS flash: enabled
I (904) wifi: config nano formating: disabled
I (919) wifi: Init dynamic tx buffer num: 32
I (919) wifi: Init dynamic rx buffer num: 64
I (920) wifi: wifi driver task: 3ffcd994, prio:23, stack:4096
I (922) wifi: Init static rx buffer num: 10
I (925) wifi: Init dynamic rx buffer num: 0
I (929) wifi: Init rx ampdu len mblock:7
I (933) wifi: Init lldesc rx ampdu entry mblock:4
I (938) wifi: wifi power manager task: 0x3ffd2d6c prio: 21 stack: 2560
I (1601) wifi: wifi timer task: 3ffd3dfc, prio:22, stack:3584
I (1619) wifi: mode : softAP (24:0a:c4:05:b0:85)
I (4923) wifi: n:1 0, o:1 0, ap:1 1, sta:25??`?MM??ɽ??j??
	?S??J??????хѥ?????2??
????*麊R????
?E??b:???j??
```

And after this second crash **the WDT won't reboot the board**.

Here is my `sdkconfig`, thanks for testing on your hardware with this one: https://gist.github.com/MicroJoe/41196c6bb51890c29efd24f36d7c06e3

MicroJoe, 2017-06-02:
I would add to this that I have an STA project that also has this kind of issues: after it connects to a router it crashes with lots of ????? and not a single line of debug information.

liuzfesp, 2017-06-07:
Hi @MicroJoe,  are you Chinese,  if yes, could you mail me your phone number, then we can have a discussion on the phone, my mail liuzhifu@espressif.com, thanks!

liuzfesp, 2017-06-07:
BTW, I just can't open the code in " https://gist.github.com/MicroJoe/41196c6bb51890c29efd24f36d7c06e3"

MicroJoe, 2017-06-09:
For me the `.sdkconfig` is available when I click on this link, make sure you entered the URL correctly.

I have managed to make the board work again but I don't know how. Now I do not have random crashes when I connect to a router. I did not retry in AP mode though but I think this was a hardware failure. Maybe the flash reset helped.

closed 2017-06-09 by MicroJoe

alkberto, 2018-01-15:
I was having