espressif/esp-idf #652: Can't obtain IP address from ethernet example

paul1419, 2017-05-30:
Hello,

I'm trying to run the ethernet example with my esp32, using a custom breakout board based on the schematic for the Waveshare LAN8720 ETH board. However, when I run the program in the terminal window, all the network info repeatedly shows all zeros, and it never can obtain a valid IP address. Does anyone know how I could go about fixing this?

This is what shows up on the terminal window:
[image]

Thanks

mmp9090, 2017-05-30:
Have you considered ending it?

projectgus, 2017-05-31:
Do you have a schematic for your custom breakout board? I've seen similar symptoms caused by missing connections in the RMII interface.

lixianyu, 2017-05-31:
I use 'LAN8720', it work very well.

paul1419, 2017-05-31:
@projectgus Here's a capture of the Eagle schematic for the breakout board I made, sorry if it's a little tricky to read:
[image]

The bottom left is just pin headers, and I've also grounded the SHIELD pins on the Magjack.

hugo-hur, 2018-01-23:
I'm having exactly the same problem with the base ethernet example configured to use LAN8720 when using the Waveshare board. Everything seems to be fine according to the software verbose log output but absolutely no ethernet packets are sent to the network by the lan8720 (confirmed with wireshark on same ethernet hub). Connected my waveshare module according to this post: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/ which seems to be an appropriate connection.

This means the module is not getting an ip address and no dhcp request is actually made. This seems to me as a some sort of RMII interface problem. The module reports back to esp correctly the link status, duplex and speed modes but no packets cannot be sent or received from/to the network. What could be causing this?

pucmelda, 2018-03-21:
I'm having exactly the same problem as hugo-hur (written on 23 Jan). Please help...

hugo-hur, 2018-03-22:
It appears that the 50MHz signals should be routed as close to the esp pins and these should be of same length. In my case the problem arised when prototyping on breadboard with the module connected to the esp using jumper wires. When I made a custom PCB to mount these all the problems vanished and everything works as it should.

paul1419, 2018-03-22:
hugo-hur, sorry for leaving this open for so long. I had the exact same experience as you. The custom board I made was connected to the ESP with jumper wires. After triple checking all the connections and revisting the circuit numerous times, I decided to just revert back to the original circuit and get the board manufactured and populated with all surface mount components, which fixed whatever the problem was. It very well may have been a proximity issue.

projectgus, 2018-03-23:
@hugo-hur , @paul1419 good to hear that these were signal integrity issues and resolved. I'm going to close this issue now.

closed 2018-03-23 by projectgus

marekburiak, 2018-08-15:
I am using ESP32 Devkit board together with Waveshare LAN8720 ETH Board. I initially had it connected on a breadboard but after reading this issue I connected it on a prototyping board as is described and shown here: 
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/. 
I uploaded this sketch: 
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720_internal_clock/ETH_LAN8720_internal_clock.ino with these defines:
```#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   17

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I2C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I2C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I2C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18
```
When the sketch started (with verbose debug level) I only got:
```[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 20 - ETH_START
ETH Started
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 22 - ETH_CONNECTED
ETH Connected
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 23 - ETH_DISCONNECTED
ETH Disconnected
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 22 - ETH_CONNECTED
ETH Connected
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 23 - ETH_DISCONNECTED
ETH Disconnected
```
it never got to getting an IP address (SYSTEM_EVENT_ETH_GOT_IP). After searching multiple sites I stumbled upon a note from espressif about c15 capacitor :
https://github.com/espressif/esp-idf/blob/master/docs/en/get-started/get-started-devkitc.rst#note-on-c15
that is connected between IO0 (through a resistor) and GND forming an RC charging circuit which was unwantedly smoothing out the 50MHz signal coming from the oscillator.
For me this capacitor was the main reason why it never got an IP. After removing it from the board I got:
```[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 20 - ETH_START
ETH Started
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 22 - ETH_CONNECTED
ETH Connected
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 24 - ETH_GOT_IP
[D][WiFiGeneric.cpp:425] _eventCallback(): ETH IP: 192.168.3.124, MASK: 255.255.255.0, GW: 192.168.3.1
ETH MAC: xx:xx:xx:xx:xx:xx, IPv4: 192.168.3.124, FULL_DUPLEX, 100Mbps

connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Wed, 15 Aug 2018 19:44:33 GMT
Expires: Fri, 14 Sep 2018 19:44:33 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
```
I hope this helps somebody.

felixstorm, 2018-11-19:
Thanks a lot @marekburiak, this did indeed help me!

I had the same issues (ETH connecting but no IP/DHCP), removed the capacitor and now it does work.

felixstorm, 2019-12-02:
Last link from https://github.com/espressif/esp-idf/issues/652#issuecomment-413315779 is not working anymore, current link: https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/get-started-devkitc.html#note-on-c15 ([permalink](https://github.com/espressif/esp-idf/blob/62ed221dafdd94077cd251d449c65de0b46502d5/docs/en/hw-reference/get-started-devkitc.rst#note-on-c15))