espressif/esp-idf #653: Make Error ESP32

jumokeaa, 2017-05-30:
Hello,

I've been trying for a few days through different platforms to compile a blink or hello world example for ESP32 DevKit on Ubuntu and everything always seems to work until I hit the "make" command which gives me the following error:

/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PD);
     ^
make[1]: *** [src/bootloader_random.o] Error 1
make: *** [bootloader_support-build] Error 2

I've tried using multiple methods and toolchains but toolchain always compiles fine but I cannot make or flash.

negativekelvin, 2017-05-31:
Are you building the blink example from within esp-idf?  It looks like your code is calling PIN_PULLDWN_EN which is marked as deprecated.  

https://github.com/espressif/esp-idf/blob/07b61d54f7e502bb12527e4b582de121ffa6fa02/components/soc/esp32/include/soc/io_mux_reg.h#L83

jumokeaa, 2017-06-01:
Hi there,
I've just been following the step by step instructions from different sites but I always get stuck when it's time to make. I get the bootloader error every time.

Spritetm, 2017-06-02:
What do you mean by 'the bootloader error'?

jumokeaa, 2017-06-02:
Hello,

The error that I keep getting is:

make[1]: *** [src/bootloader_random.o] Error 1
make: *** [bootloader_support-build] Error 2

I'm assuming there's an error with the bootloader even though when I run "make bootloader", the bootloader builds.

jumokeaa, 2017-06-02:
`jumoke@jumoke-VirtualBox:~/Desktop/iotbits_esp32/esp-idf/examples/get-started/blink$ make
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-59-g8d95cad
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-59-g8d95cad
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC src/bootloader_random.o
In file included from /home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:20:0,
                 from /home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/wdev_reg.h:17,
                 from /home/jumoke/Desktop/iotbits_esp32/esp-idf/components/bootloader_support/src/bootloader_random.c:16:
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_DIS':
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:169:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:96:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:169:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:96:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_EN':
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:101:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PU);
     ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
[... cut ...]
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/home/jumoke/Desktop/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PD);
     ^
make[1]: *** [src/bootloader_random.o] Error 1
make: *** [bootloader_support-build] Error 2`

krzychb, 2017-06-02:
Hi @jumokeaa,

The first message you posted above:
````
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-59-g8d95cad Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
````
indicates you likely have some mixed installation - an old toolchain and newer ESP-IDF.
I think it may be because of "following instructions from different sites".

Instead of now trying to figure out what is mixed and how, I would start over from scratch following http://esp-idf.readthedocs.io/en/latest/get-started/index.html

krzychb, 2017-06-02:
@jumokeaa, sorry, by saying "start over from scratch" I meant deleting everything and loading again.

I would follow "Standard Setup of Toolchain" and use precompiled toolchain instead of "Customized Setup of Toolchain" and building the toolchain. This should be quicker with less potential traps as for the first exposure to ESP32.

jumokeaa, 2017-06-06:
Hello, I've gotten as far as the "make menuconfig" command but it cannot find the toolchain for some reason even though I followed the steps.

This is the error after "make menuconfig"
`jumoke@jumoke-VirtualBox:~/esp/hello_world$ make menuconfig
/home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
/home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
`

These are my PATHs:
`jumoke@jumoke-VirtualBox:~$ printenv PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/jumoke/esp/xtensa-esp32-elf/bin

jumoke@jumoke-VirtualBox:~/esp/hello_world$ printenv IDF_PATH
/home/jumoke/esp/esp-idf`

jumokeaa, 2017-06-06:
I'm missing a "snap/bin" in my PATH but don't know how to add it.

igrr, 2017-06-20:
Can you check if you can run the compiler directly?
```
/home/jumoke/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
```
it should print something like
```
xtensa-esp32-elf-gcc: fatal error: no input files
compilation terminated.
```

noticeable, 2017-07-05:
hello ,i have as far a as issue as you ,have you solved it ?and how?thx

noticeable, 2017-07-05:
it cant build project ,have anyone gotten the same problem?
make[1]: *** [/cygdrive/d/eclipse/esp-idf/make/component_wrapper.mk:211: src/bootloader_random.o] Error 1
make: *** [/cygdrive/d/eclipse/esp-idf/make/project.mk:387: bootloader_support-build] Error 2

projectgus, 2017-07-05:
@noticeable This looks like a different issue to the one above. Can you please delete your comment and then open a [New Issue](https://github.com/espressif/esp-idf/issues/new) for this one? I will reply to you there.

jumokeaa, 2017-07-05:
For anyone wondering- I got it to work. I have no idea why Ubuntu 64-bit was not compatible with the 64-bit. I used the 32-bit on Ubuntu 64-bit.

closed 2017-07-05 by jumokeaa