espressif/esp-idf #655: Feature request: Run tests more easily

yaqwsx, 2017-05-31:
It would be nice if I could run all tests of all components in the current project by issuing e.g. `make tests`. This should build the test-utility-app and run it. Also test shouldn't run interactively - the monitor app should run all test/selected tests based on argument and exit with 0 if all the tests passed or exit with 1 to show error. This would allow for easy integration in other projects.

The current way, where I need to manually put components in test-utility-app and build it separately is inconvenient. Also there is no possibility to automatically detect whether tests succeeded or failed.

projectgus, 2017-05-31:
(@yaqwsx I discussed this on IRC and I agreed it's a good feature for streamlining test use for end projects - as opposed to tests inside IDF.)

projectgus, 2019-03-01:
Nearly two years later, this has gotten a little better now you can run unit tests via idf.py and there is the `unit_test.py` wrapper script which can manage the unit test run.

However these tools are still only loosely integrated, and I think the wrapper script is undocumented. So there is still some work to do here.

AshUK, 2020-07-08:
Would love to see further enhancements to running custom esp-idf component tests. Being able to run them off target in a cli environment would be the dream.

hfudev, 2025-01-07:
Now we've refactored all tests with pytest-embedded in ESP-IDF. Please follow the [ESP-IDF Tests with Pytest Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/esp-idf-tests-with-pytest.html) to check how it works.