espressif/esp-idf #656: problem with adding IDF_PATH in mac OS

shirish47, 2017-05-31:
[image]
I following steps as shown in esp-idf.readthedocs.io but above picture shows the problem. IDF path is not set if I check with printenv IDF_PATH. also echo $IDF_PATH does not show anything.

igrr, 2017-05-31:
The guide mentions `~/.profile`, not `~/.bash`, is that right? Also don't forget to reopen the terminal after modifying `~/.profile` (or do `. ~/.profile`, note the leading dot).

shirish47, 2017-06-01:
[image] if its so than there is problem in document.  I have added the file to .profile. its fine I will do other steps later tomorrow I think this is resolved.  Just need to change documentation.

jcwren, 2017-06-01:
What would be really nice is to move 'env' to 'env.example' in git, so that every time you don't wipe out the env file, and don't end up trying to commit it when you submit a patch.

shirish47, 2017-06-02:
now having trouble with this .. make menu config even after adding to .profile ..[image]

shirish47, 2017-06-02:
my mistake ... ~/esp not ~esp.. sorry ..

closed 2017-06-06 by FayeY