espressif/esp-idf #659: unable to build esp-idf-template in Eclipse

Eman-Hussein, 2017-05-31:
I am trying to build esp-idf-template in eclipse but getting this error
Description	Resource	Path	Location	Type
make: *** [/home/esp-idf/components/bootloader_support/Makefile.projbuild:7: /home/esp_proj_1/secure_boot_signing_key.pem] Error 1	esp_proj_1	
[image]
	 	C/C++ Problem

nkolban, 2017-06-01:
Also being worked in the forum here:

https://esp32.com/viewtopic.php?f=13&t=1999

jumokeaa, 2017-06-01:
I'm having the same problem in Ubuntu

nkolban, 2017-06-01:
@jumokeaa Maybe have a look at the forum post above and post there with your  value of the question asked in the forum?

closed 2017-06-09 by FayeY