espressif/esp-idf #665: make monitor error

Eman-Hussein, 2017-06-02:
I am trying the blink example to get started. I followed the steps here http://esp-idf.readthedocs.io/en/latest ... index.html and when it comes to the last step ,which is "make monitor" to see that the application is running on the board, I get these errors:
Traceback (most recent call last) :
File "C:/Users/emanh/Desktop/ESP32/esp32_win32_msys2_environment_and_toolchain-20170330/msys32/home/esp-idf/tools/idf_monitor.py", line 31, in <module>
import subprocess
File "/mingw32/lib/python2.7/subprocess.py", line 73, in <module>
import select
ImportError: No module named select
make: *** [/home/esp-idf/components/esptool_py/Makefile.projbuild:94: monitor] Error 1

edmund-huber, 2017-06-03:
The `select` module is a standard Python system module. Something about the python installation in msys2 is broken.

Eman-Hussein, 2017-06-06:
I tried to reinstall python using "pacman --noconfirm -S python2" , but I am still getting the same error. 
To make sure that python is not broken I create a script file named "test.py" that imports select and run it using "python test.py" and gave no error.

projectgus, 2017-06-07:
Hi @Eman-Hussein ,

Try `pacman -S mingw-w64-i686-python2`. (MSYS2 ships multiple different kinds of Python, we need the "mingw" flavour so it can natively use the serial ports.) If this doesn't work, please confirm:

* Does your MSYS2 shell prompt contain the word "MINGW32"? (There are different kinds of shells, as well.)
* What is the output of `which python` and `python -V`

Thanks,

Angus

FayeY, 2017-06-11:
Hi @Eman-Hussein , how about this issue now?

Eman-Hussein, 2017-06-11:
Thanks @projectgus , @FayeY and @edmund-huber for your replies. I made a workaround and it is fixed. Here is it :
in /home/esp-idf/components/esptool_py/Makefile.projbuild:94 , I changed $python to python, and it worked, it seems that the $python variable refers to a different version of python that doesn't work.
Thanks again.

projectgus, 2017-06-13:
HI @Eman-Hussein ,

Glad you found a solution. Maybe check the value given in "make menuconfig" under "SDK Tool Configuration" -> "Python 2 Interpreter". It should also be "python" (meaning $python expands to python) but it may have been set to something else?

Angus

dcontris, 2017-09-05:
@projectgus 
I was having the same problem, but instead of needing to replace $python, I had to replace "winpty python" with just "python".

Seems to be working so far...

projectgus, 2017-09-06:
@dcontris What environment are you running IDF from? The prebuilt MSYS environment, or something else?

dcontris, 2017-09-06:
Yes.  The prebuilt msys32

St7ven, 2018-01-21:
winpty python works for me too in the prebuilt msys32

St7ven, 2018-01-21:
For me its writing "pacman -S winpty" on the msys32 that make it works.

I use msys32 on windows 64bits

projectgus, 2018-01-21:
Hi @St7ven ,

How did you install msys32, from the prebuilt environment? This environment should include winpty already, unless it's quite an old version.

Angus

St7ven, 2018-01-28:
Hi @projectgus 

I used an old version. Now, I'm using a new version and I have only write python under "SDK Tool Configuration" -> "Python 2 Interpreter"

projectgus, 2018-01-29:
Hi @St7ven ,

Thanks for the update. Does this mean it's working as expected for you, now?

Angus

colesnicov, 2019-02-22:
When I create 'TARGET' in the eclipse: 'monitor' or 'flash monitor' and I use it (build -> target -> [flash] monitor)

```
make -j2 flash monitor 
including esp/esp-idf/components/bootloader/Makefile.projbuild...
including esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including esp/esp-idf/components/coap/Makefile.projbuild...
including esp/esp-idf/components/esp32/Makefile.projbuild...
including esp/esp-idf/components/esptool_py/Makefile.projbuild...
including esp/esp-idf/components/mbedtls/Makefile.projbuild...
including esp/esp-idf/components/nghttp/Makefile.projbuild...
including esp/esp-idf/components/partition_table/Makefile.projbuild...
including esp/esp-idf/components/ulp/Makefile.projbuild...
PROJECT_PATH= COMPONENT_DIRS= make -C esp/esp-idf/components/bootloader/subproject V=1 BUILD_DIR_BASE=Ovladac/build/bootloader TEST_COMPONENTS= TESTS_ALL= EXCLUDE_COMPONENTS= Ovladac/build/bootloader/bootloader.bin

...
...
...

Target 'component-xtensa-debug-module-build' responsible for 'Ovladac/build/xtensa-debug-module/libxtensa-debug-module.a'
Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000 )...
python esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port "/dev/ttyUSB1" --baud 921600 --before "default_reset" --after "hard_reset" write_flash -z --flash_mode "dio" --flash_freq "40m" --flash_size detect 0x1000 Ovladac/build/bootloader/bootloader.bin 0x10000  Ovladac/build/hello-world.bin 0x8000 Ovladac/build/partitions_singleapp.bin
esptool.py v2.6
Serial port /dev/ttyUSB1
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:1a:be:30
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 19472 bytes to 11604...

Writing at 0x00001000... (100 %)
Wrote 19472 bytes (11604 compressed) at 0x00001000 in 0.2 seconds (effective 905.3 kbit/s)...
Hash of data verified.
Compressed 143280 bytes to 67990...

Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 143280 bytes (67990 compressed) at 0x00010000 in 1.3 seconds (effective 861.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 8135.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[ -f Ovladac/build/hello-world.elf ] || echo "*** 'make monitor' target requires an app to be compiled and flashed first."
[ -f Ovladac/build/hello-world.elf ] || echo "*** Run 'make flash monitor' to build, flash and monitor"
[ -f Ovladac/build/hello-world.elf ] || echo "*** Or alternatively 'make simple_monitor' to view the serial port as-is."
[ -f Ovladac/build/hello-world.elf ] || exit 1
python esp/esp-idf/tools/idf_monitor.py --baud 115200 --port "/dev/ttyUSB1" --toolchain-prefix "xtensa-esp32-elf-" --make "make" --print_filter "" Ovladac/build/hello-world.elf
Traceback (most recent call last):
  File "esp/esp-idf/tools/idf_monitor.py", line 743, in <module>
    main()
  File "esp/esp-idf/tools/idf_monitor.py", line 656, in main
    monitor = Monitor(serial_instance, args.elf_file.name, args.print_filter, args.make, args.toolchain_prefix, args.eol)
  File "esp/esp-idf/tools/idf_monitor.py", line 281, in __init__
    self.console = miniterm.Console()
  File ".local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 124, in __init__
    self.old = termios.tcgetattr(self.fd)
termios.error: (25, 'Inappropriate ioctl for device')
esp/esp-idf/components/esptool_py/Makefile.projbuild:101: recipe for target 'monitor' failed
make: *** [monitor] Error 1
"make -j2 flash monitor" terminated with exit code 2. Build might be incomplete.
```

If I do the same command at the terminal, it will do it without any problems.

projectgus, 2019-02-25:
Hi @colesnicov ,

Thanks for commenting. This looks like a different issue to the one mentioned. Can you please open a new issue for this, with these details plus the additional information that is asked for in the "New Issue" template (this is shown when you open a new issue)?

Angus