espressif/esp-idf #666: panic

dannybr, 2017-06-02:
the same code when compiled with an older version of esp-idf works fine,
but with the resent 2.0 it panics, any idea where to look?, the stack is not very helpful.

[panic.txt](https://github.com/espressif/esp-idf/files/1048716/panic.txt)

negativekelvin, 2017-06-02:
Seems to indicate stack overflow, make your stack bigger?

dannybr, 2017-06-04:
> 
> Seems to indicate stack overflow, make your stack bigger?
did that, but it did not help,
out of a hunch, I removed many of the ESP_LOGx, that were
printing much debugging info, and now the program has been running for
several hours without panics, returning the debug prints it panics very quickly

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Tmr Svc) 
Register dump:
PC      : 0x4000bff0  PS      : 0x00060236  A0      : 0x800849c1  A1      : 0x3ffbe360  
A2      : 0x00000000  A3      : 0x00060223  A4      : 0x00060220  A5      : 0x00000041  
A6      : 0x00000001  A7      : 0x00000005  A8      : 0x8000bd86  A9      : 0x3ffbe520  
A10     : 0x00000000  A11     : 0x40082fb0  A12     : 0x3ffbed50  A13     : 0x00000041  
A14     : 0x3ffae314  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x4000bff0:0x3ffbe360 0x400849c1:0x3ffbe430 0x40086533:0x3ffbe450 0x40086680:0x3ffbe490 0x40081d78:0x3ffbe4b0 0x40081e58:0x3ffbe4e0 0x40082fdc:0x3ffbe500 0x400df8c8:0x3ffbe520 0x4000bd86:0x3ffbe540 0x40001180:0x3ffbe560 0x40059301:0x3ffbe580 0x4005937d:0x3ffbe5a0 0x40058bc2:0x3ffbe5c0 0x400d5ff4:0x3ffbe5f0 0x400df361:0x3ffbe620 0x400d81e4:0x3ffbe930 0x40081c84:0x3ffbe960 0x400e1f30:0x3ffbe9b0 0x400e1ffb:0x3ffbea20 0x4011d9bc:0x3ffbea40 0x400e162b:0x3ffbea60 0x40087071:0x3ffbea90 0x400870a4:0x3ffbeab0 0x400871af:0x3ffbeae0

0x4000bff0: ?? ??:0
0x400849c1: xTaskGetSchedulerState at /vol/src/esp32/esp-idf/components/freertos/./tasks.c:3439
0x40086533: xQueueGenericReceive at /vol/src/esp32/esp-idf/components/freertos/./queue.c:2034
0x40086680: xQueueTakeMutexRecursive at /vol/src/esp32/esp-idf/components/freertos/./queue.c:2034
0x40081d78: lock_acquire_generic at /vol/src/esp32/esp-idf/components/newlib/./locks.c:147
0x40081e58: _lock_acquire_recursive at /vol/src/esp32/esp-idf/components/newlib/./locks.c:161
0x40082fdc: uart_write at /vol/src/esp32/esp-idf/components/vfs/./vfs_uart.c:61
0x400df8c8: esp_vfs_write at /vol/src/esp32/esp-idf/components/vfs/./vfs.c:217 (discriminator 4)
0x4000bd86: ?? ??:0
0x40001180: ?? ??:0
0x40059301: ?? ??:0
0x4005937d: ?? ??:0
0x40058bc2: ?? ??:0
0x400d5ff4: __sprint_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:439
0x400df361: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1782 (discriminator 1)
0x400d81e4: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:40
0x40081c84: esp_log_write at /vol/src/esp32/esp-idf/components/log/./log.c:107
0x400e1f30: pcd_transfer at /vol/src/esp32/work/elockc/main/./mfrc522.c:83
0x400e1ffb: pcd_status at /vol/src/esp32/work/elockc/main/./mfrc522.c:321
0x4011d9bc: mfrcStatus at /vol/src/esp32/work/elockc/main/./mfrc.c:324
0x400e162b: hbCallBack at /vol/src/esp32/work/elockc/main/./elock_task.c:196
0x40087071: prvProcessExpiredTimer at /vol/src/esp32/esp-idf/components/freertos/./timers.c:839
0x400870a4: prvProcessTimerOrBlockTask at /vol/src/esp32/esp-idf/components/freertos/./timers.c:839
0x400871af: prvTimerTask at /vol/src/esp32/esp-idf/components/freertos/./timers.c:839

igrr, 2017-06-04:
`Debug exception reason: Stack canary watchpoint triggered (Tmr Svc)` in your log certainly means that this is a stack overflow in the timer task. Increasing timer task stack size (menuconfig: component config > freertos >  freertos timer task stack size) should help.

dannybr, 2017-06-05:
> 
> Debug exception reason: Stack canary watchpoint triggered (Tmr Svc) in your log certainly means that this is a stack overflow in the timer task. Increasing timer task stack size (menuconfig: component config > freertos > freertos timer task stack size) should help.
I forgot about that one! thanks, that fixed it.

closed 2017-06-05 by igrr