espressif/esp-idf #668: Panic after wifi initialization

Sermus, 2017-06-03:
Hello,
I added wifi initialization to my code based on IDF 6873c19 and got this panic:

I (2650) wifi: wifi firmware version: 8a1ad86
I (2650) wifi: config NVS flash: enabled
I (2650) wifi: config nano formating: disabled
I (2670) wifi: Init dynamic tx buffer num: 32
I (2670) wifi: Init dynamic rx buffer num: 64
I (2670) wifi: wifi driver task: 3ffd5490, prio:23, stack:4096
I (2670) wifi: Init static rx buffer num: 10
I (2680) wifi: Init dynamic rx buffer num: 0
I (2680) wifi: Init rx ampdu len mblock:7
I (2680) wifi: Init lldesc rx ampdu entry mblock:4
I (2690) wifi: wifi power manager task: 0x3ffda85c prio: 21 stack: 2560
I (2690) wifi: wifi timer task: 3ffdb8e0, prio:22, stack:3584
I (2740) wifi: mode : sta (18:fe:34:6a:b0:e2)
/home/andrey/iotgizmo_thermostat/espressif_3rdparty/esp32/esp-idf/components/freertos/./queue.c:715 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008450f on core 0
0x4008450f: xQueueGenericSend at /home/andrey/iotgizmo_thermostat/espressif_3rdparty/esp32/esp-idf/components/freertos/./queue.c:2034

Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffd52a0 0x40007d16:0x3ffd52c0 0x400f8c55:0x3ffd5300 0x400f9c0a:0x3ffd5320 0x400fe55d:0x3ffd5340 0x400ffcac:0x3ffd5390 0x4010028e:0x3ffd53c0 0x4010f732:0x3ffd53e0
0x400f8c55: esp_event_send at /home/andrey/iotgizmo_thermostat/espressif_3rdparty/esp32/esp-idf/components/esp32/./event_loop.c:76

0x400f9c0a: wifi_send_event at ??:?

0x400fe55d: wifi_restart_process at ??:?

0x400ffcac: wifi_start_process at ??:?

0x4010028e: ieee80211_ioctl_process at ??:?

0x4010f732: ppTask at ??:?

Any ideas what can cause that?
I'm attaching sdkconfig i used just in case it contains anything useful for the analysis.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/1049390/sdkconfig.txt)

Sermus, 2017-06-03:
Answering my own question. Not registering system event handler through esp_event_loop_init causes this. Is it required? If so can you add a more specific diagnostic message?

FHFS, 2017-06-03:
You can close it, you answered it yourself.
As you can see events will be posted by wifi to a non existing Queue 
From esp-idf/components/esp32/event_loop.c
```
esp_err_t` esp_event_loop_init(system_event_cb_t cb, void *ctx)
{
    if (s_event_init_flag) {
        return ESP_FAIL;
    }
    s_event_handler_cb = cb;
    s_event_ctx = ctx;
    s_event_queue = xQueueCreate(CONFIG_SYSTEM_EVENT_QUEUE_SIZE, sizeof(system_event_t));

    xTaskCreatePinnedToCore(esp_event_loop_task, "eventTask",
            ESP_TASKD_EVENT_STACK, NULL, ESP_TASKD_EVENT_PRIO, NULL, 0);

    s_event_init_flag = true;
    return ESP_OK;
}
```

Sermus, 2017-06-04:
Yes, i could close, but the point was doing an extra check in wifi implementation internals to let dev know if he/she didn't do necessary initializations (i.e. didn't assign the event callback). At the same time i may not need the sys event callback at all and may expect wifi initialization to work even without informing me through callback.

igrr, 2017-06-04:
When you call `esp_wifi_init` with `WIFI_INIT_CONFIG_DEFAULT`, you are passing a pointer to the default event processing function, `esp_event_send`, to the WiFi driver. For this reason, event loop must be initialized. You may very well not use a callback, in this case you can pass two NULL arguments to `esp_event_loop_init` function. Good point about adding an explicit check to `esp_event_send` and failing via `abort` inside `esp_event_send` instead of `xQueueGenericSend`.

FayeY, 2017-06-09:
Hi @Sermus , is this problem solved? Can we close this issue?

Sermus, 2017-06-09:
I dunno. The Ivan's explanation is clear and this is not an issue for me anymore. But as Ivan said, it would be good to add an explicit check and a bit more specific debug message for the situation i faced.
If your internal dev process allows doing this after closing the github ticket, then i'm fine with this.

igrr, 2017-06-09:
I have a PR which includes an explicit check for this in pipeline. Let's keep this open for now.

closed 2017-06-19 by igrr