espressif/esp-idf #672: BLE connection parameter set/update

frax84, 2017-06-05:
I'm trying to understand how connection parameters (min/max connection interval, latency and timeout) are set when a BLE connection between gatt server and client is (going) on.
1) I didn't find any call to these parameters. Are the parameters set by default? Are the parameters suggested by the client ignored?
2) I think that gatt server event handler is missing the case "CONNECTION PARAMETERS UPDATE REQUEST". Am i right? It will be implemented?

Thank you,
frax

weijianzhi, 2017-06-07:
@frax84 , this bug has been fixed in the latest idf sdk, you can update the sdk and try . In the latest gatts demo, gatt server will set the connection parameters when a BLE connection between gatt server and client is  on. ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT in gap_event_handler callback function will get the result.  gatts_demo.c has a detailed code.

frax84, 2017-06-07:
Awesome! Thank you as usual for the support!

Weijian-Espressif, 2017-06-08:
@frax84   If solved the problem, please let me know. Thanks.

frax84, 2017-06-13:
@Weijian-Espressif 

Ok, i had the chance to try the updated code you put on for connection parameters. Unfortunately it's not working as intended. 

This is what happens:
1) When i press "Connect" on my BLE client (PC side) it sends a connection packet with some connection parameters in it, as BLE protocol says to happen
2) On gatt server side (esp side) the GAP detect a connection event and triggers the callback on CONNECT_EVT that runs smoothly 
3) Considering how the code is now the connection parameters included in the exchange between gatt and client during connection establishment are completely ignored. In fact the UPDATE_CONN_PARAMS_EVT is triggered only by the update function included in the CONNECT_EVT code. If i comment that line nothing happens.

This is my console output when i don't comment che update params function:
I (17332) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 88:6b:0f:19:ed:da:, is_conn 1
I (17332) GATTS_DEMO: SERVICE_START_EVT, conn_id 0, remote 88:6b:0f:19:ed:da:, is_conn 1
I (17572) GATTS_DEMO: update connetion params status = 13, min_int = 30, max_int = 50,                  conn_int = 0,latency = 0, timeout = 400
I (17572) GATTS_DEMO: update connetion params status = 0, min_int = 30, max_int = 50,                  conn_int = 0,latency = 0, timeout = 400

As you can notice the connection parameters are the ones you define in CONNECT_EVT code. If i comment the update_params function my console output is just this one:

I (17332) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 88:6b:0f:19:ed:da:, is_conn 1
I (17332) GATTS_DEMO: SERVICE_START_EVT, conn_id 0, remote 88:6b:0f:19:ed:da:, is_conn 1

I would like to add a couple of things:
1) It would be nice having a "get method" for connection parameters, something like: "esp_ble_gap_get_conn_params"
2) There is one more try i need to investigate but it will need me a bit more of time: trying to send a pure "update connection parameters" BLE package to dynamically chance the parameters with connection already on and not during its ESTABLISHMENT.

Waiting for your feedback,
frax84

Weijian-Espressif, 2017-06-14:
@frax84  
esp_ble_gap_update_conn_params()  register the ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT callback function, so your console output is right. we don't support similar"esp_ble_gap_get_conn_params" function so far.

frax84, 2017-06-14:
@Weijian-Espressif 
i'm not sure to have well understood. I'll try formulating direct questions:
I send the connection parameters that my connection should have during the connection initialization. From what i can see the ESP seems to ignore these parameters. Maybe i'm a bit confused on the flow of events but imho an update conn param event should trigger just before the connection event because the negotiation of the parameters happens just before the connection. Is not what happens? Are the initial parameters negotiated/managed in another way than UPDATE_PARAM_EVENT?

I'm going to give a shot to the SmartRF Packet Sniffer. Maybe it will make things more clear.
Thanks

Weijian-Espressif, 2017-06-15:
hi @frax84 ,  in btm_ble_conn_complete() function of btm_ble.c,  you can add some log .

void btm_ble_conn_complete(UINT8 *p, UINT16 evt_len, BOOLEAN enhanced)
{
#if (BLE_PRIVACY_SPT == TRUE )
    UINT8       *p_data = p, peer_addr_type;
#endif  ///BLE_PRIVACY_SPT == TRUE
    UINT8       role, status, bda_type;
    UINT16      handle;
    BD_ADDR     bda;
#if (BLE_PRIVACY_SPT == TRUE)
    BD_ADDR     local_rpa, peer_rpa;
#endif  ///BLE_PRIVACY_SPT == TRUE
    UINT16      conn_interval, conn_latency, conn_timeout;
    BOOLEAN     match = FALSE;
    UNUSED(evt_len);

    STREAM_TO_UINT8   (status, p);
    STREAM_TO_UINT16   (handle, p);
    STREAM_TO_UINT8    (role, p);
    STREAM_TO_UINT8    (bda_type, p);
    STREAM_TO_BDADDR   (bda, p);
    BTM_TRACE_DEBUG("status = %d, handle = %d, role = %d, bda_type = %d",status,handle,role,bda_type);
    if (status == 0) {
#if (BLE_PRIVACY_SPT == TRUE )
        peer_addr_type = bda_type;
        match = btm_identity_addr_to_random_pseudo (bda, &bda_type, TRUE);

        if (enhanced) {
            STREAM_TO_BDADDR   (local_rpa, p);
            STREAM_TO_BDADDR   (peer_rpa, p);
        }

        /* possiblly receive connection complete with resolvable random on
           slave role while the device has been paired */
        if (!match && role == HCI_ROLE_SLAVE && BTM_BLE_IS_RESOLVE_BDA(bda)) {
            btm_ble_resolve_random_addr(bda, btm_ble_resolve_random_addr_on_conn_cmpl, p_data);
        } else
#endif
        {
            STREAM_TO_UINT16   (conn_interval, p);
            STREAM_TO_UINT16   (conn_latency, p);
            STREAM_TO_UINT16   (conn_timeout, p);
           // ===========add some log here ================//

          printf("conn_interval = %d, conn_latency = %d, conn_timeout = %d\n", conn_interval, conn_latency, conn_timeout);

            handle = HCID_GET_HANDLE (handle);

            btm_ble_connected(bda, handle, HCI_ENCRYPT_MODE_DISABLED, role, bda_type, match);

            l2cble_conn_comp (handle, role, bda, bda_type, conn_interval,
                              conn_latency, conn_timeout);

#if (BLE_PRIVACY_SPT == TRUE)
            if (enhanced) {
                btm_ble_refresh_local_resolvable_private_addr(bda, local_rpa);

                if (peer_addr_type & BLE_ADDR_TYPE_ID_BIT) {
                    btm_ble_refresh_peer_resolvable_private_addr(bda, peer_rpa, BLE_ADDR_RANDOM);
                }
            }
#endif
        }
    } else {
        role = HCI_ROLE_UNKNOWN;
        if (status != HCI_ERR_DIRECTED_ADVERTISING_TIMEOUT) {
            btm_ble_set_conn_st(BLE_CONN_IDLE);
#if (defined BLE_PRIVACY_SPT && BLE_PRIVACY_SPT == TRUE)
            btm_ble_disable_resolving_list(BTM_BLE_RL_INIT, TRUE);
#endif
        } else {
#if (defined BLE_PRIVACY_SPT && BLE_PRIVACY_SPT == TRUE)
            btm_cb.ble_ctr_cb.inq_var.adv_mode  = BTM_BLE_ADV_DISABLE;
            btm_ble_disable_resolving_list(BTM_BLE_RL_ADV, TRUE);
#endif
        }
    }

    btm_ble_update_mode_operation(role, bda, status);
}

Weijian-Espressif, 2017-06-19:
@frax84  do you get a shot to the the SmartRF Packet Sniffer? I have used Lightblue(ios app) connect to esp32 server,  esp32 update connect parameter correctly that lightblue send.

frax84, 2017-06-19:
@Weijian-Espressif I downloaded the software but i noticed i need a specific hardware for sniffing. Looking for that on some reseller.
By the way, just to understand, do you connect with standard parameters and then send an update conn params request or do you connect with a specific conn params?
Just answer to this and i suppose we can close the issue.
Thanks for the support

Weijian-Espressif, 2017-06-19:
I did two tests:
1. iPhone app connect to esp32 , the connect parameter is set by phone .
2. after then, esp32 update connect parameters with  a specific conn params.

frax84, 2017-06-19:
@Weijian-Espressif 
Ok, i suppose this should be enough to close the issue.
Thank you

closed 2017-06-19 by frax84