espressif/esp-idf #673: uart_set_line_inverse() can not set mask back to 0

zelll, 2017-06-06:
As mentioned in https://github.com/espressif/esp-idf/pull/489 , `uart_set_line_inverse()` does not allow a `0` mask. Then there is not a way to clear line inverse. Is it a bug?

> UART_CHECK((((inverse_mask & ~UART_LINE_INV_MASK) == 0) && **(inverse_mask != 0)**), "inverse_mask error", ESP_FAIL);

hwmaier, 2017-06-06:
What would be the use case for having it to clear again?

I would have thought for a given hardware you either set it or don't depending on your physical layer hardware but never have to change it after you done your hardware initialisation.

negativekelvin, 2017-06-06:
Possibly if you were talking on a multidrop rs485 bus and switching between different phy settings?

hwmaier, 2017-06-06:
Ok, if you have to change phys during run-time, then it is useful.

Looking at the code it is intended to work, and  uart_set_line_inverse(port, 0) does work and clears it.

zelll, 2017-06-06:
I'm adding IDF uart driver to nodemcu. I think someone may want to re-configure uart. Maybe a 485 bus, may be uart controlled by switches...

@hwmaier I added some logs to uart_set_line_inverse(), it does return before clearing the register.
```
    ESP_LOGE(UART_TAG, "### 1");
    UART_CHECK((((inverse_mask & ~UART_LINE_INV_MASK) == 0) && (inverse_mask != 0)), "inverse_mask error", ESP_FAIL);
    ESP_LOGE(UART_TAG, "### 2");
```
outputs:
> [0;31mE (2952) uart: ### 1[0m
[0;31mE (2952) uart: uart_set_line_inverse(195): inverse_mask error[0m
[0;32mI (2992) uart: queue free spaces: 3[0m

hwmaier, 2017-06-06:
Maybe we are using different versions of code? uart_set_line_inverse(UART_NUM_1, 0) does work in head revision and not throw the inverse_mask error assertion. What value are you passing to clear? It should be a zero to clear.

hwmaier, 2017-06-06:
You are right, this appears to be a bug. I think the logic should be:

     UART_CHECK((((inverse_mask & ~UART_LINE_INV_MASK) == 0) || (inverse_mask == 0)), "ixnverse_mask error", ESP_FAIL);

negativekelvin, 2017-06-06:
Isn't that redundant though because if inverse_mask=0 the first test will be true

hwmaier, 2017-06-06:
Yes, so the test could be even more simplified.

FayeY, 2017-08-01:
Hi @zelll , is this problem solved now? Can we close this issue?

closed 2017-08-03 by zelll