espressif/esp-idf #678: Documentation of RS-485 related features

EndlessDelirium, 2017-06-08:
Earlier versions of the datasheet state that there is integrated RS-485 support:
> ESP32 has three UART interfaces, i.e. UART0, UART1 and UART2, which provide asynchronous communication (RS232 and RS485) and IrDA support, and communicate at up to 5 Mbps.

The current technical reference manual (v1.7) still mentions RS-485 support on page 189:
> Supports RS485 Protocol

There is also a very brief documentation of RS-485 related registers.

Due to the very limited documentation, unfortunately no one seems to have managed to successfully use these functionalities yet. Better documentation and an example would therefore be highly appreciated.

This topic is also discussed in the ESP32 forums: https://esp32.com/viewtopic.php?f=13&t=1858

@hwmaier was so kind to provide a software based implementation for the most imminent issue of automated enabling/disabling of the transceiver: https://github.com/espressif/esp-idf/pull/667

negativekelvin, 2017-06-22:
Also need these docs

tusharvb19, 2017-06-27:
I also need the docs and sample code..help me
TIA.

hrgdavor, 2017-07-07:
RS485 support looks like a very nice addition. subscribed to changes on this. Interested to try it when it works :)

voxadam, 2017-08-26:
Proper RS-485 support would be incredibly useful.

bookerdong, 2017-12-08:
Proper RS-485 support would be incredibly useful.

casmeiron, 2017-12-22:
yeah, that would be a gr8 feature to have!

igrr, 2018-08-23:
RS485 support is included since https://github.com/espressif/esp-idf/commit/909eec9c3b174a080e8fed10c0315d43ded5dfd5.

closed 2018-08-23 by igrr