espressif/esp-idf #679: wifi disconnect function - deleting many times

SirRibeiro, 2017-06-08:
Hi,
I'm using raspberry pi as AP for esp32, and i was checking the events generated on the raspberry wifi module using the command 'iw event -r'. This is the output:

**5.899425: wlan0: del station 24:0a:c4:03:53:b8
0.000870: wlan0: del station 24:0a:c4:03:53:b8
0.000583: wlan0: del station 24:0a:c4:03:53:b8
0.000428: wlan0: del station 24:0a:c4:03:53:b8
0.000482: wlan0: del station 24:0a:c4:03:53:b8
0.000573: wlan0: del station 24:0a:c4:03:53:b8
0.000333: wlan0: del station 24:0a:c4:03:53:b8
0.000425: wlan0: del station 24:0a:c4:03:53:b8
0.006560: wlan0: del station 24:0a:c4:03:53:b8
0.000894: wlan0: del station 24:0a:c4:03:53:b8
0.000475: wlan0: del station 24:0a:c4:03:53:b8
9.222153: wlan0: new station 24:0a:c4:03:53:b8**

My function is: 
`void disconnet()
{
    esp_wifi_stop();
    esp_wifi_deinit();
    esp_wifi_disconnect();
    esp_restart();
}`
I've tried many combinations of this code, and it's mostly the same result.

After the restart it does the standard wifi connect used in the basic examples.

When i dissociate using for example my phone or my computer this is the output:
**0.693711: wlan0: del station b4:9d:0b:1d:64:9c
0.001025: wlan0: del station b4:9d:0b:1d:64:9c
11.567075: wlan0: new station b4:9d:0b:1d:64:9c**

I'm doing something wrong? Is there a way to solve this? 
Thanks

liuzfesp, 2017-06-09:
Ｈｉ @SirRibeiro could you provide the whole WIFI event handling code?  Then I can know what happens when disconnect the connections.
BTW, if esp_restart is called, then all other APIs, esp_wifi_stop(); esp_wifi_deinit(); esp_wifi_disconnect(); are not necessary because esp_restart will call esp_wifi_stop and esp_wifi_stop will disconnect all connections internally.  If you really want to disconnect the WIFI yourself, the right sequence should be esp_wifi_disconnect, esp_wifi_stop, esp_wifi_deinit.

SirRibeiro, 2017-06-09:
Thanks for the response @liuzfesp,

I didn't knew that the esp_restart() was already doing that sequence. 

Perfect this way i don't have to worry about that.

So my event_handler code is:

```cpp

char MyAddr[20] = { 0 };

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
		sprintf(MyAddr,"%d.%d.%d.%d",IP2STR(&event->event_info.got_ip.ip_info.ip));
		printf("\n\nMy Ip is: %s\n\n",MyAddr);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}
```
And my initialise_wifi:
```cpp

struct addrinfo hints;
static EventGroupHandle_t wifi_event_group;
String network = "SSID";
String network_pass = "SSID_PWD";
char macStr[18] = { 0 };

static void initialise_wifi(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );

    wifi_config_t wifi_config;

    strcpy(reinterpret_cast<char*>(wifi_config.sta.ssid), network.c_str());
    strcpy(reinterpret_cast<char*>(wifi_config.sta.password), network_pass.c_str());

    printf("Setting WiFi configuration SSID: %s\n", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );

    uint8_t mac[6];
    esp_wifi_get_mac(WIFI_IF_STA, mac);
    sprintf(macStr, "%02X:%02X:%02X:%02X:%02X:%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    printf("\n\n\n\n\n\nMy Mac is: %s\n\n\n\n\n",macStr);

    memset(&hints, 0, sizeof(hints));
    hints.ai_family   = AF_INET;
    hints.ai_socktype = SOCK_STREAM;
}
```

Thanks

FayeY, 2017-11-21:
Hi, sorry that we cannot reproduce your problem, if your problem is still unsolved on the latest IDF, please feel free to reopen it.

closed 2017-11-21 by FayeY