espressif/esp-idf #680: i2c timeout error

tdesmet, 2017-06-08:
I'm polling a i2c device every half second and after a while (sometimes 4 minutes, sometimes half an hour). The i2c_master_cmd_begin function returns ESP_ERR_TIMEOUT, and every next call to i2c_master_cmd_begin returns ESP_ERR_TIMEOUT.

I tried deleting the driver using i2c_driver_delete and then i2c_driver_install but it keeps giving the error.

At this time the SDA line remains low and the SCL line remains high. I also tried to delete the driver and then change the pins to output and change to no avail.

The device I am communicating with is a battery monitor (LC709203F). But I have the same problem with an accelerometer.

Curclamas, 2017-06-10:
Maybe this could be related to https://github.com/espressif/arduino-esp32/issues/349 . I've noticed similar things when I poll some I2C-GPIO-Expander every 20ms. At some point the request will either always fail or just lock up.

tdesmet, 2017-06-13:
debugging a bit more it seems something locks up the lines. I tried disconnecting the device but the lines remain in their locked state.

This is really frustrating as the only solution seems to be to reset the device when this happens. But that means data loss since we use the esp32 to measure sensors continuously and report this data to the cloud.
Well we can continue our measurements but we will be missing all info we get via i2c (battery and accelerometer at this point).

Curclamas, 2017-06-19:
Also this kind of problem seams to be noticed in the forum:
- https://esp32.com/viewtopic.php?f=2&t=2181&p=10333#p10333
- https://esp32.com/viewtopic.php?f=2&t=2178&p=10265&hilit=i2c#p10265
I would guess this more widespread than one may think.

tbnobody, 2017-08-13:
I think I have the same problem but with a DS3231 / DS1307. The interesting thing is, that I have exactly your behavior using the ESP-IDF. But if I use the ESP-Arduino SDK (as a component in IDF) it works without any problems.

Sermus, 2017-09-11:
+1.
I experience this with ft6206 touch controller and mvh3002d temperature/humidity sensor.

costaud, 2017-09-13:
Weird, is there a soft reboot during reading the I2C slave?

costaud, 2017-09-13:
As far as I know, the device will only get stuck when the slave hold the bus wires and get the bus into a wrong state. Would you please post some of your reading code here? @Sermus @Curclamas @tdesmet

Sermus, 2017-09-13:
Nop, there is no soft reboot (until i force it)

chegewara, 2017-10-08:
I have no problem with I2C and MPU6050, even i can polling every 20ms, but im having problem with mcp23017.

costaud, 2017-10-18:
The patch will be merged into master branch recently

Curclamas, 2017-10-18:
@costaud  how did you fix it? Can we already see the fix on master?

costaud, 2017-10-18:
The state machine of I2C bus could be dead if the bus is influenced in some circumstance(for example, short the sda/scl to gnd for a while). The hardware would get stuck and be always sending clock on scl or only sending a single pulse on scl. We can only reset the I2C hardware to get the hardware back working.

To reset the hardware, we can call periph_module_disable then periph_module_enable, but all the register settings will be lost.  We will try to merge this patch ASAP. In the next version of chips, we will find some way to void this situation.

closed 2017-11-21 by FayeY

AmrutaCh, 2019-04-22:
I am facing the same issue with I2C example provided in idf. 
ESP32 Arduino works as expected whereas esp-idf gives the following errors

- timeout : 0x107 (generated by I2C_STATUS_TIMEOUT)
-  failed: -1 (generated by I2C_STATUS_ACK_ERROR)

The slave is an Arduino device. 

ESP-idf version: v3.1
Code:

```

#include <stdio.h>
#include "driver/i2c.h"

#define I2C_EXAMPLE_MASTER_SDA_IO 22           /*!< gpio number for I2C master data  */
#define I2C_EXAMPLE_MASTER_SCL_IO 23           /*!< gpio number for I2C master clock */
#define I2C_EXAMPLE_MASTER_NUM I2C_NUM_0       /*!< I2C port number for master dev */
#define I2C_EXAMPLE_MASTER_TX_BUF_DISABLE 0    /*!< I2C master do not need buffer */
#define I2C_EXAMPLE_MASTER_RX_BUF_DISABLE 0    /*!< I2C master do not need buffer */
#define I2C_EXAMPLE_MASTER_FREQ_HZ (10 * 1000) /*!< I2C master clock frequency */

#define DATA_LENGTH 1   //18             /*!<Data buffer length for test buffer*/
#define I2C_SLAVE_ADDR (8)         /*!< ESP32 slave address, you can set any 7bit value */
#define WRITE_BIT I2C_MASTER_WRITE /*!< I2C master write */
#define READ_BIT I2C_MASTER_READ   /*!< I2C master read */
#define ACK_CHECK_EN 0x1           /*!< I2C master will check ack from slave*/
#define ACK_CHECK_DIS 0x0          /*!< I2C master will not check ack from slave */
#define ACK_VAL 0x0                /*!< I2C ack value */
#define NACK_VAL 0x1               /*!< I2C nack value */

static esp_err_t i2c_example_master_read_slave(i2c_port_t i2c_num, uint8_t *data_rd, size_t size)
{
    esp_err_t err = ESP_OK;

    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (I2C_SLAVE_ADDR << 1) | READ_BIT, ACK_CHECK_EN);
    if (size > 1)
    {
        i2c_master_read(cmd, data_rd, size - 1, ACK_VAL);
    }
    i2c_master_read_byte(cmd, data_rd + size - 1, NACK_VAL);
    i2c_master_stop(cmd);
    err = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);

    return err;
}

static void i2c_example_master_init()
{
    int i2c_master_port = I2C_EXAMPLE_MASTER_NUM;
    i2c_config_t conf;
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = I2C_EXAMPLE_MASTER_SDA_IO;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_io_num = I2C_EXAMPLE_MASTER_SCL_IO;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = I2C_EXAMPLE_MASTER_FREQ_HZ;
    esp_err_t err = i2c_param_config(i2c_master_port, &conf);
    if (ESP_OK == err)
    {
        err = i2c_driver_install(i2c_master_port, conf.mode,
                                 I2C_EXAMPLE_MASTER_RX_BUF_DISABLE,
                                 I2C_EXAMPLE_MASTER_TX_BUF_DISABLE, 0);
        printf("i2c_driver_install 0x%x \n", err);
    }
    else
    {
        printf("i2c_param_config fail 0x%x \n", err);
    }
}

static void i2c_test_task(void *arg)
{
    uint8_t data_rd[DATA_LENGTH] = {0};
    int cnt = 0;
    esp_err_t ret = 0;

    while (1)
    {
        printf("cnt: %d\n", cnt++);

        vTaskDelay(100 / portTICK_RATE_MS);

        ret = i2c_example_master_read_slave(I2C_EXAMPLE_MASTER_NUM, data_rd, DATA_LENGTH);

        if (ESP_OK == ret)
        {
            data_rd[DATA_LENGTH] = 0;
            printf("Data: %s \n", data_rd);
        }
        else
        {
            printf("err: 0x%x %s \n", ret, esp_err_to_name(ret));
        }
    }
}

void app_main()
{
    i2c_example_master_init();
    xTaskCreate(i2c_test_task, "i2c_test_task_1", 1024 * 5, (void *)1, 10, NULL);
}

```

arsinio, 2019-04-22:
I can confirm this error still exists...I Incorporated the one-line
fix/cludge into a local fork of the esp-idf repo and keep track of it that
way.

AmrutaCh, 2019-04-22:
@arsinio can you share the changes which are working for you?
It would of great help.

AmrutaCh, 2019-04-24:
> I am facing the same issue with I2C example provided in idf.
> ESP32 Arduino works as expected whereas esp-idf gives the following errors
> 
>     * timeout : 0x107 (generated by I2C_STATUS_TIMEOUT)
> 
>     * failed: -1 (generated by I2C_STATUS_ACK_ERROR)
> 
> 
> The slave is an Arduino device.
> 
> ESP-idf version: v3.1
> Code:
> 
> init
> 
> ```
>  int i2c_master_port = I2C_EXAMPLE_MASTER_NUM;
>     i2c_config_t conf;
>     conf.mode = I2C_MODE_MASTER;
>     conf.sda_io_num = I2C_EXAMPLE_MASTER_SDA_IO;
>     conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
>     conf.scl_io_num = I2C_EXAMPLE_MASTER_SCL_IO;
>     conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
>     conf.master.clk_speed = I2C_EXAMPLE_MASTER_FREQ_HZ;
>     esp_err_t err = i2c_param_config(i2c_master_port, &conf);
>     if (ESP_OK == err)
>     {
>         err = i2c_driver_install(i2c_master_port, conf.mode,
>                                  I2C_EXAMPLE_MASTER_RX_BUF_DISABLE,
>                                  I2C_EXAMPLE_MASTER_TX_BUF_DISABLE, 0);
>         printf("i2c_driver_install 0x%x \n", err);
>     }
>     else
>     {
>         printf("i2c_param_config fail 0x%x \n", err);
>     }
> ```
> 
> read data from slave
> 
> ```
>     i2c_cmd_handle_t cmd = i2c_cmd_link_create();
>     i2c_master_start(cmd);
>     i2c_master_write_byte(cmd, (I2C_SLAVE_ADDR << 1) | READ_BIT, ACK_CHECK_EN);
>     if (size > 1)
>     {
>         i2c_master_read(cmd, data_rd, size - 1, ACK_VAL);
>     }
>     i2c_master_read_byte(cmd, data_rd + size - 1, NACK_VAL);
>     i2c_master_stop(cmd);
>     err = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_RATE_MS);
>     i2c_cmd_link_delete(cmd);
> ```

More details:

On Arduino slave side, it gives `TW_BUS_ERROR`

arsinio, 2019-04-24:
@AmrutaCh 
You'll need to modify `components/driver/i2c.c`...I'm not up-to-date with the latest version so I'll just walk you through the change:

In `esp_err_t i2c_param_config(i2c_port_t i2c_num, const i2c_config_t* i2c_conf)` there is an if block...something like:

```
if (i2c_conf->mode == I2C_MODE_SLAVE) {  //slave mode
...    
} else {
...        
}
```

You'll need to modify the `else` clause changing
```
I2C[i2c_num]->timeout.tout = cycle * I2C_MASTER_TOUT_CNUM_DEFAULT;
```

To
```
I2C[i2c_num]->timeout.tout = 1048575;
```

That number was picked because, to the best of my knowledge, it's the largest supported by the hardware.

That being said there is a "newer" function that may help:
`i2c_set_timeout`

When first introduced I found that it did not resolve my issue, but I should really give it another try.

AmrutaCh, 2019-04-26:
@arsinio 
Sadly, it didn't work for me.
Setting the value to 0xfffff (=I2C_TIME_OUT_REG_V) resulted in salve being able to write 2 bytes.
Previously it was not able to write it at all

AmrutaCh, 2019-04-26:
```i2c_set_timeout``` worked 

Thanks @arsinio

saeidimajid, 2023-06-01:
Hi

I tried to read SHT20 registers with I2C tool with "i2cget" command, but it has error
"ESP_ERR_TIMEOUT Operation timeout because the bus is busy"
esp_err_t = 0x107
what does it mean?
Maybe it depends on the timout setting. (i2c_set_timeout)
I trid to set timeout in multi value but it doesn't work

IvanAkiy, 2024-04-15:
> Hi
>
> I tried to read SHT20 registers with I2C tool with "i2cget" command, but it has error "ESP_ERR_TIMEOUT Operation timeout because the bus is busy" esp_err_t = 0x107 what does it mean? Maybe it [...]

Hi
Wondering if you solved this issue, because I'm facing the same problem with esp32-s3-pico and pn532. If yes, can you provide the solution that worked for you?

saeidimajid, 2024-04-18:
> Hi Wondering if you solved this issue, because I'm facing the same problem with esp32-s3-pico and pn532. If yes, can you provide the solution that worked for you?

I couldn't solve this problem, but with a program that I wrote myself based on the SHT20 datasheet, it worked easily. I also used a DS1307 module with the i2c tool, which worked properly as well. I was able to read and write to the registers perfectly. I think in some i2c devices, there's a need for a short response time for the device to respond and send data, which hasn't been considered in the i2c tool code.

yallico, 2024-07-17:
Also having the same issue in v5.2.2, having said that I'm using the legacy driver I think. I will git it a go with the new driver in i2c_master.c

Mem21, 2024-10-21:
> Also having the same issue in v5.2.2, having said that I'm using the legacy driver I think. I will git it a go with the new driver in i2c_master.c

Have you been succesfull with the new driver?

I am having ESP_ERR_TIMEOUT with the legacy I2C driver in the ESP32S3 with idf 5.3.1. 
The exat same code runs without any problem on esp32 with espidf 4.X.

yallico, 2024-10-23:
I defaulted to using v5.1.4, (legacy driver) hybrid approach. Port 0 on IDF and port 1 in Arduino wire function!