espressif/esp-idf #683: Component lwip looks having a bug

rillhu, 2017-06-09:
Now the esp-idf component lwip supports both ipv6 and ipv4. However I find a potential issue in function lwip_getaddrinfo() of netdb.c(located at esp-idf/components/lwip/api/netdb.c)

If you look at below code, you will find that it uses IPV6 address structure sockaddr_storage. 

```
if (nodename != NULL) {
    /* copy nodename to canonname if specified */
    ai->ai_canonname = ((char*)ai + sizeof(struct addrinfo) + sizeof(struct sockaddr_storage));
    MEMCPY(ai->ai_canonname, nodename, namelen);
    ai->ai_canonname[namelen] = 0;
  }
  ai->ai_addrlen = sizeof(struct sockaddr_storage);
```

sockaddr_storage is define as below:

```
struct sockaddr_storage {
  u8_t        s2_len;
  sa_family_t ss_family;
  char        s2_data1[2];
  u32_t       s2_data2[3];
#if LWIP_IPV6
  u32_t       s2_data3[3];
#endif /* LWIP_IPV6 */
};

```

And we can see that current code definse LWIP_IPV6 to 0 in esp-idf/components/lwip/include/lwip/lwip/opt.h, but when we call function lwip_getaddrinfo(), it still returns IPV6 adress length. I mean I get a address length with 28 instead of 16.

**Is this a bug here?**

Currenly I add some code as a workaround to avoid such issue. I put it here. But I strongly suggest we dig into the issue.

```
  if (nodename != NULL) {
    /* copy nodename to canonname if specified */
    if(ai->ai_family ==AF_INET){
        ai->ai_canonname = ((char*)ai + sizeof(struct addrinfo) + sizeof(struct sockaddr_in));
    }else{
    ai->ai_canonname = ((char*)ai + sizeof(struct addrinfo) + sizeof(struct sockaddr_storage));
    }
    MEMCPY(ai->ai_canonname, nodename, namelen);
    ai->ai_canonname[namelen] = 0;
  }
  if(ai->ai_family ==AF_INET){  
    ai->ai_addrlen = sizeof(struct sockaddr_in);  
  }else{
  ai->ai_addrlen = sizeof(struct sockaddr_storage);
  }

```

projectgus, 2017-06-09:
LWIP_IPV6 defaults to disabled in opt.h (which has default LWIP options), but we enable it in lwipopts.h (which contains build-specific options):
https://github.com/espressif/esp-idf/blob/master/components/lwip/include/lwip/port/lwipopts.h#L570

At the moment there isn't a config option in IDF to disable IPv6 support.

liuzfesp, 2017-06-09:
@projectgus is right:)

rillhu, 2017-06-09:
Thanks @projectgus

closed 2017-06-09 by rillhu