espressif/esp-idf #684: Current consumption not varying for different Transmitter powers

Raienryu97, 2017-06-09:
I tried setting different values from 0 to 20 dBm for the Tx of WiFi radio using the setting  
Under make menuconfig > Component config > PHY >  Max WiFi TX Power, flashed a program and kept checking for the current the board is drawing. I've observed that the current consumption is same for all values of Tx Power.

Can someone verify if they're facing the same issue?
Or am I supposed to change something else too?

Thanks

Raienryu97, 2017-07-01:
Bump. Any update?

jack0c, 2018-01-14:
@Raienryu97 Do you select Do phy calibration and store calibration data in NVS?

TimXia, 2018-01-15:
@Raienryu97 What are the values of current consumption that you have measured before and after setting Max WiFi Tx Power?

jack0c, 2018-03-30:
@FayeY maybe we can close this issue.

FayeY, 2018-03-30:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2018-03-30 by FayeY