espressif/esp-idf #687: Failed to enter RAM download mode. Only got 2 byte status response

nccchirag, 2017-06-10:
...
CC hello_world_main.o
AR libmain.a
LD hello-world.elf
esptool.py v2.0-beta3
Flashing binaries to serial port COM4 (app at offset 0x10000)...
esptool.py v2.0-beta3
Connecting......
Uploading stub...

A fatal error occurred: Failed to enter RAM download mode. Only got 2 byte status response.
make: *** [/c/msys32/esp-idf/components/esptool_py/Makefile.projbuild:53: flash] Error 2

**Hardware Setup Details**
ESP_Module_Testboard (ESP32)
CP2104 USB - TTL
Windows 10 x64 with msys32 toolchain

**Boot log**
 ets Jul  5 2015,rst cause:1, boot mode:(13)

	 _stack_sentry: 0x3fffe1d0, 	 __stack: 0x40000000, 
 	 _bss_start: 0x3fffcd48, 	 _bss_end: 0x3fffe1d0, 
 	 _data_start: 0x3fffc000, 	 _data_end: 0x3fffc864

projectgus, 2017-06-13:
Hi @nccchirag ,

Thanks for the detailed issue report. Judging from the boot log you are seeing, your board is an ESP31B (the beta ESP32 boards which were supplied in small numbers to beta testers.) IDF doesn't support this chip, there are significant changes between ESP31B and the production ESP32.

There is an SDK for this chip here: https://github.com/espressif/ESP31_RTOS_SDK - however IDF is significantly different. If you plan to develop for ESP32, then I would suggest buying a new development board and not using the ESP31B.

closed 2017-06-13 by projectgus

nccchirag, 2017-06-14:
Yes it was a beta board. Thank you @projectgus