espressif/esp-idf #688: logging: make monitor vs. stty/cat

mehrvarz, 2017-06-10:
Issues when using _make monitor_ vs. _stty/cat_:

1. Quitting and restarting _make monitor_ seems to reset the dev-board. I find this troublesome. When in logging mode, the way I use my logging tool should NOT influence what is happening on the dev-board in any way. Quitting and restarting _stty/cat_ will never reset the dev-board, which is much better.

2. When using _make monitor_ (or minicom/miniterm), I need to be extra careful not to touch my keyboard in some unintended way. Doing so will send characters to the board, which may affect the board in some (negative) way. If it happens, it usually means that I need to start from scratch with whatever I was doing. When using _stty/cat_, I can enter characters on the keyboard without running any risk: they are not being forwarded to the board. This is much better behavior.

So for me, using _stty/cat_ does solve two bugging issues. But I think the underlying problem really needs to be resolved on the device. The act of logging should be **fully passive** and on the ESP32 this is currently not the case. (I am even able to make the board crash when in Bluetooth mode, by how I go about the logging: https://github.com/espressif/esp-idf/issues/674).

When in logging-mode, the board should fend off all "influence" coming from the logging terminal. At this point in time I do not know where the required code changes need to go. But I would like to find out, if consensus can be achieved on this issue in general.

FHFS, 2017-06-11:
Have a look at [idf_monitor](https://github.com/espressif/esp-idf/blob/master/tools/idf_monitor.py#L296) And lines containing "CTRL_*"

mehrvarz, 2017-06-12:
These ctrl-keys are not a problem. Other keys are. For instance, if I accidentally hit the space key while in _make monitor_ (or the return key, which I actually do quite often), the logging session is immediately dead. What is this good for?

Using stty/cat solves this issue (kind of). But if it was possible to tell the device to ignore all arbitrary inbound traffic (from the logging device), that would be better.

igrr, 2017-06-12:
You can remap the standard input stream in your `app_main` if you don't need standard input:
```
_GLOBAL_REENT->_stdin  = (FILE*) &__sf_fake_stdin;
stdin = (FILE*) &__sf_fake_stdin;
```
After that, stdin of the current task (app_main task) and all new tasks will not use `/dev/uart/n` for input.

mehrvarz, 2017-06-12:
Sounds like a useful idea. But it does not seem to solve the issue. I added the two lines on top of my app_main. But when I hit CR, monitor still grid locks.

And to bring up my other item: why does monitor auto-toggle rts? This seems like a really bad idea. And it is not necessary. (Through, it is useful to have ctrl-r.)

igrr, 2017-06-12:
Could you please share the code example which reproduces the issue (hitting CR locks the monitor)?

The reason why `make monitor` resets the board is because that allows one to observe the application start-up from the very beginning. It is synonymous to 'run the application' rather than 'observe what the application is doing'. You can also use tools like `screen` or even `miniterm.py` if you wish to avoid resetting the board.

mehrvarz, 2017-06-12:
> hitting CR locks the monitor
Happens with /examples/bluetooth/gatt_client (my goto mini-app) Did not change any code, only switched Default baud rate to 921600 baud. gatt_client shows lots of live BLE advertisements, until I hit CR (or space) in monitor.

I also added these two lines to app_main() in gattc_demo.c now, but no change:
    _GLOBAL_REENT->_stdin  = (FILE*) &__sf_fake_stdin;
    stdin = (FILE*) &__sf_fake_stdin;

mehrvarz, 2017-06-14:
I don't think this is specific to GATT client example. Whoever is using _make monitor_ with _their_ app should be able to confirm the issue.

igrr, 2017-06-15:
It turned out to be specific to Bluetooth examples. Bluetooth stack used an interrupt which was also attached to UART0 interrupt source by default, which caused Bluetooth stack to receive stray interrupts when UART0 received a character. We have a fix in pipeline, thank you for the report.

FHFS, 2017-06-15:
So any activity on GPIO0 would generate a interrupt for the Bluetooth stack that could cause a crash?

mehrvarz, 2017-06-20:
I should try to be strong. But this bug makes my whole life miserable.

igrr, 2017-06-20:
This issue has been fixed in https://github.com/espressif/esp-idf/commit/09f7cf45.

closed 2017-06-20 by igrr

mehrvarz, 2017-06-20:
This particular issue "stops working when CR is pressed on the console" is indeed solved now. Thank you.

However, other, similar UART+Bt specific issues still persist. One reason you don't see these issues, may be due to how _make monitor_, when executed, is resetting the device. My advice would be, for you to play a few minutes with gatt_client, either using screen, miniterm.py or stty/cat (not idf_monitor).

Please also take a look at: https://github.com/espressif/esp-idf/issues/674