espressif/esp-idf #689: ble client cannot make successful connection with server

joicetm, 2017-06-10:
Hi,

I was working on BLE client to connect with a pulse-oximeter server, i was able to make the connection and read the data. but i am not able to make the connection consistently, ie.. i need to restart the device for around 10 to 15 times to make a successful connection.  Connection fail to establish error(bta_gattc_conn_cback() reason0x003e ) occurs. and sometimes l2cap connection canceled error.  

[image]
[image]
[image]

Weijian-Espressif, 2017-07-07:
we need more information to locate the reason, please open the bt log in bt_trace.h. Please inform if there is new information

joicetm, 2017-07-12:
Hi thank you for the reply.
I think the problem was that i couldn't pair the device properly. it would be much helpful if you give some guidance on how to perform a successful pairing from gatt client. the target server device is using just works pairing.
i was able to read data from other devices which does not require a pairing without any problem.
details of bt server : NONIN 3230 pulse-oximeter
url : http://www.nonin.com/OEMSolutions/Nonin_3230_Bluetooth_SMART

Weijian-Espressif, 2017-07-13:
hi @joicetm , we have a demo "gatt_security_server" as server. you can refer to the demo. 

In our gattc_demo , when get server pairing request, gattc would get ESP_GAP_BLE_SEC_REQ_EVT in gap callback func, you need send a response like this  :

static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{
    uint8_t *adv_name = NULL;
    uint8_t adv_name_len = 0;
    switch (event) {
    case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT: {
        //the unit of the duration is second
        uint32_t duration = 30;
        esp_ble_gap_start_scanning(duration);
        break;
    }
    case ESP_GAP_BLE_SCAN_START_COMPLETE_EVT:
        //scan start complete event to indicate scan start successfully or failed
        if (param->scan_start_cmpl.status != ESP_BT_STATUS_SUCCESS) {
            ESP_LOGE(GATTC_TAG, "Scan start failed");
        }
        break;
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
            esp_log_buffer_hex(GATTC_TAG, scan_result->scan_rst.bda, 6);
            ESP_LOGI(GATTC_TAG, "Searched Adv Data Len %d, Scan Response Len %d", scan_result->scan_rst.adv_data_len, scan_result->scan_rst.scan_rsp_len);
            adv_name = esp_ble_resolve_adv_data(scan_result->scan_rst.ble_adv,
                                                ESP_BLE_AD_TYPE_NAME_CMPL, &adv_name_len);
            ESP_LOGI(GATTC_TAG, "Searched Device Name Len %d", adv_name_len);
            esp_log_buffer_char(GATTC_TAG, adv_name, adv_name_len);
            ESP_LOGI(GATTC_TAG, "\n");
            if (adv_name != NULL) {
                if (strlen(device_name) == adv_name_len && strncmp((char *)adv_name, device_name, adv_name_len) == 0) {
                    ESP_LOGI(GATTC_TAG, "Searched device %s\n", device_name);
                    if (connect == false) {
                        connect = true;
                        ESP_LOGI(GATTC_TAG, "Connect to the remote device.");
                        esp_ble_gap_stop_scanning();
                        esp_ble_gattc_open(gl_profile_tab[PROFILE_A_APP_ID].gattc_if, scan_result->scan_rst.bda, true);
                        //esp_ble_gattc_open(gl_profile_tab[PROFILE_B_APP_ID].gattc_if, scan_result->scan_rst.bda, true);
                    }
                }
            }
            break;
        case ESP_GAP_SEARCH_INQ_CMPL_EVT:
            break;
        default:

            break;
        }
        break;
    }

    case ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT:
        if (param->scan_stop_cmpl.status != ESP_BT_STATUS_SUCCESS){
            ESP_LOGE(GATTC_TAG, "Scan stop failed");
        }
        else {
            ESP_LOGI(GATTC_TAG, "Stop scan successfully");
        }
        break;

    case ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT:
        if (param->adv_stop_cmpl.status != ESP_BT_STATUS_SUCCESS){
            ESP_LOGE(GATTC_TAG, "Adv stop failed");
        }
        else {
            ESP_LOGI(GATTC_TAG, "Stop adv successfully");
        }
        break;
    //******** pair request need response *************//
    case ESP_GAP_BLE_SEC_REQ_EVT:
        ESP_LOGI(GATTC_TAG, "-----ESP_GAP_BLE_SEC_REQ_EVT");
        esp_ble_gap_security_rsp(param->ble_security.ble_req.bd_addr, true);
        break;

    case ESP_GAP_BLE_LOCAL_ER_EVT:
        ESP_LOGI(GATTC_TAG, "-----ESP_GAP_BLE_LOCAL_ER_EVT");
        break;
    case ESP_GAP_BLE_KEY_EVT:
        ESP_LOGI(GATTC_TAG, "-----ESP_GAP_BLE_KEY_EVT");
        ESP_LOGI(GATTC_TAG, "key type = %s", esp_key_type_to_str(param->ble_security.ble_key.key_type));
        break;
    case ESP_GAP_BLE_AUTH_CMPL_EVT:
        ESP_LOGI(GATTC_TAG, "-----ESP_GAP_BLE_AUTH_CMPL_EVT");
        ESP_LOGI(GATTC_TAG, "address type = %d", param->ble_security.auth_cmpl.addr_type);
        ESP_LOGI(GATTC_TAG, "pair status = %s",param->ble_security.auth_cmpl.success ? "success" : "fail");
        break;
    default:
        ESP_LOGI(GATTC_TAG, "=========gap event %d", event);
        break;
    }
}

joicetm, 2017-07-17:
Hi Weijan,

thanks for the reply, much appreciated.
I have already tried this, and worked a few times out of too many attempts.
actually my server is damaged now. until i get the new device which i have
ordered, i could not test it.
now i know that i was going in the right direction with pairing,

once i get my new server module, i will reply with more details

once again thanks for the support.
Joice

Weijian-Espressif, 2017-07-17:
👌

FayeY, 2017-08-03:
Hi @joicetm , is there any update?

FayeY, 2017-08-07:
Hi @joicetm , I‘m going to close this issue. If your problem is still unsolved, please feel free to reopen it.

closed 2017-08-07 by FayeY