espressif/esp-idf #691: LWIP build fails when enabling lwip option MEMP_USE_CUSTOM_POOLS

ukfasthands, 2017-06-10:
ESP_THREAD_SAFE macros have same name as some structure members so preprocessor is substituting macros.

eg in sockets.h there os a macro:
#define recv(s,mem,len,flags)                     lwip_recv_r(s,mem,len,flags)

in the udp.h the structure struct udp_pcb has a member variable recv.

When compiling referencing code to udp_pcb.recv are been substituted to udp_pcb.lwip_recv_r

zhangyanjiaoesp, 2017-09-22:
@ukfasthands  I want to know whether you have tested this :"When compiling referencing code to udp_pcb.recv are been substituted to udp_pcb.lwip_recv_r".

zhangyanjiaoesp, 2017-09-22:
@ukfasthands I have test about this, When compiling referencing code to udp_pcb.recv, it didn't substitute to udp_pcb.lwip_recv_r.
when call pcb_new() , the "udp_pcb.recv" will be assigned a callback function, its function is to send a mbox to recvmbox when data is received.   However, lwip_recv_r() is to fetch  from recvmbox, and save the data into netbuf.

ukfasthands, 2017-09-25:
I'm building under osx 
I can recreate by building the ethernet example in esp-idf/examples/ethernet/ethernet
I set MEMP_USE_CUSTOM_POOLS to 1  in  esp-idf/components/lwip/include/lwip/lwip/opt.h

build error is: 

/Users/scott/esp32/esp-idf/components/lwip/core/raw.c: In function 'raw_input':
/Users/scott/esp32/esp-idf/components/lwip/core/raw.c:158:20: error: 'struct raw_pcb' has no member named 'lwip_recv_r'
         eaten = pcb->recv(pcb->recv_arg, pcb, p, ip_current_src_addr());
                    ^
make[1]: *** [core/raw.o] Error 1
make: *** [lwip-build] Erro

gcc version:
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-61-gab8375a)

zhangyanjiaoesp, 2017-09-26:
@ukfasthands ok, I will test it again according to your steps, and reply you as soon as possible.

zhangyanjiaoesp, 2017-12-08:
Hi @ukfasthands this issue is already fixed in latest IDF, please check. 
commit ID: 1c0543fb17ec700d27eaad9c379407047e82eb61

closed 2017-12-11 by FayeY