espressif/esp-idf #692: Value of esp_get_idf_version()

nkolban, 2017-06-11:
When I examine the value of esp_get_idf_version() as of the date of this post using the current master version of the ESP-IDF, I find that its value is:

v2.0-rc1-930-g058eb26

My puzzle is that we are now post V2.0.  I am thinking that the release value should be something other than v2.0 ... for example v3.0 or ???

tomROLI, 2017-06-11:
Pretty sure `v2.0-rc1-930-g058eb26` would indicate that your current version is 930 commits past the v2.0-rc1 tag and that you're commit hash is `58eb26`. You should see the same thing after running `git describe` when in the repo directory.

joicetm, 2017-06-13:
Hi Kolban,

Have you worked with ble client ?, if yes what is the correct procedure to perform bonding in ble client.
Thank you for all your support.

igrr, 2017-08-17:
We have started tagging development versions, so now the version output should be less confusing. For example, the next version to be released is 3.0, and the output will look like
```
v3.0-dev-NNN-gXXXXXX
```
where NNN is the number of commits since v3.0 development has started, and XXXXXX is the short commit ID.

closed 2017-08-17 by igrr