espressif/esp-idf #695: btm_sec_disconnected clearing pending flag handle: 0 reason:40

Nicholas3388, 2017-06-13:
Hi, I create a gatt server based on the gatts sample provided in esp-idf. I got "btm_sec_disconnected clearing pending flag handle: 0 reason:40" sometimes when I connect to my gatt server using Android phone. My phone is 360 N4s, with Android 6.0 inside. Sometimes, I got "btm_sec_disconnected clearing pending flag handle: 0 reason:31". I didn't find any error number define as 40 or 31. Is there anyone know what's the meaning for disconnect reason 40/31? and is there any suggestion on how to solve the disconnection?

Nicholas3388, 2017-06-15:
It seems like the connection is disconnected by some Android phone.

Nicholas3388, 2017-06-15:
When I use the gatt_server provided by esp-idf, I face the same issue. Some Android phone will disconnect with disconnect reason 40

Yulong-espressif, 2017-08-09:
@Nicholas3388 Are you using the Smart Phone's system setting BT to connect to the ESP32 gatt server directly? If that, it can't connect successfuly, Smart Phone's system setting BT should achieve BLE security method, and the gatt server demo don't have the security. you can use the APP like BLEdeng to connect to the ESP32 gatt server instead of system setting BT directly, hope this answer can help you, thanks.

closed 2017-09-20 by FayeY